ghdl/ghdl #2652: Constant definitions in pure functions using attributes of signals generates unexpected warnings

Araneidae, 2024-04-23:
**Description**
The attributes of signals are constant values, so it would seem reasonable to be able to use them within pure functions.  If the VHDL standard forbids this this still ought to be permitted when setting `-frelaxed`.  This report arises from exploring issues arising from issue #2651.

**Expected behaviour**
The following example code is accepted without comment by ModelSim and it would be good if ghdl also accepted this code.

Edit: see below, I hadn't spotted that these are suppressible warnings, but I think there possibly remains a case for not generating these warnings.

**How to reproduce?**
```vhd :file: func.vhd
library ieee;
use ieee.std_logic_1164.all;
 
entity func is
    port (
        data_i : in std_ulogic_vector;
        data_o : out std_ulogic_vector
    );
end;
 
architecture arch of func is
    function extend(data : std_ulogic_vector) return std_ulogic_vector
    is
        constant WIDTH_IN : natural := data_i'LENGTH;
        constant WIDTH_OUT : natural := data_o'LENGTH;
        variable result : std_ulogic_vector(WIDTH_OUT-1 downto 0);   
    begin
        result(WIDTH_IN-1 downto 0) := data;
        result(WIDTH_OUT-1 downto WIDTH_IN) := (others => '0');
        return result;
    end;
begin
    data_o <= extend(data_i);
end;
```
analysed by command
```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 -frelaxed ent.vhd
```
generates error messages:
```
func.vhd:14:40:warning: reference to port "data_i" violate pure rule for function "extend" [-Wpure]
        constant WIDTH_IN : natural := data_i'LENGTH;
                                       ^
func.vhd:15:41:warning: reference to port "data_o" violate pure rule for function "extend" [-Wpure]
        constant WIDTH_OUT : natural := data_o'LENGTH;
                                        ^
```
Oh.

Didn't spot that these were only warnings.  Hmm.  I see that `-Wno-pure` suppresses these messages.

Is there an argument for `-frelaxed` accepting this construction (and the `'SUBTYPE` usage mentioned in https://github.com/ghdl/ghdl/issues/2651#issuecomment-2071665564) without demur?  Otherwise close this, and apologies for the noise!

**Context**
- OS: RHEL8
- Origin: Built from sources using NIX at commit https://github.com/ghdl/ghdl/commit/98e90ffe8547aa03763887730fa38da61ede65c4

tgingold, 2024-04-23:
Yes, I agree with this one: it makes sense to be allowed with `-frelaxed`.  I will have a closer look.

tgingold, 2024-04-24:
Currently `-frelaxed` turns any pure error into a warning which can be made silent with `-Wno-pure` (as you have noticed).

Araneidae, 2024-04-24:
Hmm, that doesn't sound right to me.  Would the following code pass with `-frelaxed`:
```vhdl
signal blah : natural;
function add_blah(value : natural) return natural is begin return blah + value; end;
```
?

I wouldn't want it to ... but on the other hand we're perilously close to inventing a new language here.  Certainly code like this isn't accepted by ModelSim.

tgingold, 2024-04-24:
It is accepted by ghdl *if* you use `-frelaxed` but you get a warning.
I think this is almost required as even the standard vital libraries do such a reference.

Araneidae, 2024-04-25:
Can we pin this down a little?  My understanding of VHDL functions is that they are "pure" in the sense that variable outputs _only_ depend on arguments, but constants can be used within the definition of a function.  I would like to keep this notion of a "pure" function, as I think it is valuable (and the `impure` keyword is lurking there for exactly this application), however I think there is some ambiguity about what constitutes a "constant".

In particular, the attributes of variables and signals are constants (apart from the dynamic attributes), and I would have expected them to be treated as such in function definitions, that has been my experience with Vivado and ModelSim, and it matches my conception of function purity.

So, I have two questions:
1. Does the VHDL standard really forbid the use of constant signal and variable attributes in a pure function definition?
2. Would it be possible for `-frelaxed` to still be strict about function purity, or to have some other way to keep this?

Paebbels, 2024-04-25:
A pure function has no side effects and does not rely on non-static parameters outside of it's scope. Thus, a function called with same parameters yields the same result. That means: computation results of a such a function can be cached and once synthesized functions can be reused because the represented circuitry will be identical.

In your case, `data_i'LENGTH` is from an outer scope. At analysis time, that size is unknown, because the entity uses an unconstrained type for it's port `data_i`, which is known at elaboration time.

I suspect pure functions of VHDL &le;2019 need that information at analysis time.

Araneidae, 2024-04-25:
This is kind of true ... *but* if you assign `data_i'LENGTH` to a constant in the outer scope then the function is treated as pure, and I think this undermines this argument -- a function _is_ allowed to depend on (computed) constants in an outer scope ... let me check this ... yes:
```vhdl
entity test_others is
    port (                            
        data_i : in std_ulogic_vector;
--        value1_o : out natural;
        value2_o : out natural
    );
end;
                                   
architecture arch of test_others is                 
    constant DATA_LENGTH : natural := data_i'LENGTH;
--    function test1 return natural is              
--    begin                    
--        return data_i'LENGTH; 
--    end;                            
    function test2 return natural is
    begin                   
        return DATA_LENGTH;                                                     
    end;                                                                        
begin                                                                           
--    value1_o <= test1;
    value2_o <= test2;                                                          
end;
```
This works, but the commented out part fails; both functions are effectively identical.

I don't think the distinction between analysis and elaboration is terribly helpful except as an implementation detail, the behaviour of an entity is entirely determined by its particular instantiation.

nselvara, 2024-04-25:
The length attribute leads to a constant that's why you can use it in a (pure) function definition. Even if the arrays of outer scope's ports/signals/variables are unconstrained, the final port mapping in the end have to lead to constrained ones! So the length can be evaluated at early stage. Any pure function has to return always the same value for given algorithm and optionally arguments. If you begin to use global port's/signal's/variable's _**value(s)**_, then the algorithm is not anymore predictable.
So in my opinion it shouldn't be considered a warning at all.

tgingold, 2024-04-25:
@Araneidae, it's in LRM 4.3 subprogram bodies:
```
Similarly, if a pure function subprogram contains a reference to an explicitly declared signal or variable
object, or a slice or subelement (or slice thereof) of an explicit signal, then that object shall be declared
within the declarative region formed by the function;
```

For your second point, I have to check.

tgingold, 2024-04-25:
@nselvara I think we all agree with that, except this is not what the LRM says.

nselvara, 2024-04-25:
@tgingold, but from the wording it appears to me that any reference to an object (signal, port and/or variable) itself is considered impure but not the attributes themselves though. Because calling an attribute yields to something that's not _explicitely_ defined.

tgingold, 2024-04-25:
@nselvara But how can you reference the attribute of an object without reference first the object ?

Only variables, signals and files are considered for pure functions.  Types or subtypes are not.

Araneidae, 2024-04-26:
I guess we do have to argue about the exact language, and I guess @tgingold you have to be the judge, but it does seem to me that we should strive to seek an interpretation of the language that makes the best sense in terms of language design.  Is the word "reference" clearly defined in the standard, or is this just normal language?  Are we _obliged_ to treat an expression of the form `symbol'ATTR` as a "reference" to `symbol`?

If so, this strikes me as a clear defect in the language specification which surely doesn't capture the underlying intention of the definition of a "pure" function.

I guess you're trying to maintain two versions of VHDL, which is a bit of a painful place to be:
* The pedantic version that matches the strictest reading of the language specification
* The (IMHO) "sensible" version that relaxes the pointless constraints introduced by the language

The problem with this, of course, is what is the "sensible" version of VHDL?  I guess I'm arguing that Vivado<sub>2008</sub>∩ModelSim<sub>2008</sub> isn't a bad place to start.

There are a frustrating number places in VHDL where I can't write something ... just because ... but if I declare some functions and some auxiliary signals and constants and write it out the long winded way it can be made to work.  I do think that VHDL is a deeply flawed language ... but it's the best we've got 😢.

nselvara, 2024-04-26:
@tgingold, I kind of see your point that the references of an object in a function should be in the declarative region of such functions but now the million dollar question is if the LRM also counts evaluating attributes as a reference to the object therefore also needs to be defined in the region or is it just an edge case that the LRM didn't see coming. But as [Araneidae](https://github.com/Araneidae) said, it's kind of a bit of a grey zone. 
And tbh a strict interpretation of that doesn't make any sense, as generic functions are also kind of then contradicting. Because the reference to a generic type in the function is formed in the argument's list and if that also counts as declarative region then we  have a generic type that's defined but we just used a small workaround. 🙃

tgingold, 2024-04-27:
So now there is not anymore message for the attributes.
Should pure issues still be a warning (and not an error) with `-frelaxed` ? I fear this may break some designs.
(and I'd prefer not to add another flag).

Araneidae, 2024-04-29:
I need to get my colleague to rebuild the latest ghdl, but as you already have `-Werror` as a flag I think I can focus on the combination `--std=08 -frelaxed -Werror` to give the language variant I'm hoping for.  Sounds good so far.  So yes, please keep true pure issues (actual _uses_ of external signals and variables) as warnings!

I'll do some more experiments and report back if I find anything amiss.

nselvara, 2024-04-29:
@tgingold, I think it would make sense to output a warning as it's kind of an error if we interpret it strictly. But maybe you could link the warning to this issue. I don't know if that's a thing here resp. adheres to the philosophy, as I just started using GHDL only recently.

tgingold, 2024-05-02:
Yes, we should handle with `-Wpure`.  The default should be an error (`-Werror=pure`), so that it could be turned to a warning with `-Wno-error=pure` or completely off with `-Wno-pure`.  `-frelax` will have no impact on that.
I will try to implement that and to see the fallout on the testsuite.

tgingold, 2024-05-09:
I tried, but this is not that simple: there are many packages (including vendor libraries) which violate pure rules.
(returning access or using files in functions).
I prefer to keep the current behaviour, unless someone has a better suggestion.

closed 2024-05-09 by tgingold

Araneidae, 2024-05-09:
Interesting and irritating: the `impure` keyword is lurking there shouting "use me"!  Ah well 🤷

What flag combination are you using to read these libraries now?

tgingold, 2024-05-09:
There is no particular flags: the default is `--std=93c` which is equivalent to `--std=93 -frelaxed`.
Of course, you get warnings.

commit 889d4ed9 (2024-04-27): vhdl: call check_pure when an object is read/modified.

Fix #2652