ghdl/ghdl #2656: .all in access type variable assignment reveals a bug

Dotto92, 2024-05-04:
Assignment to access type variable a (a(a'range).all), at the syntax check (ghdl -s ......), returns with a bug error:

```
......

process is
  variable a : mem_ptr;
  variable l : line;
  begin
	a := new mem_t;
	a(a'range).all := (others => (others =>'1'));

.......

```
returns with:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.1.0 (tarball) [Dunoon edition]
Compiled with GNAT Version: Community 2020 (20200818-84)
Target: x86_64-apple-darwin17.7.0
Command line:
ghdl -s -fsynopsys example.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
******************************************************************
```

Without .all the code works correctly

Paebbels, 2024-05-04:
Isn't the syntax `a.all(a'range)`?

First dereference the pointer, then access a range in the array.  
An explicit dereference is not needed, that's why `a(a'range)` works.  
`a(a'range).all` would try to dereference an array element, but not the pointer.

------
Sure, GHDL should not crash and report a proper error.

Paebbels, 2024-05-04:
@Dotto92 please provide the type of `mem_t` and `mem_ptr`.

Dotto92, 2024-05-04:
> @Dotto92 please provide the type of `mem_t` and `mem_ptr`.

The type of mem_t and mem_ptr are:

```
type mem_t is array (0 to 15) of std_logic_vector (7 downto 0);
type mem_ptr is access mem_t;
```

Dotto92, 2024-05-04:
> Isn't the syntax `a.all(a'range)`?
>
> First dereference the pointer, then access a range in the array. An explicit dereference is not needed, that's why `a(a'range)` works. `a(a'range).all` would try [...]

Thanks for your help.
Ok, it’s my fault.
I opened this issue just to report the bug.

closed 2024-05-05 by tgingold

Paebbels, 2024-05-05:
> Ok, it’s my fault.

We appreciate your issue, because we also want to close all corner cases where GHDL isn't printing a clean error message instead of a crash. So even if your code was not fully correct, that code snippet and your issue helped us to improve the VHDL parser and error reporting.

@tgingold derived a test case from your code, so it's now part of the regression testing.

commit 4a84b473 (2024-05-05): vhdl-sem_names: avoid a crash on invalid dereference

Fix #2656