ghdl/ghdl #2658: GHDL crash when dynamic slicing is used.

consanii, 2024-05-13:
**Description**
GHDL is failing to synthesise my design and crash. It is probably due to the inability of the tool to manage the dynamic slicing of a vector.

**Expected behaviour**
The code to be synthesised correctly. Commercial tools are able to synthesise the design and simulate it. Tried with ModelSim SE-64 10.7 revision 2017.12 .

**How to reproduce?**
The problematic code is the following. I have two variables that represent an 8 bit window, and these variable moves to slice the vector. For example the right bound start at 0 and the left one at 7, and then they are both incremented by 8 till end of vector. They are of type integer and they are limited to range 0 to maxDimension.

```vhd
-- Window signals
signal rxRightBoundxDP, rxRightBoundxDN : integer range 0 to voltageChangeRegLengthC := 0;
signal rxLeftBoundxDP, rxLeftBoundxDN   : integer range 0 to voltageChangeRegLengthC := 7;

[...]

txDataxDN <= paddingVector
                & voltageChangeIntervalxDI(rxLeftBoundxDP downto rxRightBoundxDP)
                when (rxLeftBoundxDP - rxRightBoundxDP) < 7 else 
                voltageChangeIntervalxDI(rxLeftBoundxDP downto rxRightBoundxDP);
```

The command used to compile is the following:
```sh
yosys -m ghdl -p 'ghdl --no-formal --std=08 $(VHDL_FILES) -e $(TOP_MODULE);
```

**Context**
GHDL crash. Before crashing, the following errors are thrown:

```sh
../hdl/UartLogic.vhd:93:43:error: cannot extract same variable part for dynamic slice
                & voltageChangeIntervalxDI(rxLeftBoundxDP downto rxRightBoundxDP)
                                          ^
../hdl/UartLogic.vhd:95:41:error: cannot extract same variable part for dynamic slice
                voltageChangeIntervalxDI(rxLeftBoundxDP downto rxRightBoundxDP);
```

Crash report:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r903.g2803c2d4b) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-apple-darwin17.7.0
/Users/mattia/CERN/eval_board_tests_rtl/accurate/build/
Command line:

Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:783
******************************************************************
```

tgingold, 2024-05-13:
Please, provide a complete reproducer so that the issue could be investigated (and the crash reproduced).

consanii, 2024-05-14:
Sure. The following code is able to reproduce the issue and the crash on my machine.

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity BugTest is
end entity BugTest;

architecture rtl of BugTest is
	constant voltageChangeRegLengthC : integer := 52;

    -- Window signals
    signal rxRightBoundxDP, rxRightBoundxDN : integer range 0 to voltageChangeRegLengthC := 0;
    signal rxLeftBoundxDP, rxLeftBoundxDN   : integer range 0 to voltageChangeRegLengthC := 7;

	signal voltageChangeIntervalxDI : std_logic_vector(voltageChangeRegLengthC-1 downto 0);
	signal txDataxDN : std_logic_vector(7 downto 0);
	
	constant paddingNumber : natural := voltageChangeRegLengthC mod 8;
    constant paddingVector : std_logic_vector(7 - paddingNumber downto 0) := (others => '0');
    

begin
    
    -- Problematic part of the code
	txDataxDN <= paddingVector
                & voltageChangeIntervalxDI(rxLeftBoundxDP downto rxRightBoundxDP)
                when (rxLeftBoundxDP - rxRightBoundxDP) < 7 else 
                voltageChangeIntervalxDI(rxLeftBoundxDP downto rxRightBoundxDP);

end architecture rtl;
```
This gives the following output:
```sh
BugTest.vhd:25:43:error: cannot extract same variable part for dynamic slice
                & voltageChangeIntervalxDI(rxLeftBoundxDP downto rxRightBoundxDP)
                                          ^
BugTest.vhd:27:41:error: cannot extract same variable part for dynamic slice
                voltageChangeIntervalxDI(rxLeftBoundxDP downto rxRightBoundxDP);
                                        ^

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r903.g2803c2d4b) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-apple-darwin17.7.0
/Users/mattia/Desktop/
Command line:

Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:783
******************************************************************
```
I am launching GHDL with the following command:
```sh
yosys -m ghdl -p 'ghdl --no-formal --std=08 BugTest.vhd -e BugTest'
```

tgingold, 2024-05-14:
Yes, I can reproduce the crash.

closed 2024-05-14 by tgingold

tgingold, 2024-05-15:
So the crash is now fixed, however the dynamic slicing is still not supported as it is.
GHDL follows the ieee 1076.6 standard: the width of the slice must be a constant.  So you need to slightly rewrite the assignment.

commit bdac074f (2024-05-14): synth-vhdl_stmts: avoid a crash after error.

Fix #2658