ghdl/ghdl #2659: pyGHDL installation broken with latest pytooling (6.1.0)

augustofg, 2024-05-15:
Trying to install the GHDL language server seems to have no effect with recent versions of pytooling:
```bash
$ python setup build
[pyTooling.Packaging] Extracting requirements from 'pyGHDL/requirements.txt'.
[pyTooling.Packaging]   Extracting requirements from 'pyGHDL/libghdl/requirements.txt'.
[pyTooling.Packaging]   - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]   Extracting requirements from 'pyGHDL/dom/requirements.txt'.
[pyTooling.Packaging]     Extracting requirements from 'pyGHDL/dom/../libghdl/requirements.txt'.
[pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]   - pyVHDLModel==0.25.1
[pyTooling.Packaging]   Extracting requirements from 'pyGHDL/lsp/requirements.txt'.
[pyTooling.Packaging]     Extracting requirements from 'pyGHDL/lsp/../libghdl/requirements.txt'.
[pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]   Extracting requirements from 'pyGHDL/cli/requirements.txt'.
[pyTooling.Packaging]     Extracting requirements from 'pyGHDL/cli/../dom/requirements.txt'.
[pyTooling.Packaging]       Extracting requirements from 'pyGHDL/cli/../dom/../libghdl/requirements.txt'.
[pyTooling.Packaging]       - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]     - pyVHDLModel==0.25.1
[pyTooling.Packaging]   - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]   - pyAttributes>=2.3.2
[pyTooling.Packaging] Extracting requirements from 'doc/requirements.txt'.
[pyTooling.Packaging]   Extracting requirements from 'doc/../pyGHDL/requirements.txt'.
[pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/libghdl/requirements.txt'.
[pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/dom/requirements.txt'.
[pyTooling.Packaging]       Extracting requirements from 'doc/../pyGHDL/dom/../libghdl/requirements.txt'.
[pyTooling.Packaging]       - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]     - pyVHDLModel==0.25.1
[pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/lsp/requirements.txt'.
[pyTooling.Packaging]       Extracting requirements from 'doc/../pyGHDL/lsp/../libghdl/requirements.txt'.
[pyTooling.Packaging]       - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/cli/requirements.txt'.
[pyTooling.Packaging]       Extracting requirements from 'doc/../pyGHDL/cli/../dom/requirements.txt'.
[pyTooling.Packaging]         Extracting requirements from 'doc/../pyGHDL/cli/../dom/../libghdl/requirements.txt'.
[pyTooling.Packaging]         - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]       - pyVHDLModel==0.25.1
[pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]     - pyAttributes>=2.3.2
[pyTooling.Packaging] - sphinx
[pyTooling.Packaging] - python-dateutil>=2.8.2
[pyTooling.Packaging] - autoapi>=2.0.1
[pyTooling.Packaging] - myst-parser
[pyTooling.Packaging] - sphinx_fontawesome>=0.0.6
[pyTooling.Packaging] - sphinx_autodoc_typehints>=1.19.5
[pyTooling.Packaging] - furo
[pyTooling.Packaging] - btd.sphinx.autoprogram>=0.1.7.post1
$ python setup.py install --root=/tmp/pyghdl --optimize=1 --skip-build
[pyTooling.Packaging] Extracting requirements from 'pyGHDL/requirements.txt'.
[pyTooling.Packaging]   Extracting requirements from 'pyGHDL/libghdl/requirements.txt'.
[pyTooling.Packaging]   - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]   Extracting requirements from 'pyGHDL/dom/requirements.txt'.
[pyTooling.Packaging]     Extracting requirements from 'pyGHDL/dom/../libghdl/requirements.txt'.
[pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]   - pyVHDLModel==0.25.1
[pyTooling.Packaging]   Extracting requirements from 'pyGHDL/lsp/requirements.txt'.
[pyTooling.Packaging]     Extracting requirements from 'pyGHDL/lsp/../libghdl/requirements.txt'.
[pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]   Extracting requirements from 'pyGHDL/cli/requirements.txt'.
[pyTooling.Packaging]     Extracting requirements from 'pyGHDL/cli/../dom/requirements.txt'.
[pyTooling.Packaging]       Extracting requirements from 'pyGHDL/cli/../dom/../libghdl/requirements.txt'.
[pyTooling.Packaging]       - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]     - pyVHDLModel==0.25.1
[pyTooling.Packaging]   - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]   - pyAttributes>=2.3.2
[pyTooling.Packaging] Extracting requirements from 'doc/requirements.txt'.
[pyTooling.Packaging]   Extracting requirements from 'doc/../pyGHDL/requirements.txt'.
[pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/libghdl/requirements.txt'.
[pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/dom/requirements.txt'.
[pyTooling.Packaging]       Extracting requirements from 'doc/../pyGHDL/dom/../libghdl/requirements.txt'.
[pyTooling.Packaging]       - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]     - pyVHDLModel==0.25.1
[pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/lsp/requirements.txt'.
[pyTooling.Packaging]       Extracting requirements from 'doc/../pyGHDL/lsp/../libghdl/requirements.txt'.
[pyTooling.Packaging]       - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/cli/requirements.txt'.
[pyTooling.Packaging]       Extracting requirements from 'doc/../pyGHDL/cli/../dom/requirements.txt'.
[pyTooling.Packaging]         Extracting requirements from 'doc/../pyGHDL/cli/../dom/../libghdl/requirements.txt'.
[pyTooling.Packaging]         - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]       - pyVHDLModel==0.25.1
[pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
[pyTooling.Packaging]     - pyAttributes>=2.3.2
[pyTooling.Packaging] - sphinx
[pyTooling.Packaging] - python-dateutil>=2.8.2
[pyTooling.Packaging] - autoapi>=2.0.1
[pyTooling.Packaging] - myst-parser
[pyTooling.Packaging] - sphinx_fontawesome>=0.0.6
[pyTooling.Packaging] - sphinx_autodoc_typehints>=1.19.5
[pyTooling.Packaging] - furo
[pyTooling.Packaging] - btd.sphinx.autoprogram>=0.1.7.post1
$ ls /tmp/pyghdl # empty directory
```
The installation directory is empty. This issue was triggered by some change introduced in recent versions of pytooling I believe.

- OS: Archlinux x86-64
- Origin:
  - [x] Built from sources: `dc17919eb3d401da6843cb4e298b9005d36d93e9`

tgingold, 2024-05-17:
That's weird because pyTooling < 5.0 is part of the requirements.

Paebbels, 2024-05-21:
Hi. GHDL currently requires pyTooling below 5.0. Usually, this is enforced and checked when the environment is setup correctly and/or pyGHDL is packaged as a wheel and installed from wheel.

I know that this is e.g. not enforced in ArchLinux packages and some other repositories, because the requirements.txt is not used/obeyed to. I think it's directly installed from sources but not from wheel or so. (I have no ArchLinux to verify it...)

Building a wheel should be done like this:  
`python -m build --wheel`  
now you have a `*.whl` file in `dist`  
`python -m pip install .\pyGHDL-5.0.0.dev0-py3-none-any.whl`

* This uses `build` and `wheel` to build Python packages instead of `distutils`.
* This uses `pip` and `wheel` to install the package, but not `distutils`.

----
I started to work on an update for pyGHDL to use pyTooling 6.1. I need to adjust one of the APIs to avoid errors when parsing the example files.

Are you blocked by the current issue?

-----
/cc @skoehler

Paebbels, 2024-12-19:
Latest GHDL uses pyTooling v8.0+. This should be all fixed.

Please try again and reopen this issue if you see further problems.

closed 2024-12-19 by Paebbels