ghdl/ghdl #2661: pyGHDL dom/libGHDL array port item constraints might not be fully populated

Tcenova, 2024-05-23:
**Description**
It looks like pyGHDY.dom does not yet parse the constraints from std_logic_vectors and array types. I am trying to use one of the internal dom functions to get the constraints that libGHDL parsed so I can use them. When attempting to access the constraints of a portItem with an unconstrained array type, where the port definition declares the sizes, the constraints returned from libGHDL only contain the first layer array range, not any of the subtypes ranges as well. 

i_test : in t_unc_array(0 to 5)(7 downto 0)

Will return a single item in a list that defines the 0 to 5 range.

**Expected behaviour**
When requesting the constraints, I expected a list with all constraints defined after the type. So a list with (0 to 5) and (7 downto 0). 

I am not sure what the expected behavior of libGHDL is. If the current functionality is expected, is there a different node I need to request the constraints of to get the range for the subtypes?

**How to reproduce?**
Run the below code. The code also includes an example of a 2d array with the same issue. The output should look like:
```
test
Constraints:
0 to 5
  - i_test : in t_unc_array?
Constraints:
0 to 9
  - i_test2 : in t_unc_array_2d?
```

```vhd :file: test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package my_types_pkg is 
  type t_unc_array is array (natural range <>) of std_logic_vector;
  type t_unc_array_2d is array (natural range <>) of t_unc_array;
end package;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.my_types_pkg.all;

entity test is
  port (
    i_test : in t_unc_array(0 to 5)(7 downto 0);
    i_test2 : in t_unc_array_2d(0 to 9)(0 to 5)(7 downto 0)
  );
end entity;

architecture rtl of test is

begin

end architecture;
```

```python :file: test.py
from pathlib import Path
import os
from pyGHDL.dom.NonStandard import Design, Document
from pyGHDL.dom.Symbol import ConstrainedCompositeSubtypeSymbol
from pyGHDL.dom._Translate import GetArrayConstraintsFromSubtypeIndication

dutFilepath = os.path.join(os.path.dirname(__file__), "test.vhd")

dutFilePathPath = Path(dutFilepath)

design = Design()
design.LoadDefaultLibraries()
document = Document(dutFilePathPath)
design.Documents.append(document)

for entity in document.Entities.values():
    print(f"{entity.Identifier}")
    for port in entity.PortItems:
        if isinstance(port.Subtype, ConstrainedCompositeSubtypeSymbol):
            constraints = GetArrayConstraintsFromSubtypeIndication(port.Subtype._iirNode)
            print("Constraints:")
            for constraint in constraints:
                print(constraint)
        identifiers = ", ".join([str(i) for i in port.Identifiers])
        print(f"  - {identifiers} : {port.Mode!s} {port.Subtype}")
```

**Context**
Please, provide the following information:

- OS: Ubuntu 22.04
- Origin:
  - [ ] Package manager: `version`
  - [X] Released binaries: [Wheel](https://github.com/ghdl/ghdl/releases/download/v4.1.0/pyGHDL-4.1.0-py3-none-any.whl) [GHDL Binary](https://github.com/ghdl/ghdl/releases/download/v4.1.0/ghdl-gha-ubuntu-22.04-mcode.tgz)
  - [ ] Built from sources: `commit SHA`

Paebbels, 2024-05-23:
How urgent is this topic for you? I'm currently trying to fix #2659 (aligning dependencies).

Paebbels, 2024-05-23:
As you figured out, that part is not fully implemented, because constraints can be very complex. What you're looking for is not a list of constraints (this would be the case for `foo(7 downto 0, 7 downto 0)`, it's nested constraints. Because your case is not a 2-dimensional array, but a 1-dim. array in another 1-dim. array.

Tcenova, 2024-05-23:
Thank you for the clarification! TIL that VHDL allows for 2d arrays like that. Nested constraints also exist for 1d arrays with unconstrained subtype though too (2008). My example shows that as well (t_unc_array)

It would be very nice to have some form of a workaround/fix within a few weeks if possible. I am also happy to figure out a workaround for the issue on my own if this is a limitation of the python side versus a limitation of libGHDL. I would just need a hint as to how to get the info from libGHDL is all. Or even just a way to get a reference to the string representation that I can re-parse for my use cases.

Tcenova, 2024-06-10:
Small update. I was able to find a way to get constraints for the simple array cases by looking at some printing functions in ghdl and converting them over to the python interface to ghdl. 
https://github.com/ghdl/ghdl/blob/dab33f753521b5786f82d3f24ea43108bceb658e/src/vhdl/vhdl-prints.adb#L525-L547

I am unblocked for now. I am still trying to learn how to get constraints from types that are constrained at definition (Versus fully unconstrained as in my example). And I also need to get constraints of record elements (constrained or unconstrained).

Paebbels, 2024-12-19:
It's winter time and I'll spend the next 4-5 weeks on pyGHDL updates (besides other Python related topics) :)