ghdl/ghdl #2662: Help with "error: use of synopsys package "std_logic_arith" needs the -fsynopsys option"

ghelbig, 2024-05-24:
I'm using Ubuntu 22.04 in WSL with GHDL 1.0.0 (Ubuntu 1.0.0+dfsg-6) [Dunoon edition] and Vivado 2019.3

I get that error from an instantiated primitive: Xilinx/Vivado/2019.1/data/vhdl/src/unisims/primitive/IDELAYE2.vhd:28:10

I re-compiled the Vivado unisim making sure that the -fsynopsys option was set (I added it to the Tcl file):
 
/usr/bin/ghdl -a --mb-comments -fexplicit -fsynopsys -Wbinding --ieee=synopsys --no-vital-checks --std=93c -frelaxed -P/home/ghelbig/Projects/<redacted>/sim/xilinx-vivado -v --work=unisim --workdir=/home/ghelbig/Projects/FiberLinkInterface/sim/xilinx-vivado/unisim/v93 "/home/ghelbig/Xilinx/Vivado/2019.1/data/vhdl/src/unisims/primitive/IDELAYE2.vhd"

I also recompiled all of the files in my simulation with the same options.

I'm still getting the error message.  Perhaps I'm doing something wrong?  Perhaps the error message is leading me astray?

Thanks,
Gary.

ghelbig, 2024-05-24:
I think I found it:  The -fsynopsys has to be added to the elaborate step?

Paebbels, 2024-05-24:
GHDL v1.0.0 is very old and not supported anymore. Please update and install v4.1.0 (latest stable release) or compile from sources (v5.0.0-dev). This will not solve your issue, but there have been many updates, improvements and bugfixes since v1.0.0.

All analyzed files need to use the same VHDL standard. So mixing VHDL-93 and VHDL-2008 is not possible. In addition, many options need to be the same for analysis and elaboration and running the simulation. E.g. with `-fsynopsys`, additional non IEEE packages (used by Xilinx) get mapped. This mapping needs to be active in your analysis calls and elaboration calls.

Hint: GHDL provides pre-compile scripts to pre-compile vendor libraries like Vivado.  
See https://ghdl.github.io/ghdl/getting.html#precompile-vendor-primitives

closed 2024-05-26 by tgingold