ghdl/ghdl #2665: Synthesis when using fixed_pkg produces `asserts`

rapgenic, 2024-05-29:
**Description**

I am writing some code to perform fixed point arithmetic, using the `fixed_pkg` package. I am using yosys+ghdl+nextpnr flow, and nextpnr complains that `cell type '$assert' is unsupported`. I checked the netlist produced with `ghdl synth` and I actually found some `assert` directives, which I believe are not synthesizable, hence the errors...

I commented them out and included the synthesized vhdl netlist in yosys, which correctly produced a netlist that nextpnr accepted.

**Expected behaviour**
I suppose the correct behaviour should be that the assert instructions are not present in the synthesized netlist.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.NUMERIC_STD.all;
use ieee.fixed_pkg.all;

entity demo is
	port (
		data_in : in sfixed(0 downto -15);
		data_out : out unsigned(31 downto 0)
	);
end entity;

architecture behavioral of demo is
	constant K : sfixed(5 downto -11) := to_sfixed(23.13037109375, 5, -11);
	signal temp : sfixed(21 downto -11);
begin
	temp <= (data_in * K);
	data_out <= unsigned(to_signed(temp, data_out'length));
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl analyze --std=08 bug.vhd 
ghdl synth --std=08 demo | grep assert && false
```

```vhdl
-- SYNTHESIZED OUTPUT
-- File: synth.vhdl

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.fixed_pkg.all;
entity demo is
  port (
    data_in: in sfixed (0 downto - 15);
    data_out: out unsigned (31 downto 0)
  );
end entity;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture rtl of demo is
  subtype typwrap_data_in is std_logic_vector (15 downto 0);
  signal wrap_data_in: typwrap_data_in;
  subtype typwrap_data_out is std_logic_vector (31 downto 0);
  signal wrap_data_out: typwrap_data_out;
  -- MORE CODE
begin
  -- MORE CODE
  -- ../../src/ieee2008/fixed_generic_pkg-body.vhdl:89:5
  n22: postponed assert n21_o = '1' severity error; --  assert    <================================================
  -- ../../src/ieee2008/fixed_generic_pkg-body.vhdl:553:24
  n32_o <= std_logic_vector (resize (signed (wrap_data_in), 33));  --  sext
  -- ../../src/ieee2008/fixed_generic_pkg-body.vhdl:553:24
  n35_o <= std_logic_vector (resize (signed (n32_o) * signed'("000000000000000001011100100001011"), 33));
  -- ../../src/ieee2008/fixed_generic_pkg-body.vhdl:89:5
  n69_o <= not '1';
  -- ../../src/ieee2008/fixed_generic_pkg-body.vhdl:89:5
  n70_o <= n69_o or '1';
  -- ../../src/ieee2008/fixed_generic_pkg-body.vhdl:89:5
  n71: postponed assert n70_o = '1' severity error; --  assert    <================================================
  -- ../../src/ieee2008/fixed_generic_pkg-body.vhdl:3457:19
  n73_o <= temp (32 downto 11); 
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS: Fedora 40
- Origin:
  - [ ] Package manager: `version`
  - [x] Released binaries: `oss-cad-suite (by yosys) version`
     ```
     OSS CAD Suite info

     Product: oss-cad-suite
     Architecture: linux-x64
     Version: 20240511
     Package: oss-cad-suite-linux-x64-20240511
     ```
  - [ ] Built from sources: `commit SHA`

tgingold, 2024-05-30:
Did you try to use `--no-formal` option ?

rapgenic, 2024-05-30:
Thank you very much for the fast answer!! Yes it actually synthesizes correctly now! 

Obviously it was very clearly written in the documentation... sorry for bothering you

On a side note can I ask on what circumstance would i _not_ use the `--no-formal` option?

closed 2024-05-30 by rapgenic

tgingold, 2024-05-30:
If you do formal verification or if your synthesize script remove them by default.