ghdl/ghdl #2667: Synthesis (for formal) misbehaves

Topi-ab, 2024-06-02:
**Description**
Two different ways in writing same thing in VHDL produces two different outputs in synthesis (formal), but equal outputs in simulation.

**Expected behaviour**
The segs_1_out and segs_2_out producing always same output

**How to reproduce?**
Run cover case with symbiyosys for the following example.

The cover result is:

[image]

Simulation (you need to comment PSL part away from bin_to_7seg.vhdl) result:

[image]

```vhd :file: bin_to_7seg.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity bin_to_7seg is
	port(
		clk_in: in std_logic;
		
		bin_in: in unsigned(3 downto 0);
		
		segs_1_out: out std_logic_vector(6 downto 0);
		segs_2_out: out std_logic_vector(6 downto 0)
	);
end;

architecture rtl of bin_to_7seg is
	type seg_a is array(0 to 15) of std_logic_vector(6 downto 0);
	constant seg_c: seg_a := (
		"1111110", -- 0
		"0110000", -- 1
		"1101101", -- 2
		"1111001", -- 3
		"0110011", -- 4
		"1011011", -- 5
		"1011111", -- 6
		"1110000", -- 7
		"1111111", -- 8
		"1111011", -- 9
		"1110111", -- A
		"0011111", -- B
		"1001110", -- C
		"0111101", -- D
		"1001111", -- E
		"1000111" -- F
	);
	
	signal cnt: integer range 0 to 6 := 6;
begin
	process(clk_in)
		variable s: std_logic_vector(6 downto 0);
	begin
		if rising_edge(clk_in) then
			if cnt < 6 then
				cnt <= cnt + 1;
			end if;
			if cnt = 6 then
				cnt <= 0;
			end if;

			segs_1_out <= (others => '0');
			s := seg_c(to_integer(bin_in));
			segs_1_out(cnt) <= s(cnt);
			
			segs_2_out <= (others => '0');
			segs_2_out(cnt) <= seg_c(to_integer(bin_in))(cnt);
		end if;
	end process;
	
	default clock is rising_edge(clk_in);
	
	a_1: assume always stable(bin_in);
	
	f_1: assert always {true; onehot0(segs_1_out)};
	
	c_1: cover {bin_in = 1; [*15]};
end;
```

```sby :file: ent.sby
[tasks]
bmc
cover

[options]
bmc: mode bmc
bmc: depth 250
cover: mode cover
cover: depth 250

[engines]
bmc: smtbmc
cover: smtbmc

[script]
ghdl --std=08 bin_to_7seg.vhdl -e bin_to_7seg
prep -top bin_to_7seg

[files]
bin_to_7seg.vhdl
```

```vhd :file: bin_to_7seg.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tb_bin_to_7seg is
end;

architecture tb of tb_bin_to_7seg is
	signal clk: std_logic;
	signal bin: unsigned(3 downto 0);
	
	signal segs_1: std_logic_vector(6 downto 0);
	signal segs_2: std_logic_vector(6 downto 0);
begin
	clk_pr: process
	begin
		clk <= '0';
		wait for 5 ns;
		clk <= '1';
		wait for 5 ns;
	end process;
	
	bin_pr: process
	begin
		bin <= to_unsigned(1, bin);
		wait;
	end process;

	dut: entity work.bin_to_7seg
		port map(
			clk_in => clk,
			bin_in => bin,
			segs_1_out => segs_1,
			segs_2_out => segs_2
		);

end;
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS: WSL (Win 11)
- Origin:
docker image "hdlc/formal:all"
ghdl --version
GHDL 5.0.0-dev (v4.1.0-24-gdc17919eb) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2024-06-04:
I can reproduce the issue; I think this is an error during synthesis, but I am investigating.

tgingold, 2024-06-05:
Should now be fixed.  Thank you for the issue!

commit 3cb2027d (2023-05-16): netlists-memories: avoid a crash.  Fix #2667
commit 406a7cba (2024-06-05): netlists-memories: handle multi-dim memories with any ranges

Fix handling of ranges which are not a power of 2

Fix #2667
closed 2024-06-05 by tgingold