ghdl/ghdl #2668: formal: requires initial value

NikLeberg, 2024-06-03:
**Description**
When running a formal test with SymbiYosys I encountered an issue with non initialized signals. If a signal is non initialized and only has an assume of its value, an assert for the assumed value triggers.

**Expected behaviour**
I would assume for the `assume` to enough. It's not clear to me why there needs to be an initial value.

I'm not sure if my PSL statements are correct though.. I sadly don't have access to the official PSL/VHDL standard to read up on how the initial values are treated. If it describes something and you have access to it, I'd love to have a copy of the initial value handling definition.

**How to reproduce?**
Minimal Working Example (MWE):

```vhd :file: foo.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity foo is
  port(
    clk : in std_logic;
    reset : in std_logic
  );
end foo;

architecture arch of foo is
  signal a : std_logic; -- fails
  -- signal a : std_logic := '0'; -- passes
  -- signal a : std_logic := 'U'; -- fails
begin
  assume always a = '1' @rising_edge(clk);
  assert always a = '1' @rising_edge(clk);
end arch;
```

```sby :file: foo.sby
[tasks]
bmc

[options]
bmc: mode bmc
multiclock on

[engines]
smtbmc --progress --nopresat yices

[script]
ghdl --std=08 foo.vhd -e foo
prep -top foo

[files]
foo.vhd
```

Run with `SynbiYosys` + `ghdl-yosys-plugin`:
```sh :image: ghcr.io/nikleberg/spinalhdl-with-ghdlyosys:0.1-dev-staging
sby --yosys "yosys -m ghdl" -f foo.sby
```

**Context**
Please, provide the following information:

- OS: Ubuntu-24.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: git tag `v4.1.0` or `7188e92cf3cc60f5d97cfe83bc2c85f0250373ab`
```
GHDL 4.1.0 (4.1.0.r0.g7188e92cf) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 static elaboration, mcode JIT code generator
```

Thanks!

tmeissner, 2024-06-03:
I think, "plain" assume only works on input ports. If you wanna use it with internal signals, you can define an anyconst or anyseq attribute for these signals. Then assumptions on these signals should work. For an example, see the yosys_anyconst.vhd file in my psl_with_ghdl repository. There is always some information in the yosys docs about these special attributes: https://symbiyosys.readthedocs.io/en/latest/verilog.html (unconstrained variables).

NikLeberg, 2024-06-03:
Only being able to `assume` on inputs makes actually a lot of sense then. Your `psl_with_ghdl` is an real treasure trove by the way. I always come back to it when I want to learn something new about PSL :) .

Thanks for the good example!

So basically, either use `assume` only on input signals or on unconstrained internal signals.

```vhd
entity foo is
  port(
    clk : in std_logic;
    reset : in std_logic;
    a : in std_logic
  );
end foo;

architecture arch of foo is
  signal b : std_logic;

  attribute anyconst : boolean;
  signal c : std_logic;
  attribute anyconst of c : signal is true;
begin
  default clock is rising_edge(clk);

  -- input (OK)
  ASSUME_a: assume always a = '1';
  ASSERT_a: assert always a = '1';

  -- internal signal (FAIL)
  --ASSUME_b: assume always b = '1';
  --ASSERT_b: assert always b = '1';

  -- internal unconstrained signal (OK)
  ASSUME_c: assume always c = '1';
  ASSERT_c: assert always c = '1';
end arch;
```

closed 2024-06-03 by NikLeberg