ghdl/ghdl #2669: formal: bug when assert with `async_abort`

NikLeberg, 2024-06-03:
**Description**
When `async_abort` is used in a PSL style `assert` an error/exception is thrown. If the seemingly equivalent `abort` (which should only be an alias) is used, the statement works (at least can be analyzed).

**Expected behaviour**
Identical behaviour between `async_abort` and `abort`.

**How to reproduce?**
Minimal Working Example (MWE):

```vhd :file: foo.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity foo is
  port(
    clk : in std_logic;
    reset : in std_logic
  );
end foo;

architecture arch of foo is
  signal a : std_logic;
begin
  assert always (a = '1')       abort (reset = '1') @rising_edge(clk); -- ok
  assert always (a = '1')  sync_abort (reset = '1') @rising_edge(clk); -- ok
  assert always (a = '1') async_abort (reset = '1') @rising_edge(clk); -- error
end arch;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 foo.vhd
```

**Context**
Please, provide the following information:

- OS: `Ubuntu-24.04`
- Origin:
  - [x] Built from sources: git tag `v4.1.0` or `7188e92cf3cc60f5d97cfe83bc2c85f0250373ab`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
build_property_fa: cannot handle N_ASYNC_ABORT

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.1.0 (4.1.0.r0.g7188e92cf) [Dunoon edition]
Compiled with GNAT Version: 13.2.0
Target: x86_64-linux-gnu
/workspaces/SpinalHDL/simWorkspace/FormalApiTest_31/
Command line:
ghdl -a --std=08 rtl/unamed.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : psl-errors.adb:37
******************************************************************
```

**Additional context**
As `abort` should be identical to `async_abort` this has no impact on my workflow. I put this here simply for future low-priority fixes.

tgingold, 2024-06-06:
I think async_abort is handled correctly when used without an explicit clock.
I am investigating.

NikLeberg, 2024-06-06:
With explicit clock it shows the same behaviour for me.

```vhd
  default clock is rising_edge(clk);
  assert always (a = '1')       abort (reset = '1'); -- ok
  assert always (a = '1') async_abort (reset = '1'); -- error
```

closed 2024-06-10 by tgingold

NikLeberg, 2024-06-12:
Wow, this did trigger quite a big change... thank you for the amazing work! The elaboration now works for me.

I'm not sure if this now did change behaviour/simulation though. I have a contrived entity:
```vhd
entity AssertUnderResetTest is
  port(
    reset : in std_logic;
    clk : in std_logic
  );
end AssertUnderResetTest;
```

For which the following architecture did __not__ fail formal verification previous to the fixes:

```vhd
architecture previous of AssertUnderResetTest is
begin
  assert always (false) async_abort (reset = '1') @rising_edge(clk);
  assume always (reset = '1') @rising_edge(clk);
end previous;
```

To get the same behaviour now with the fix I need to change the `async_abort` to a `sync_abort`. Otherwise the formal verification fails in step 0 (`Checking assertions in step 0..`):
```vhd
architecture after_fix of AssertUnderResetTest is
begin
  assert always (false) sync_abort (reset = '1') @rising_edge(clk);
  assume always (reset = '1') @rising_edge(clk);
end after_fix;
```

Previous relates to GHDL version: `4.1.0 (4.1.0.r0.g7188e92cf)`
After fix relates to GHDL version: `5.0.0-dev (4.1.0.r61.gdab33f753)`

I'm not saying that my use-case is at all valid. I'm simply stating that the behaviour changed.

NikLeberg, 2024-06-13:
Alright, makes perfect sense in hindsight. Thanks for the explanation!
So I basically relied on the previous wrong handling of the abort.

commit e9001dad (2024-06-10): Fix handling of async_abort

Add PSL_Abort and remove PSL_Abort_Flag
Previously, async_abort was handled if at the top of the property
(before always), which is incorrect.

Fix #2669