ghdl/ghdl #2675: Attribute of component (e.g. keep) not kept in the synthesis netlist

donnie-j, 2024-06-21:
**Description**
When using the GHDL plugin to yosys, and probably more generally, attributes are not attached to instantiations.  This causes (for example) yosys to optimise away instantiated blocks that are marked keep in the VHDL.

**Expected behaviour**
Should be able to attach attributes to instantiations that need to be preserved.  This should be passed to yosys (if using the plugin) or for example passed as (* keep = '1' *) in the verilog synthesis netlist.

**How to reproduce?**
The below should result in `bar` having the attribute (* keep = "yes" *) in the resulting verilog netlist.

```vhd :file: atr.vhd
-- a simple synthesis test for passing attributes
library ieee;
use ieee.std_logic_1164.all;

entity foo is port (
  a : in  std_logic;
  y : out std_logic);
end entity;

architecture beh of foo is

component bar is
end component;
attribute keep : string;
attribute keep of bar: component is "yes";

begin
   y <= a;

   b0 : bar;
end beh;
```

```sh
ghdl -a atr.vhd ; yosys -m ghdl -p "ghdl foo; write_verilog foo.v"
```

The resulting netlist contains the bar() module, but it is not labeled keep:

```verilog:file:foo.v
/* Generated by Yosys 0.41 (git sha1 c1ad37779, g++ 13.2.0-23ubuntu4 -fPIC -Os) */

module foo(a, y);
  input a;
  wire a;
  output y;
  wire y;
  bar b0 (
  );
  assign y = a;
endmodule
```

**Context**
In this case, `bar()` may be a macro that needs to be instantiated, but has no ports or connections on the final die.

Any yosys optimization pass will remove `bar()`.  The same may happen in other cases, with (otherwise) redundant logic instantiations.

tgingold, 2024-06-22:
Ok, I have to check whether it is supported or not.
But I think the `keep` attribute should be set on the instance, not on the entity unless other tools accept it.

marzoul, 2024-07-02:
In a similar direction,
I have just realized the attribute `ram_style` is also removed, however it is important for yosys logic synthesis.
Maybe a simple and general behaviour of GHDL could be to just pass all non-ghdl-specific attributes to backend ?

donnie-j, 2024-07-02:
I'm a fan of the simple pass though approach.  We tend to attach attributes to signals etc (yes, this issue is about instantiations, still) that are post processed by other tools, e.g. using [this](https://github.com/j-core/jcore-soc/blob/master/lib/hwutils/attr_pkg.vhd)

tgingold, 2024-07-02:
@marzoul but this is the default behaviour: ghdl keeps all the attribute attached to signals and variables.
The problem is that some attributes are attached to instances.
For `ram_style`, can you create a separate issue with a reproducer ?

marzoul, 2024-09-03:
@tgingold here is a reproducer with a simple memory component :

```vhd :file: top.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity top is
	generic (
		WDATA : natural := 32;
		CELLS : natural := 64;
		WADDR : natural := 6
	);
	port (
		clk    : in  std_logic;
		wren   : in  std_logic;
		wraddr : in  std_logic_vector(WADDR-1 downto 0);
		wrdata : in  std_logic_vector(WDATA-1 downto 0);
		rdaddr : in  std_logic_vector(WADDR-1 downto 0);
		rddata : out std_logic_vector(WDATA-1 downto 0)
	);
end top;

architecture synth of top is

	type mem_type is array(0 to CELLS-1) of std_logic_vector(WDATA-1 downto 0);
	signal mem : mem_type;

	attribute ram_style : string;
	attribute ram_style of mem : signal is "distributed";

begin

	process(clk)
	begin
		if rising_edge(clk) then
			if wren = '1' then
				mem(to_integer(unsigned(wraddr))) <= wrdata;
			end if;
		end if;
	end process;

	rddata <= mem(to_integer(unsigned(rdaddr)));

end architecture;
```
Do the synthesis with :
```sh
ghdl -a top.vhd
ghdl synth --out=verilog top
```
The output verilog is the following, it contains no attribute :
```verilog
module top
  (input  clk,
   input  wren,
   input  [5:0] wraddr,
   input  [31:0] wrdata,
   input  [5:0] rdaddr,
   output [31:0] rddata);
  wire [31:0] n19_data; // mem_rd
  assign rddata = n19_data;
  /* top.vhd:41:23  */
  reg [31:0] mem[63:0] ; // memory
  assign n19_data = mem[rdaddr];
  always @(posedge clk)
    if (wren)
      mem[wraddr] <= wrdata;
  /* top.vhd:41:23  */
  /* top.vhd:36:37  */
endmodule
```
Additional note for the VHDL export : if we ask for a vhdl output then the attribute is also lost, except there is a comment to indicate an attribute was placed on the signal, but in addition that sign no longer exists.

So, yes there is an issue with propagation of attributes.

marzoul, 2024-09-06:
Sorry you asked for a separate issue, I'll do it.

marzoul, 2024-09-06:
@tgingold separate issue created in #2747

EverythingElseWasAlreadyTaken, 2025-05-15:
This issue seems to be fixed! 

> [@tgingold](https://github.com/tgingold) here is a reproducer with a simple memory component :
> 
> library ieee;
> use ieee.std_logic_1164.all;
> use ieee.numeric_std.all;
> 
> entity top is
> 	generic (
> 		WDATA : natural := 32;
> 		CELLS : natural := 64;
> 		WADDR : natural := 6
> 	);
> 	port (
> 		clk    : in  std_logic;
> 		wren   : in  std_logic;
> 		wraddr : in  std_logic_vector(WADDR-1 downto 0);
> 		wrdata : in  std_logic_vector(WDATA-1 downto 0);
> 		rdaddr : in  std_logic_vector(WADDR-1 downto 0);
> 		rddata : out std_logic_vector(WDATA-1 downto 0)
> 	);
> end top;
> 
> architecture synth of top is
> 
> 	type mem_type is array(0 to CELLS-1) of std_logic_vector(WDATA-1 downto 0);
> 	signal mem : mem_type;
> 
> 	attribute ram_style : string;
> 	attribute ram_style of mem : signal is "distributed";
> 
> begin
> 
> 	process(clk)
> 	begin
> 		if rising_edge(clk) then
> 			if wren = '1' then
> 				mem(to_integer(unsigned(wraddr))) <= wrdata;
> 			end if;
> 		end if;
> 	end process;
> 
> 	rddata <= mem(to_integer(unsigned(rdaddr)));
> 
> end architecture;
> 
> Do the synthesis with :
> 
> ghdl -a top.vhd
> ghdl synth --out=verilog top
> 
> The output verilog is the following, it contains no attribute :
> 
> module top
>   (input  clk,
>    input  wren,
>    input  [5:0] wraddr,
>    input  [31:0] wrdata,
>    input  [5:0] rdaddr,
>    output [31:0] rddata);
>   wire [31:0] n19_data; // mem_rd
>   assign rddata = n19_data;
>   /* top.vhd:41:23  */
>   reg [31:0] mem[63:0] ; // memory
>   assign n19_data = mem[rdaddr];
>   always @(posedge clk)
>     if (wren)
>       mem[wraddr] <= wrdata;
>   /* top.vhd:41:23  */
>   /* top.vhd:36:37  */
> endmodule
> 
> Additional note for the VHDL export : if we ask for a vhdl output then the attribute is also lost, except there is a comment to indicate an attribute was placed on the signal, but in addition that sign no longer exists.
> 
> So, yes there is an issue with propagation of attributes.

This seems to be fixed. 
I've tested your example with `GHDL 6.0.0-dev (4.1.0.r593.gdd13f1d38) [Dunoon edition] mcode backend`.
GHDL outputs the following result:

```
top.vhdl:27:19:warning: unhandled attribute "ram_style" [-Wunhandled-attribute]
        attribute ram_style of mem : signal is "distributed";
                  ^
top.vhdl:24:16:note: found RAM "mem", width: 32 bits, depth: 64
        signal mem : mem_type;
               ^
module top
  (input  clk,
   input  wren,
   input  [5:0] wraddr,
   input  [31:0] wrdata,
   input  [5:0] rdaddr,
   output [31:0] rddata);
  wire [31:0] n19_data; // mem_rd
  assign rddata = n19_data; //(module output)
  /* top.vhdl:40:23  */
  (* ram_style="distributed" *) reg [31:0] mem[63:0] ; // memory
  assign n19_data = mem[rdaddr];
  always @(posedge clk)
    if (wren)
      mem[wraddr] <= wrdata;
  /* top.vhdl:40:23  */
  /* top.vhdl:35:37  */
endmodule

```

commit 56a4d3e4 (2025-05-17): testsuite/synth: add a test for #2675

Close #2675
closed 2025-05-17 by tgingold