ghdl/ghdl #2677: Subtype from unconstrained record

cameronweston, 2024-06-25:
**Description**
ghdl exits during runtime before reaching simulation stop time or producing a waveform. 

**Expected behaviour**
The example file should produce an output (cnt) that increments every clock cycle. If the output is changed to a std_logic_vector everything works as expected. 

**How to reproduce?**
Minimal Working Example:

```vhd :file: pkg.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package pkg is
  type count_t is record
    count : std_logic_vector;
  end record;
end package;
```

```vhd :file: counter.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.pkg.all;

entity counter is
  port (
    clk     : in std_logic;
    cnt     : out count_t(count(7 downto 0))
  );
end counter;

architecture rtl of counter is
  signal test : cnt.count'subtype := (others => '0');
begin
count : process (clk)
begin
  if rising_edge(clk) then
    test <= std_logic_vector(unsigned(test) + 1);
  end if;
end process;

cnt.count <= test;
end rtl;
```

```vhd :file: tb_counter.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.pkg.all;

entity tb_counter is
end tb_counter;

architecture tb of tb_counter is
  constant period : time := 25 ns;
  signal clk : std_logic := '0';
  signal cnt : count_t(count(7 downto 0));
begin
  clk <= not clk after period / 2;

  uut : entity work.counter
   port map(
      clk => clk,
      cnt => cnt
  );

end tb;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 pkg.vhd counter.vhd tb_counter.vhd
ghdl -e --std=08 tb_counter
ghdl -r --std=08 tb_counter --stop-time=1us --wave=counter.ghw
```

**Context**
Please, provide the following information:

- OS: Ubuntu 22.04 (wsl)
- Origin:
  - [x] Built from sources: `7188e92cf3cc60f`

diogratia, 2024-06-26:
Changing how the subtype indication for test is derived:

```vhdl
architecture rtl of counter is
  -- signal test : cnt.count'subtype := (others => '0');
    signal test:  std_logic_vector (cnt.count'range) := (others => '0');
begin
count : process (clk)
begin
  if rising_edge(clk) then
    test <= std_logic_vector(unsigned(test) + 1);
  end if;
end process;

cnt.count <= test;
end rtl;
````
allows successful simulation.

This also clearly defines the issue as the use of the 'SUBTYPE predefined attribute.

By commenting out the clock relaxation oscillator

```none
%: ghdl -r --std=08 tb_counter --stop-time=1us --wave=tb_counter.ghw
./tb_counter:error: NULL access dereferenced
in process .tb_counter(tb).P0
```
became

```none
%: ghdl -r --std=08 tb_counter --stop-time=1us --wave=tb_counter.ghw
./tb_counter:error: NULL access dereferenced
in process .tb_counter(tb).uut@counter(rtl).P0
./tb_counter:error: simulation failed
```

pointing to the actual issue (using an llvm version (4.1.0.r37.g869f16434) on a Mac.

Getting the subtype from an alias:

```vhdl
architecture rtl of counter is
    alias cnt_count: std_logic_vector is cnt.count;
    signal test:   cnt_count'subtype := (others => '0');  -- works
  -- signal test : cnt.count'subtype := (others => '0');  -- doesn't work
  -- signal test:  std_logic_vector (cnt.count'range) := (others => '0');  -- works
begin 
count : process (clk)
begin
  if rising_edge(clk) then
    test <= std_logic_vector(unsigned(test) + 1);
  end if;
end process;

cnt.count <= test;
end rtl;
```
also works which tells us the 'subtype isn't being extracted from selected named record element.

tgingold, 2024-06-26:
You can use `--fst=tb.fst` to dump waves instead of `--ghw=xx`.
It's an issue with the waveform dumper.

cameronweston, 2024-06-26:
If I change the last line to ```ghdl -r --std=08 tb_counter --stop-time=1us --fst=tb.fst``` then I get the following error:
```
./tb_counter:error: NULL access dereferenced
in process .tb_counter(tb).P0
./tb_counter:error: simulation failed
```

Also I didn't think fst supported VHDL records?

tgingold, 2024-06-26:
I am pretty sure it works if you use the mcode backend.  For llvm or gcc backends, ghw and fst use the same input which aren't complete enough. (I plan to deprecate the gcc backend and to use the llvm backend like the mcode one).

I was not able to get a crash with the mcode backend.

Records are dumped in the FST files.

cameronweston, 2024-06-26:
I was using llvm. It looks like records are only supported with the mcode backend? I was using llvm for better performance but maybe for the time being I'll switch to mcode.