ghdl/ghdl #2678: error: unhandled dyn operation: IIR_PREDEFINED_IEEE_NUMERIC_STD_UNSIGNED_TO_SLV_NAT_NAT

MJoergen, 2024-06-28:
**Description**
I'm using ghdl in combination with yosys to do formal verification. It generally works fine for me, except for the following problem:
During compilation I get the error:
```
error: unhandled dyn operation: IIR_PREDEFINED_IEEE_NUMERIC_STD_UNSIGNED_TO_SLV_NAT_NAT
```
The source code line in question is:
```
user_ram_b_addr <= to_stdlogicvector(column, C_USER_RAM_ADDR_SIZE);
```
where I further up have declared:
```
constant C_USER_RAM_ADDR_SIZE : natural := log2(G_ROW_SIZE);
signal   user_ram_b_addr      : std_logic_vector(C_USER_RAM_ADDR_SIZE - 1 downto 0);
```
and `G_ROW_SIZE` is a generic declared as a natural and given the value 4. I'm including the library numeric_std_unsigned.

Note: GHDL can easily handle this in simulation; the problem only appears during formal verification with yosys, which I believe is using GHDL in synthesis mode.

**Expected behaviour**
I expect formal verification to proceed without error.

I'm wondering whether this feature is not supported, and if so, whether you can suggest a work-around.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `commit 768b6d3ba`

tgingold, 2024-06-28:
I think I have a fix.
But it would be much simpler if you could file a reproducer.  The function is overloaded.

MJoergen, 2024-06-28:
Here is a MWE:

```
library ieee;
   use ieee.std_logic_1164.all;
   use ieee.numeric_std_unsigned.all;

entity ent is
   port (
      clk_i  : in    std_logic;
      col_i  : in    natural;
      addr_o : out   std_logic_vector(1 downto 0)
   );
end entity ent;

architecture synth of ent is

   constant C_ROW_SIZE : natural  := 4;

   pure function log2 (
      arg : natural
   ) return natural is
   begin
      --
      for i in 0 to arg loop
         if 2 ** i >= arg then
            return i;
         end if;
      end loop;

      return -1;
   end function log2;

   constant C_ADDR_SIZE : natural := log2(C_ROW_SIZE);

   signal   addr : std_logic_vector(C_ADDR_SIZE - 1 downto 0);
   signal   col  : natural range 0 to C_ROW_SIZE - 1;

begin

   col    <= col_i;
   addr   <= to_stdlogicvector(col, C_ADDR_SIZE);
   addr_o <= addr(1 downto 0);

end architecture synth;
```

Reproduce with the commands:
```
ghdl -a --std=08 issue2678.vhd
ghdl synth --std=08 ent
```

Error produced:
```
issue2678.vhd:39:14:error: unhandled dyn operation: IIR_PREDEFINED_IEEE_NUMERIC_STD_UNSIGNED_TO_SLV_NAT_NAT
   addr   <= to_stdlogicvector(col, C_ADDR_SIZE);
```

tgingold, 2024-06-28:
Thank you.  I will soon commit a fix for it.
(Otherwise, you can use numeric_std).

commit 4db9b33d (2024-06-28): synth-vhdl_oper: handle to_stdlogicvector

Fix #2678
closed 2024-06-28 by tgingold