ghdl/ghdl #2680: [question]: Adding GHDL to scoop.

Gaweringo, 2024-06-28:
I would like to add GHDL to [scoop](https://scoop.sh). But I want to make sure that it's ok.

GHDL is available for Windows and from what I could figure out, the recommended way of installing would be through MSYS2. (Is this correct?)

I personally like scoop the most as a "package manager" for Windows, and I would like to add GHDL to it, so that it's easy to install and update. I would use the `ghdl-MINGW32.zip` and `ghdl-UCRT64.zip` for 32Bit and 64Bit respectively from the releases page.

The final scoop setup would look like this: https://github.com/Gaweringo/Extras/blob/ghdl/bucket/ghdl.json
You can test this out by installing Scoop. Adding my forked bucket as a scoop bucket:
```
scoop bucket add gaweringo-extras-ghdl https://github.com/Gaweringo/Extras
```
and then installing ghdl with:
```
scoop install ghdl
```

Or just download the `ghdl.json` file and install it with scoop like so: `scoop install .\ghdl.json`

## My questions

I am wandering, if there would be any problems, which I am not aware of, with adding GHDL to scoop for easier installation on Windows?
I have used this method on a clean Windows install and ran `testsuite/vests` and everything seems to work fine.
Are there any things I am missing, because I would not want to add this to scoop if this could mean, that people installing it that way would get a bugged or otherwise incomplete version.

## TL:DR
- Is there a dependency/technological reason why GHDL is not on Scoop, Chocolatey or winget?
- Are `ghdl-MINGW32.zip` and `ghdl-UCRT64.zip` for 32Bit and 64Bit respectively?

Sorry for the long text. Just want to make sure you are okay with this and that I am not missing anything.

tgingold, 2024-06-29:
I am not a Windows user, so I don't know what is the best way to distribute GHDL.
If it is easy to support scoop, go for it!  However that's not something I could maintain.

Apparently, there is no more Ada compiler for Windows32, so `ghdl-MINGW32.zip` is not built anymore.

Gaweringo, 2024-06-30:
Just created the PR to add it to scoop. It's pretty easy to maintain, you don't have to do anything. It automatically looks for new releases on GitHub and updates the package on scoop to just download the new version.

closed 2024-06-30 by Gaweringo

tgingold, 2024-06-30:
Ok, thank you.