ghdl/ghdl #2681: Testsuite does not pass on macos 14.5 (intel) for ghdl-mcode

yikkrubz, 2024-06-30:
**Description**
Whereas the llvm-based version of ghdl passes the testsuite on macos 14.5, the mcode-based version fails on the testcases issue2510 and bug081.

**Expected behaviour**
The testsuite should pass like the llvm-based version of ghdl.

gna issue1377: ok
gna issue2510: failed
gna issue1379: ok
analyze repro1.vhdl
repro1.vhdl:21:19:warning: declaration of a protected object before the protected body [-Welaboration]
  shared variable sh : prot_t;
                  ^
elaborate and simulate (failure expected) repro1
repro1.vhdl:28:31:error: reference to a declaration before its elaboration
    variable val : natural := c;
                              ^

raised STORAGE_ERROR : stack overflow
gna bug081: failed
try to analyze -Werror=elaboration dummy.vhdl
dummy.vhdl:4:55:error: function "STICKS" is called before elaborated of its body
  constant IM_TOTALLY_NOT_NEGATIVE: natural := FISH - STICKS;
                                                      ^
dummy.vhdl:7:14:error: circular reference of package "dummypackage"
package body DummyPackage is
             ^
dummy.vhdl:16:10:error: circular reference of package "dummypackage"
use work.DummyPackage.all;
         ^
dummy.vhdl:20:38:error: no declaration for "im_totally_not_negative"
  signal i_data: in std_logic_vector(IM_TOTALLY_NOT_NEGATIVE-1 downto 0);                                              
                                     ^
dummy.vhdl:21:39:error: no declaration for "im_totally_not_negative"
  signal o_data: out std_logic_vector(IM_TOTALLY_NOT_NEGATIVE-1 downto 0)                                              
                                      ^
dummy.vhdl:25:14:error: circular reference of entity "dummyentity"
architecture arch of DummyEntity is
             ^
dummy.vhdl:27:3:error: no declaration for "o_data"
  o_data <= i_data;
  ^
dummy.vhdl:27:13:error: no declaration for "i_data"
  o_data <= i_data;
            ^
try to analyze -Werror=elaboration elab_func.vhdl
elab_func.vhdl:6:27:error: function "f" is called before elaborated of its body
  constant c : natural := f;
                          ^
try to analyze --std=08 -Werror=elaboration elab_prot.vhdl
elab_prot.vhdl:9:19:error: declaration of a protected object before the protected body
  shared variable v: myprot;
                  ^
analyze -Werror=elaboration test_comp.vhdl
analyze dummy.vhdl
dummy.vhdl:4:55:warning: function "STICKS" is called before elaborated of its body [-Welaboration]
  constant IM_TOTALLY_NOT_NEGATIVE: natural := FISH - STICKS;
                                                      ^
elaborate and simulate (failure expected) dummyentity
dummy.vhdl:4:12:error: value out of range
  constant IM_TOTALLY_NOT_NEGATIVE: natural := FISH - STICKS;
           ^
dummy.vhdl:20:38:error: reference to a declaration before its elaboration
  signal i_data: in std_logic_vector(IM_TOTALLY_NOT_NEGATIVE-1 downto 0);                                              
                                     ^

raised STORAGE_ERROR : stack overflow
gna issue2510: failed
analyze repro1.vhdl
repro1.vhdl:21:19:warning: declaration of a protected object before the protected body [-Welaboration]
  shared variable sh : prot_t;
                  ^
elaborate and simulate (failure expected) repro1
repro1.vhdl:28:31:error: reference to a declaration before its elaboration
    variable val : natural := c;
                              ^

raised STORAGE_ERROR : stack overflow

gna bug081: failed
try to analyze -Werror=elaboration dummy.vhdl
dummy.vhdl:4:55:error: function "STICKS" is called before elaborated of its body
  constant IM_TOTALLY_NOT_NEGATIVE: natural := FISH - STICKS;
                                                      ^
dummy.vhdl:7:14:error: circular reference of package "dummypackage"
package body DummyPackage is
             ^
dummy.vhdl:16:10:error: circular reference of package "dummypackage"
use work.DummyPackage.all;
         ^
dummy.vhdl:20:38:error: no declaration for "im_totally_not_negative"
  signal i_data: in std_logic_vector(IM_TOTALLY_NOT_NEGATIVE-1 downto 0);                                              
                                     ^
dummy.vhdl:21:39:error: no declaration for "im_totally_not_negative"
  signal o_data: out std_logic_vector(IM_TOTALLY_NOT_NEGATIVE-1 downto 0)                                              
                                      ^
dummy.vhdl:25:14:error: circular reference of entity "dummyentity"
architecture arch of DummyEntity is
             ^
dummy.vhdl:27:3:error: no declaration for "o_data"
  o_data <= i_data;
  ^
dummy.vhdl:27:13:error: no declaration for "i_data"
  o_data <= i_data;
            ^
try to analyze -Werror=elaboration elab_func.vhdl
elab_func.vhdl:6:27:error: function "f" is called before elaborated of its body
  constant c : natural := f;
                          ^
try to analyze --std=08 -Werror=elaboration elab_prot.vhdl
elab_prot.vhdl:9:19:error: declaration of a protected object before the protected body
  shared variable v: myprot;
                  ^
analyze -Werror=elaboration test_comp.vhdl
analyze dummy.vhdl
dummy.vhdl:4:55:warning: function "STICKS" is called before elaborated of its body [-Welaboration]
  constant IM_TOTALLY_NOT_NEGATIVE: natural := FISH - STICKS;
                                                      ^
elaborate and simulate (failure expected) dummyentity
dummy.vhdl:4:12:error: value out of range
  constant IM_TOTALLY_NOT_NEGATIVE: natural := FISH - STICKS;
           ^
dummy.vhdl:20:38:error: reference to a declaration before its elaboration
  signal i_data: in std_logic_vector(IM_TOTALLY_NOT_NEGATIVE-1 downto 0);                                              
                                     ^

raised STORAGE_ERROR : stack overflow

gna test failed (issue2510 bug081 )

- OS: macsos 14.5
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `4.1.0.r90.g623407590`

tgingold, 2024-06-30:
Probably you should increase the stack limit using `ulimit`.

yikkrubz, 2024-06-30:
Thank you very much for your reply. ulimit -s 16394 fixed the issue (was 8192).

closed 2024-06-30 by yikkrubz