ghdl/ghdl #2683: slow analyze of obscure PSL sequence

NikLeberg, 2024-07-03:
**Description**
Hey there, It's me with the obscure bug reports again :) I want to stress, that I have absolutely no requirement for the following to work. I just noticed this by coincidence and wanted to let you know. I find it very interesting. Feel no obligation to fix this.

When analyzing a specific PSL sequence, the analize step takes a really long time (more than a minute).

**Expected behaviour**
I expected the analize step to be.. faster.

**How to reproduce?**
Minimal Working Example (MWE):

```vhd :file: foo_ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity foo is
  port(
    a, b : in std_logic;
    clk  : in std_logic
  );
end foo;
```

```vhd :file: foo_arch-slow.vhd
architecture slow of foo is
  default clock is rising_edge(clk);
begin
  sequence s0 is {{a ; b[=2] ; a}[+] ; b ; a}; -- note "[=2]"
  sequence s1 is {s0 ; s0};
  assert (always s1);
end slow;
```

```vhd :file: foo_arch-fast.vhd
architecture fast of foo is
  default clock is rising_edge(clk);
begin
  sequence s0 is {{a ; b[=1] ; a}[+] ; b ; a}; -- note "[=1]"
  sequence s1 is {s0 ; s0};
  assert (always s1);
end fast;
```

```sh :image: ghdl/ghdl:bullseye-mcode
timeout 10 ghdl -a --std=08 foo_ent.vhd foo_arch-fast.vhd # ok
timeout 10 ghdl -a --std=08 foo_ent.vhd foo_arch-slow.vhd # timeout
```

**Context**
- OS: Ubuntu 24.04 (inside docker in WSL2 Windows Host)
- Origin:
  - [x] Built from sources: `3b05281` (Jun 26, 2024)

**Additional context**
 - The PSL sequence is meant to be used to formaly prove with SymbiYosys i.e. Yosys.
 - When analyzing the _slow_ sequence, CPU usage of GHDL jumps to 100% (of one core), memory consumption stays the same.
 - It eventually finishes processing after some time (more than three hours). Yosys reports an imense amount of logic gates (32k) in the final netlist.
 - I'm not sure what eactly in the sequence makes it so slow. Removing any part/step of the sequence makes it fast again.
 - Is it just a very bad sequence?

tgingold, 2024-07-03:
The sequence is long but not very long.  This should be fixable.

closed 2024-07-05 by tgingold

NikLeberg, 2024-07-06:
Nice! _400 ms_ compared to more than _3 h_ is an _20k_ fold improvement. Amazing. :P

commit 3652e9f0 (2024-07-05): psl/psl-optimize: fix an nfa corruption

Fix #2683