ghdl/ghdl #2685: SIM: Segmentation fault when running GHDL generated binary (LLVM)

tmeissner, 2024-07-08:
Hi. I'm in the process to adapt simulations from Questa to free simulators. It's already done for **nvc**, so I'm at adapting to **GHDL** now. I have two problems , here is the first one. Note, that the designs and testbenches are complex and confidential. So unfortunately I can't present an MWE at the moment.

I get an segmentatation fault when trying to run the GHDL (LLVM backend) generated binary:

```
$ ./queuehandlingt 
Segmentation fault
```

In gdb I get the following:

```
Reading symbols from queuehandlingt...
(gdb) run
Starting program: /srv/scratch/meissner/git/QueueHandling/queuehandlingt 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x0000000055000160 in ?? ()
```

Maybe someone has a hint what I can do with gdb (I'm not a user of this tool, so I don't have any clue what it is capable of) or what else I could do.

GHDL version:
```
$ ghdl --version
GHDL 5.0.0-dev (4.1.0.r102.g3b5854759.dirty) [Dunoon edition]
 Compiled with GNAT Version: 10.5.0
 llvm 14.0.0 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Linux:
```
$ uname -a
Linux hw-ts3 6.1.0-18-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.76-1 (2024-02-01) x86_64 GNU/Linux
```

tgingold, 2024-07-09:
Did you try with the mcode backend ?

tgingold, 2024-07-09:
Otherwise, in gdb try `bt`.

tmeissner, 2024-07-09:
Okay, I tried the mcode backend.

```
$ ghdl --version
GHDL 5.0.0-dev (4.1.0.r110.g918a63daf) [Dunoon edition]
 Compiled with GNAT Version: 10.5.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.
```

Here is the bug message I get:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r110.g918a63daf) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
/build/git/QueueHandling/work/
Command line:
ghdl -r --std=08 -P../osvvm -P../simulation queuehandlingt --wave=queuehandlingt.ghw --psl-report=queuehandling_psl_coverage.json
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-avhpi.adb:593 access check failed
******************************************************************
```

The error only occurs when the `--wave` option is used. Without the option the simulation runs with mcode. Note, that with LLVM backend it doesn't matter if `--wave` option is used or not. One always get a segmentation fault with LLVM backend, regardless of the `--wave` option.

tgingold, 2024-07-09:
Try to use `--fst=queuehandlingt.fst` instead of `--wave`.  The ghw dumper doesn't support all kinds of instances.