ghdl/ghdl #2686: GHDL bug when using generic package with string type

tmeissner, 2024-07-08:
**Description**
I have a generic package with a generic type which uses these type inside a record type. When I use the package with the generic type set to string a GHDL bug occurs when analyzing the design which instantiate the package.

**Expected behaviour**
**nvc** and **Questa** can handle the code without problems.

**How to reproduce?**

```vhd :file: test.vhd
package testp is
  generic (type G_KEY_TYPE);
  type t_dict_iter is record
    key : G_KEY_TYPE;
  end record t_dict_iter;
end package testp;

package body testp is
end package body testp;

entity test is
end entity test;

architecture sim of test is
  -- bug
  package StringPackage is new work.testp
    generic map (G_KEY_TYPE => string);
  -- works
  package NaturalPackage is new work.testp
    generic map (G_KEY_TYPE => natural);
begin
end architecture sim;
```

```sh
$ ghdl -a --std=08 --work=work test.vhd 
```

**Context**
```
$ ghdl --version
GHDL 5.0.0-dev (4.1.0.r102.g3b5854759.dirty) [Dunoon edition]
 Compiled with GNAT Version: 10.5.0
 llvm 14.0.0 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
```
$ uname -a
Linux hw-ts3 6.1.0-18-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.76-1 (2024-02-01) x86_64 GNU/Linux
```

If a `GHDL Bug occurred` block is shown in the log, please paste it here:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r102.g3b5854759.dirty) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
/build/tests/ghdl_bug/
Command line:
/opt/oss-cad-suite/libexec/ghdl1-llvm --std=08 --work=work -P/opt/oss-cad-suite/lib/ghdl/ieee/v08/ -P/opt/oss-cad-suite/lib/ghdl/ -fpic -c -o test.o test.vhd
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap3.adb:645 access check failed
Call stack traceback locations:
0x7f31b55e975c 0x7f31b55e99c5 0x7f31b55eac0c 0x7f31b55ee30f 0x7f31b55f3b0a 0x7f31b565ff64 0x7f31b56613ec 0x7f31b5663708 0x7f31b564ceac 0x7f31b5654569 0x7f31b5661519 0x7f31b5663708 0x7f
31b568d90f 0x7f31b5644224 0x7f31b5636672 0x7f31b56a8797 0x7f31b56aa1e9 0x7f31b56ab83d 0x7f31b52580b1 0x7f31ae0a4d8e 0x7f31ae0a4e3e 0x7f31b52568e3 0xfffffffffffffffe
******************************************************************
/opt/oss-cad-suite/libexec/ghdl:error: compilation error
```

**Additional context**
Note that the error doesn't occur when using another type like natural or boolean.

smaslovski, 2024-07-09:
Hi,

Here, this code passes analysis & elaboration, but fails when running:

```
$ ghdl -a --std=08 ghdl_bug.vhd
$ ghdl -e --std=08 test        
$ ghdl -r --std=08 test        

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (3.0.0.r1094.gbddf22c98) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-linux-gnu
/home/stas/vhdl/bugs/
Command line:
/home/stas/app/ghdl/bin/ghdl -r --std=08 test
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : ortho_code-types.adb:113
******************************************************************
```

Context:

```
$ ghdl --version
GHDL 5.0.0-dev (3.0.0.r1094.gbddf22c98) [Dunoon edition]
 Compiled with GNAT Version: 12.2.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

smaslovski, 2024-07-09:
Well, probably, the context from above is not very helpfull as is, because it refers to my local git repo... But it is in sync with the current GHDL's HEAD:

```commit bddf22c981bf33fe8c5c708afb164c209548e928 (HEAD -> master)
Merge: 66db3e3b0 918a63daf
Author: Stanislav Maslovski <stanislav.maslovski@gmail.com>
Date:   Tue Jul 9 12:12:16 2024 +0100

    Merge branch 'master' of https://github.com/ghdl/ghdl

commit 918a63dafb23cf3dbf596716783aaf2670cfc98c (upstream/master)
Author: Tristan Gingold <tgingold@free.fr>
Date:   Sat Jul 6 20:13:45 2024 +0200

    testsuite/gna: reproducer for #2684

commit f57fe0a12f23a4185f114caf0f21976b8766863a
Author: Tristan Gingold <tgingold@free.fr>
Date:   Sat Jul 6 20:12:33 2024 +0200

    vhdl-sem_specs: handle psl declarations
    
    Fix #2684 (by NikLeberg)

commit 1183b2f7b488baa39b4d5408d748e9e166c57255
Author: Tristan Gingold <tgingold@free.fr>
Date:   Sat Jul 6 18:47:19 2024 +0200

    ci-run: use static llvm libs on macosx

commit 87a85759c79eb58e4d46a5c1330ee13544d07d02
Author: Tristan Gingold <tgingold@free.fr>
Date:   Sat Jul 6 18:42:58 2024 +0200

    config,Makefile: introduce LLVM_LDFLAGS
    
    To add workarounds llvm-config output
```

commit 588c76b2 (2024-07-09): vhdl-sem_types: reanalyze after unbounded interface type

Fix #2686
closed 2024-07-09 by tgingold