ghdl/ghdl #2689: Crash on   simple entity with function generic.

sckoarn, 2024-07-10:
**Description**
GHDL crashes when running the post analyzed code.

**Expected behaviour**
Test should finish.

**How to reproduce?**
Run the example provided.

```library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
--  package defining functions
package generic_type_pkg is
    function increment(x: integer) return integer;
    function increment1(x: integer) return integer;
end package;

package body generic_type_pkg is
    function increment(x: integer) return integer is
    begin
        return x + 1;
    end function increment;
    
    function increment1(x: integer) return integer is
    begin
        return x + 1;
    end function increment1;
end;

--  entity with function generic
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.generic_type_pkg.all;

entity generic_type is
    generic(
        function ifunc(x: integer) return integer
    );
    port(
        clk         : in  std_logic;
        din         : in  integer;
        dout        : out integer
    );
end;

architecture rtl of generic_type is

begin

    process(clk)
    begin
        if rising_edge(clk) then
            --report integer'image(din);
            dout <= increment(din);
            
        end if;
    end process;
end;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.generic_type_pkg.all;

--  Test bench  instance and driver / checker
entity tb is
end entity;

architecture bhv of tb is
  signal clk : std_logic := '0';
  signal i   : integer := 0;
  signal o   : integer := 0;
  
begin
  -- design  instance
  i_generic_type: entity work.generic_type(rtl)
  generic map(
    ifunc  => increment1
  )
  port map(
    clk => clk,
    din => i,
    dout => o
  );

  --  counter and driver
  process
    variable cnt : integer := 0;
  begin
    wait for 1 ns;
    clk <= not clk;
    cnt := cnt +1;
    if cnt > 200 then
      report "Test Passed !!";
      wait;
    end if;
    i <= increment1(cnt);
  end process;
  -- checker
  process (i)
    variable  p : integer := 0;
    variable  c : integer;
  begin
    if clk'event and clk = '1' then
      c := i;
      assert c = p+2 report "ERROR:  output did not increment." severity failure;
      --report "Check passed ...";
      p := c;
    end if;
  end process;
end bhv;

```

``` ghdl -a --std=08 test.vhdl
ghdl -r --std=08 tb
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:  Deb  Unix
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ x] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r101.g420a1d3e) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
vh_tests/ent_3_08/gen_func1/
Command line:
ghdl -r --std=08 tb
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x55954ada7fc0 0x55954af9c6e9 0x55954af9d969 0x55954af9ed15 0x55954afcb6d1 0x55954af334cf 0x55954af30803 0x55954af3ed6e 0x55954afd0c5f 0x55954af0f0e4 0x55954ae46e60 0x55954afd45bf 0x55954abfba71 0x7f3c4e936099 0x55954abfa1d8 0xfffffffffffffffe
******************************************************************

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

maehne, 2024-07-10:
Your GHDL version is too old. The provided test case passes, e.g., with GHDL 5.0.0-dev (4.1.0.r24.gdc17919eb) [Dunoon edition]. Please consider using one of the [official builds](https://github.com/ghdl/ghdl/releases) in case your Linux distribution provides only an outdated version.

sckoarn, 2024-07-10:
Confirmed.  Sorry for the distraction ...

closed 2024-07-10 by sckoarn