ghdl/ghdl #2692: Exception when synthesising simple vhdl module

andrewrussellmalcolm, 2024-07-19:
**Description**
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:403

**Expected behaviour**
The following vhdl sythesises correctly using Lattice iCECube2, producing expected behaviour when programmed into an iCE40HX
It also works with my test simulation in GHDL

**How to reproduce?**

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
 
entity spi_slave is
  port (
    spi_clk     : in  std_logic;
    spi_in      : in  std_logic;
    spi_out     : out  std_logic;
    spi_en      : in std_logic;
    reg0_out    : out std_logic_vector(7 downto 0);
    reg0_in     : in std_logic_vector(7 downto 0)		
    );
end spi_slave;
 
architecture rtl of spi_slave is

    signal reg_addr   : std_logic_vector(7 downto 0);
    signal wr         : std_logic;
    signal reg0_data : std_logic_vector(7 downto 0) := "00000000"; 

begin
    process(spi_clk)

    variable bit_cnt : integer;

    begin
        if spi_en = '0' then	
            wr <= reg_addr(7);

            if rising_edge(spi_clk) then

                if bit_cnt < reg_addr'length then
                    reg_addr((reg_addr'length -1) - bit_cnt) <= spi_in; 
                    spi_out <= '0';
                else
                    if wr = '1' then
                        reg0_data(15 - bit_cnt) <= spi_in;
                    end if;
                end if;

                bit_cnt := bit_cnt + 1;
            end if;

            if falling_edge(spi_clk) then

                if bit_cnt > reg_addr'length -1 then

                    if wr = '0' then
                        spi_out <= reg0_in(15 - bit_cnt);
                    end if;
                end if;

            end if;

        else

            bit_cnt := 0;
            spi_out <= '0';
            reg_addr <= "00000000";
            wr <= '0';
        end if;
    end process;

    reg0_out <= reg0_data;
end rtl;

ghdl --version
GHDL 5.0.0-dev (4.1.0.r119.gabe642a3c) [Dunoon edition]
 Compiled with GNAT Version: 10.5.0

ghdl -a spi_slave.vhdl
ghdl -e spi_slave
ghdl --synth --out=verilog spi_slave > spi_slave.v

- OS: Lunux Mint
 uname -a
Linux andrews-Precision-3440 5.15.0-113-generic #123-Ubuntu SMP Mon Jun 10 08:16:17 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

- Origin:
  Built from sources

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r119.gabe642a3c) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
/home/andrew/vhdl/spi_slave/
Command line:
ghdl --synth --out=verilog spi_slave
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:403
******************************************************************

tgingold, 2024-07-19:
I can reproduce the crash.

andrewrussellmalcolm, 2024-07-20:
I have been trying to find a way to avid this bug,  and find that an extra
'if' clause for the case when the enable is false avoids the bug:

architecture rtl of spi_slave is

    signal reg_addr   : std_logic_vector(7 downto 0) ;
    signal wr         : std_logic;
    signal reg0_data  : std_logic_vector(7 downto 0) := "00000000";

begin
    process(spi_clk)

    variable bit_cnt : integer := 0;

    begin
        if spi_en = '0' then
            wr <= reg_addr(7);

            if rising_edge(spi_clk) then

if bit_cnt < reg_addr'length then
                    reg_addr((reg_addr'length -1) - bit_cnt) <= spi_in;
spi_out <= '0';
                else
                    if wr = '1' then
reg0_data(15 - bit_cnt) <= spi_in;
end if;
end if;

bit_cnt := bit_cnt + 1;
            end if;

            if falling_edge(spi_clk) then
            if bit_cnt > reg_addr'length -1 then

                    if wr = '0' then
                        spi_out <= reg0_in(15 - bit_cnt);
                    end if;

end if;
            end if;

        else

           if bit_cnt > reg_addr'length -1 then -- AVOIDS BUG
       bit_cnt := 0;
       spi_out <= '0';
       reg_addr <= "00000000";
       wr <= '0';
end if;
        end if;
    end process;

reg0_out <= reg0_data;

end rtl;

I hope this is useful.

Regards

Andrew

tgingold, 2024-07-21:
The initial design is not synthesizable.
I am fixing the crash, but it will be rejected.

andrewrussellmalcolm, 2024-07-21:
Yes, I have figured that out! Still, it would be nice to get an informative
error message and not a crash. Thanks and regards, Andrew

commit 4d3a9f70 (2024-07-22): netlists-inference: avoid a crash on incorrect FF

Fix #2692
closed 2024-07-22 by tgingold