ghdl/ghdl #2695: GHDL's run options are not documented in "ghdl help" output

Paebbels, 2024-07-22:
**Description**
When checking GHDLs command line outputs, especially the outputs of
* `help`
* `help-options`
* `help-warnings`

it was found, that `ghdl run ... [RUNOPTS]` references run options, but no (documented?) command exists to print these run options. Also `ghdl help run` doesn't print them or references an additional command to get them.

**Expected behavior**

All major options should be self-describing/self-documenting by calling `ghdl help...` commands in a shell. The documented run options should include e.g. `-gFOO=bar`, `--wave=file.ghw`, `--read-wave-opt=wave.opt` or `--stop-time=10ms`.

I would suggest:
* `help` should list how to get the list of `RUNOPTS`
* in parity to `help-options` another `help-run-options` is needed to list these options
* I would be happy if `ghdl help run` and `ghdl help elab-run` would display these run options as additional sections.

As a general question:  
Why are there additional commands to print options. I can't remember any program that has split the help over multiple commands. I only saw more complex CLIs where options are listed specifically for certain main commands like `help analyze` or `help run` to have a more specific output for the user what options are allowed in a certain context depending on the main command.

**How to reproduce?**

```
$ ghdl help
usage: C:\msys64\ucrt64\bin\ghdl.exe COMMAND [OPTIONS] ...
COMMAND is one of:
analyze [OPTS] FILEs
  Analyze one or multiple VHDL files
  aliases: -a, analyse
elaborate [OPTS] UNIT [ARCH]
  Elaborate design UNIT
  alias: -e
run UNIT [ARCH] [RUNOPTS]
  Run design UNIT
  alias: -r
elab-run [OPTS] UNIT [ARCH] [RUNOPTS]
  Elaborate and run design UNIT
  alias: --elab-run
bind [OPTS] UNIT [ARCH]
  Bind design UNIT
  alias: --bind
link [OPTS] UNIT [ARCH]
  Link design UNIT
  alias: --link
list-link [OPTS] UNIT [ARCH]
  List objects file to link UNIT
  alias: --list-link
compile [OPTS] FILEs -e UNIT [ARCH]
  Generate whole sequence to elaborate design UNIT from FILEs
  alias: -c
make [OPTS] UNIT [ARCH]
  Make design UNIT
  alias: -m
gen-makefile [OPTS] UNIT [ARCH]
  Generate a Makefile for design UNIT
  alias: --gen-makefile
gen-depends [OPTS] UNIT [ARCH]
  Generate dependencies of design UNIT
  alias: --gen-depends
disp-config
  Display tools path
  aliases: --disp-config, dispconfig, --dispconfig
bootstrap-std
  (internal) Compile std.standard
  alias: --bootstrap-standard
synth [FILES... -e] UNIT [ARCH]
  Synthesis from UNIT
  alias: --synth
import [OPTS] FILEs
  Import units of FILEs
  alias: -i
syntax [OPTS] FILEs
  Check syntax of FILEs
  alias: -s
dir [LIBs]
  Display contents of the libraries
  alias: --dir
files FILEs
  Display units in FILES
  alias: -f
clean
  Remove generated files
  alias: --clean
remove
  Remove generated files and library file
  alias: --remove
disp-standard
  Disp std.standard in pseudo-vhdl
  alias: --disp-standard
elab-order [--libraries] [OPTS] UNIT [ARCH]
  Display ordered source files
  alias: --elab-order
find-top
  Display possible top entity in work library
  alias: --find-top
chop [OPTS] FILEs
  Chop FILEs
  alias: --chop
lines FILEs
  Precede line with its number
  alias: --lines
reprint [OPTS] FILEs
  Redisplay FILEs
  alias: --reprint
fmt [OPTS] FILEs
  Format FILEs
  alias: --format
compare-tokens [OPTS] REF FILEs
  Compare FILEs with REF
  alias: --compare-tokens
pp-html FILEs
  Pretty-print FILEs in HTML
  alias: --pp-html
xref-html FILEs
  Display FILEs in HTML with xrefs
  alias: --xref-html
xref FILEs
  Generate xrefs
  alias: --xref
--vpi-compile CMD ARGS
  Compile with VPI/VHPI include path
--vpi-link CMD ARGS
  Link with VPI/VHPI library
--vpi-cflags
  Display VPI/VHPI compile flags
--vpi-ldflags
  Display VPI/VHPI link flags
--vpi-include-dir
  Display VPI/VHPI include directory
--vpi-library-dir
  Display VPI/VHPI library directory
--vpi-library-dir-unix
  Display VPI/VHPI library directory (unix form)
file-to-xml FILEs
  Dump AST in XML
  alias: --file-to-xml
--libghdl-name
  Display libghdl name
--libghdl-library-path
  Display libghdl library path
--libghdl-include-dir
  Display libghdl include directory
help [CMD]
  Display this help or [help on CMD]
  aliases: -h, --help
version
  Display ghdl version
  aliases: -v, --version
help-options
  Display help for analyzer options
  alias: --help-options, opts-help, --options-help
help-warnings
  Display help about all the warnings
  alias: --help-warnings

To display the options of a GHDL program,
  run your program with the 'help' option.
Also see 'opts-help' for analyzer options.

Please, refer to the GHDL manual for more information.
Report issues on https://github.com/ghdl/ghdl
```

```
$ ghdl help-options
Main options:
  --work=LIB         use LIB as work library
  --workdir=DIR      use DIR for the file library
  -PPATH             add PATH in the library path list
  --std=87/93/00/02/08  select vhdl 87/93/00/02/08 standard
  --std=93c          select vhdl 93 standard and allow 87 syntax
  --[no-]vital-checks  do [not] check VITAL restrictions
  -Wx or --warn-x    enable a warning (see help-warnings
  -Wall              enables all warnings.
  -Werror            turns warnings into errors
Extensions:
  -fexplicit         give priority to explicitly declared operator
  -frelaxed-rules    relax some LRM rules
  -C  --mb-comments  allow multi-bytes chars in a comment
  --bootstrap        allow --work=std
  --syn-binding      use synthesis default binding rule
  -fpsl              parse psl in comments
Compilation list:
  -l[sca]            after semantics, canon or annotation
  --lall             -lX options apply to all files
  -lv                verbose list
  -v                 disp compilation stages
Compilation dump:
  -d[psa]            dump tree after parse, semantics or annotate
  --dall             -dX options apply to all files
```

**Context**
Please, provide the following information:

- OS: Windows 11 + MSYS2 + UCRT64
- Origin:
  - [x] Package manager: `4.1.0`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

**Additional context**
*None*

tgingold, 2024-07-23:
There is `ghdl --run-help`

Paebbels, 2024-07-23:
But `--run-help` is not listed yet, right?

Do you want to list it e.g. below of `run`/`elab-run` like:
```
run UNIT [ARCH] [RUNOPTS]
  Run design UNIT
  alias: -r
  see also: --run-help for RUNOPTS
elab-run [OPTS] UNIT [ARCH] [RUNOPTS]
  Elaborate and run design UNIT
  alias: --elab-run
  see also: --run-help for RUNOPTS

[...]

help-run
  Display help for run options
  alias: --help-run, --run-ehlp
```

tgingold, 2024-07-23:
Yes, for sure this could be added.

closed 2024-07-24 by tgingold

Paebbels, 2024-07-27:
Is this limited to the mcode backend?

commit c04f8d14 (2024-07-24): ghdldrv: put --help-run help at the end

with help-options and help-warnings
Fix #2695