ghdl/ghdl #2696: Isssue with generic package instantiation with type from package

amb5l, 2024-07-23:
I get a compilation error when instantiating a generic package with a type defined in another package. This seems similar to issue #535.

To reproduce, compile the following:
```
package gen_pkg is
  generic (
    type item_t
  );
end package gen_pkg;

library ieee;
  use ieee.std_logic_1164.all;

package type_pkg is
  subtype r_item_t is std_ulogic_vector(15 downto 0);
  constant R_EMPTY : r_item_t := "XXXXXXXXXXXXXXXX";
end package type_pkg;

use work.type_pkg.all;
package r_gen_pkg is
  new work.gen_pkg generic map(item_t => r_item_t);
```

Here is my compilation command with its results:
```
ghdl -a --work=work --std=08 -fsynopsys -frelaxed -Wno-hide -Wno-shared bug.vhd
bug.vhd:21:42: cannot find a "=" declaration for type "r_item_t"
  new work.gen_pkg generic map(item_t => r_item_t);
```
I am running the following version of GHDL on Windows 11 / MSYS2:
`GHDL 4.0.0-dev (3.0.0.r147.g6c56631a7) [Dunoon edition]`

AdamKeith1, 2024-07-23:
Hey sorry I don't know the answer to your question but I was curious how you got the 'work' library to work, are you using symbiyosys?

amb5l, 2024-07-23:
> Hey sorry I don't know the answer to your question but I was curious how you got the 'work' library to work, are you using symbiyosys?

I use makefile driven flow for my FPGA projects; I synthesise and implement with vendor tools (e.g. Vivado) and simulate using Vivado, GHDL, NVC and ModelSim. The user lists a few key details in specific makefile variables, and a makefile include builds rules and recipes. Take a look at the include file for GHDL [here](https://github.com/amb5l/make-fpga/blob/main/ghdl.mak) and an example makefile [here](https://github.com/amb5l/tyto2/blob/main/test/hram/tb_model_hram/makefile).

The user can append '=library' to each source file, with `work` being the default.

AdamKeith1, 2024-07-23:
Thanks for those, I'll continue poking around in them to find what I need. In the meantime about the =library command, here is where I have my commands 
[script]
```
read -formal tb_fpga_config.sv
ghdl --std=08 --ieee=synopsys fpga_config.vhd -e fpga_config
ghdl --std=08 --ieee=synopsys -i axi_pkg.vhd
prep -top top -flatten
```
Is this =library something that needs to go in these ghdl commands and if so what is the exact syntax for that?

tgingold, 2024-07-23:
I think the error message is correct.  You need to provide visibility to the `=` operator, like this:

```vhdl
library ieee;
  use ieee.std_logic_1164.all;
use work.type_pkg.all;
package r_gen_pkg is
  new work.gen_pkg generic map(item_t => r_item_t);
```

AdamKeith1, 2024-07-23:
What is the `get_src_lib` function in your Makefiles? @amb5l

amb5l, 2024-07-23:
> I think the error message is correct. You need to provide visibility to the `=` operator, like this:
> 
> ```vhdl
> library ieee;
>   use ieee.std_logic_1164.all;
> use work.type_pkg.all;
> package r_gen_pkg is
>   new work.gen_pkg generic map(item_t => r_item_t);
> ```

Thanks Tristan, as always you are right!

amb5l, 2024-07-23:
> What is the `get_src_lib` function in your Makefiles? @amb5l

Hi - happy to help - let's discuss this away from this issue. Use my email address (from my repo).

amb5l, 2024-07-23:
No fault found!

closed 2024-07-23 by amb5l

tgingold, 2024-07-23:
You're welcome. (If it were true...)

nickg, 2024-07-24:
> I think the error message is correct. You need to provide visibility to the `=` operator, like this:
> 

I'm not sure about this. The 2008 LRM says:

> The following operations are defined for the interface type:
> — The basic operations of assignment, allocation, type qualification and type conversion
> — The predefined equality (=) and inequality (/=) operators, implicitly declared as formal generic subprograms immediately following the interface type declaration in the enclosing interface list

This doesn't seem to require that the `"="` and `"/="` operators are actually visible at this point, and it also only associates the predefined operators, not any user-defined operator that might be visible.

I believe this should be an error in 2019 however, where the LRM text was changed to:

> Each interface type definition includes the additional predefined operators and attributes that are defined for corresponding incomplete type definition (see 5.8). These additional operators are implicitly declared as formal generic subprograms with an interface subprogram default in form of a box (<>) immediately following the interface type declaration in the enclosing interface list.

Here's an example that has different behaviour in 2008/2019:

```vhdl
package pack1 is
    type t is (a, b, c);
    function "=" (l, r : t) return boolean;
end package;

package body pack1 is
    function "=" (l, r : t) return boolean is
    begin
        report "user-defined = called";
        return false;
    end function;
end package body;

-------------------------------------------------------------------------------

package pack2 is
    generic (type g);
    function equal (l, r : g) return boolean;
end package;

package body pack2 is
    function equal (l, r : g) return boolean is
    begin
        return l = r;
    end function;
end package body;

-------------------------------------------------------------------------------

entity gentype6 is
end entity;

use work.pack1.all;

architecture test of gentype6 is
    package pack3 is new work.pack2 generic map (g => t);
begin

    check: process is
    begin
        assert pack3.equal(a, a);
        wait;
    end process;

end architecture;
```

The assertion should pass in 2008 but fail in 2019. However it fails with GHDL `--std=08` (ModelSim passes in 2008 mode).

tgingold, 2024-07-24:
> I'm not sure about this. The 2008 LRM says:

That specifies only how the interface type can be used.  But this says nothing about how the operators are associated.

> This doesn't seem to require that the `"="` and `"/="` operators are actually visible at this point, and it also only associates the predefined operators, not any user-defined operator that might be visible.

I don't see how this can be inferred from the above definition of the interface.
I would add that ignoring overriding user-defined operator is almost always a mistake.

> I believe this should be an error in 2019 however, where the LRM text was changed to:

I think it only clarifies the intent.

nickg, 2024-07-24:
> That specifies only how the interface type can be used. But this says nothing about how the operators are associated.

Why does it explicitly call out "predefined" here then? Otherwise "The equality (=) and inequality (/=) operators, implicitly declared as formal generic subprograms ..." alone would have been sufficient. The other uses of "predefined "="" or "predefined equality" in the LRM indicate that any user defined operators should be ignored - for example in the definition of equality for composite types or case statements.

> 
> I don't see how this can be inferred from the above definition of the interface. I would add that ignoring overriding user-defined operator is almost always a mistake.
> 

I agree that the wording is not very clear and the behaviour is quite unintuitive but both ModelSim and RivieraPro have taken the interpretation that it should associate the predefined operator here. GHDL is the only simulator I tested that calls the user-defined function.

tgingold, 2024-07-24:
> Why does it explicitly call out "predefined" here then? Otherwise "The equality (=) and inequality (/=) operators, implicitly declared as formal generic subprograms ..." alone would have been [...]

I understand this as: the interface type provides a type as well as two operators.  I agree that the presence of 'predefined' is useless, because it depends on how the association is done as they are formal generic functions.

The operators could have been implicitly declared after the generics.

> I agree that the wording is not very clear and the behaviour is quite unintuitive but both ModelSim and RivieraPro have taken the interpretation that it should associate the predefined operator here. GHDL is the only simulator I tested that calls the user-defined function.

We agree on the fact the wording is not clear.  Do ModelSim and RivieraPro change their behaviour with vhdl 2019 ?

nickg, 2024-07-24:
> 
> Do ModelSim and RivieraPro change their behaviour with vhdl 2019 ?

The version of RivieraPro on EDA Playground doesn't call the user-defined operator in 2019 either, I'm pretty sure that's a bug.

https://www.edaplayground.com/x/Dizq

The ModelSim/Questa version I have (2021.2) doesn't support 2019 at all. Does anyone have access to another 2019 simulator?

JimLewis, 2024-07-25:
>>>— The predefined equality (=) and inequality (/=) operators, implicitly declared as formal generic subprograms immediately following the interface type declaration in the enclosing interface list

>>That specifies only how the interface type can be used. But this says nothing about how the operators are associated.

> Why does it explicitly call out "predefined" here then? Otherwise "The equality (=) and inequality (/=) operators, implicitly declared as formal generic subprograms ..." alone would have been [...]

I think the "predefined equality" operator refers to the formal generic type.   As a formal generic type, it would only have a "predefined equality" operator.   I think the actual equality operator is the one that is visible to the instance of the generic package - which in the example above is the one from pack1.

nickg, 2024-07-25:
> 
> I think the "predefined equality" operator refers to the formal generic type. As a formal generic type, it would only have a "predefined equality" operator. Note that it refers to this as the "formal generic subprogram". This means that something has to be mapped to it. Since the mapping does not have to be done explicitly, we have to assume that it implicitly does a mapping from the formal subprogram to an actual subprogram. The actual subprogram is going to be something that is visible to the instance of the generic package - which in the example above is the one from pack1.

But the 2008 LRM never specifies that it should behave as if written with `<>`. I think it's equally valid to assume that the intention is to associate the predefined equality operator for the actual type, and the package author should explicitly add `function "=" (x, y : t) return boolean is <>` to the generic list to associate the user-defined operator, which would then hide the predefined operator.

For what it's worth I did originally implement this the way @JimLewis and @tgingold suggest, by doing a lookup for a visible "=" operator for the actual type. But I changed it after a bug report similar to this one, where a user had a case like the following:

```vhdl
package pack1 is
    type t is (a, b);
end package;

-------------------------------------------------------------------------------

package pack2 is
    generic (type g);
end package;

-------------------------------------------------------------------------------

use work.pack1;

package pack3 is new work.pack2 generic map ( pack1.t );
```

ModelSim and RivieraPro accept this even though there is no visible `"="` operator for `pack1.t` at the point of the package instantiation.

JimLewis, 2024-07-25:
The original role I served in ISAC was a user grounding of decisions.   What is the benefit to the users to interpret this one way or another.

If there is an equality operator visible to the instance of the package, there is no use model where I would want to revert back to using the implicitly predefined operator.   

If there is not an equality operator visible to the instance of the package, what do I use?   Remember, there is an implicit mapping going on.  Does the subtype make a predefined operator visible?   Lets test that with the following:

```vhdl
library ieee ;
use ieee.std_logic_1164.all ; 
package type_pkg is
  subtype r_item_t is std_ulogic_vector(15 downto 0);
  constant R_EMPTY : r_item_t := "XXXXXXXXXXXXXXXX";
end package type_pkg;
-------------------------------------------------------
entity ArchTypePkg is
end entity;
architecture test of ArchTypePkg is
  use work.type_pkg.all;
begin
    check: process is
      variable a : r_item_t ; 
    begin
        assert A = A ;
        wait;
    end process;
end architecture;
```

This both GHDL and NVC agree that there is no predefined equality operator visible for a subtype.   Hence, you must reference std_logic_1164 in the architecture to get the equality operator.

JimLewis, 2024-07-25:
> ModelSim and RivieraPro accept this even though there is no visible "=" operator for pack1.t at the point of the package instantiation.

Siemens and Aldec both allow me to call OSVVM's alerts in a resolution function, but GHDL does not.    Alerts indirectly call methods of a protected type, and hence, are impure - which makes the resolution function impure - which is clearly illegal.

JimLewis, 2024-07-25:
If we look at elaboration, 
VHDL-2008:  Elaboration 14.3.3.1 General
...
Similarly, the generic association list contains an implicit association element for each generic subprogram
that is not explicitly associated with an actual or that is associated with the reserved word open; the actual
part of such an implicit association element is determined by the interface subprogram default as described
in 6.5.6.2. The generic association list also contains implicit association elements for the predefined equality
(=) operator and inequality (/=) operators of each generic type; the actual part of such an implicit association
element is the *_name_ of the predefined equality operator or inequality operator* for the base type of the
subtype indication in the actual part of the association element corresponding to the generic type.

VHDL-2019: Elaboration 14.3.3.1 General
...
Similarly, the generic association list contains an implicit association element for each generic subprogram
that is not explicitly associated with an actual or that is associated with the reserved word open; the actual
part of such an implicit association element is determined by the interface subprogram default as described
in 6.5.6.2.

Note that VHDL-2019 removes the last sentence about using the predefined equality operator.

That last sentence in VHDL-2008 looks like it throws a monkey wrench in things and makes the Siemens and Aldec implementations LRM compliant, however, I do not think so.   

What I think it says is that it starts with this:
```vhdl
package r_gen_pkg is
  new work.gen_pkg generic map(item_t => r_item_t);
```

And then transforms it to this:
```vhdl
package r_gen_pkg is
  new work.gen_pkg generic map(
       item_t => r_item_t,
       "="     => "="
       "/="    => "/="
   );
```

Note that it said *name* of the predefined equality operator.   Note that it did not say that it mapped the implicit operator.   Note that it did not say it made it made the implicit operator visible if it was not otherwise visible.   

I think once VHDL-2008 does this mapping, then normal visibility rules and overloading rules have to be taken into account.   And hence if there is an explicit operator visible, then it is used and not the predefined (implicit) operator.

Hence, this brings me back to believing that the package reference to std_logic_1164 is indeed required here as otherwise there would be no "=" or "/=" operator visible.

```vhdl
library ieee;
  use ieee.std_logic_1164.all;
use work.type_pkg.all;
package r_gen_pkg is
  new work.gen_pkg generic map(
       item_t => r_item_t,
       "="     => "="
       "/="    => "/="
   );
```

tgingold, 2024-07-26:
@JimLewis Thank you for the analysis!

nickg, 2024-07-26:
Wouldn't the "name of the predefined equality operator" here be `ieee.std_logic_1164."="` which would make it visible by selection?

nickg, 2024-07-26:
@JimLewis do you have any contacts at Siemens or Aldec you could discuss this with? It would be best if all vendors could agree on the same behaviour.

JimLewis, 2024-07-26:
@nickg  I don't think that is the intent of the language.   

That sort of an interpretation would make the language inconsistent with other parts of the language - such as is demonstrated with my example, entity ArchTypePkg.   

In addition, that interpretation in the presence of the visible "=" operator such as the one from pack1 in example entity gentype6 is, in the word s of my son, maddening, and from my perspective confusing.  Particularly if you consider what is required when mapping other generic subprograms.

I don't disagree that the wording is confusing and subject to different interpretations.   However, when this happens, I suggest that we take an interpretation that is consistent with other aspects of the language, has the only sensible use model, and is consistent with the wording of VHDL-2019.

In the past when there was such a case, ISAC (issues screening and analysis committee) would convene, do an analysis, and issue an interpretation.    At least at one point in time, IEEE started forbidding IEEE WG from making interpretations of a standard.   I am not sure if this is still the case or not.  I am also not sure if our interpretation of "interpretation" was too literal or not - as what we are doing is a clarification of the standard and it is not an interpretation that asserts anything about compliance to the standard - I think asserting compliance is actually what IEEE was worried about as there would be legal issues with that for product type stuff.

Paebbels, 2024-07-26:
> @tgingold wrote:  
> That specifies only how the interface type can be used. But this says nothing about how the operators are associated.

I consent here. When [LCS2016_059](https://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_059) was written, it only (intentionally) changed who many operations will be available. With VHDL-2008, it was limited to `=`, `/=`, `new`, `deallocate`, `type(...)`, `:=`/`<=`. The change should not touch how types are associated and what operations are associated.