ghdl/ghdl #2697: Using 'work' library

AdamKeith1, 2024-07-23:
**Description**
Tool doesn't recognize the work library.

**Expected behaviour**
Design file imports packages from work library, and while the files are included in the .sby file (im using symbiyosys) since they aren't in a 'work' library, my design doesn't recognize them.

**How to reproduce?**
I have a very large design and can't paste it all for a few reasons. My design imports packages from multiple locations, I have added these files in the [files] section of the .sby file and they show up in my src folder when I go to run a test.
ex. `USE work.axi_pkg.ALL;`

My 'testbench' is in System Verilog. I go instantiate the vhdl design in the tb and it works just fine, it's when the design is being elaborated and the 'work' library doesn't exist that the errors for unknown constants show up. Basically my question is how do I get these files into a work library for ghld/sby to recognize since I can't change my source code. Again I apologize for not being able to post the code, there are a few reasons for this.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS: Ubuntu 22.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2024-07-23:
You can do:
```
ghdl --work=liba filea.vhdl --work=libb fileb.vhdl -e
```

I am not sure if it answer your question, it would be nice to have a reproducer (or a mock-up) to understand clearly the issue.

AdamKeith1, 2024-07-23:
Kind of, I'll show what I'm able to. So I have...

design.vhd
package1.vhd
...
packagen.vhd

as my files that I have included in my .sby file

design.vhd uses these packages via `USE work.axi_pkg.ALL;`

I have testbench.sv and I **can't** include packages in that test bench since it's a paid feature. But I am able to instantiate the vhdl design, so the only way I can use those packages is via those USE statements in design.vhd but those files are just listed all together, and aren't specified to be in 'work' library. So if I have...

`ghdl --std=08 --ieee=synopsys design.vhd -e design`

So for the packages that I want in the work library I would do...

`ghdl --work=<IDK> file1.vhd file2.vhd ... filen.vhd -e`

My confusion is what is the `--work=` doing and based on that what do I specify after this?

AdamKeith1, 2024-07-24:
What does `--work=` do exactly?

tgingold, 2024-07-28:
The option `--work=NAME` just set the name of the `work` library.
By default, it is named `work`.
If you want to put analyzed files into the `mylib` library, you need to specify `--work=mylib`.

closed 2024-07-28 by tgingold