ghdl/ghdl #2700: Creating Library

AdamKeith1, 2024-07-24:
I have a lot of packages I need to use in my design. I was wondering if it's possible to make a directory in the ghdl/libraries/ directory and use that. I made a folder titled 'work' and copied my .vhd packages over and tried to access them in my code (I'm using SymbiYosys just for reference) and I get this error.

```
root@fpga-symbiyosys--0_43_0:~/demos/wiggle_test# make run MODE=bmc
sby  --yosys "yosys -m ghdl" -f test.sby bmc
SBY  9:56:03 [test_bmc] Removing directory '/root/demos/wiggle_test/test_bmc'.
SBY  9:56:03 [test_bmc] Copy '/root/demos/wiggle_test/tb.sv' to '/root/demos/wiggle_test/test_bmc/src/tb.sv'.
SBY  9:56:03 [test_bmc] Copy '/root/demos/wiggle_test/b.vhd' to '/root/demos/wiggle_test/test_bmc/src/b.vhd'.
SBY  9:56:03 [test_bmc] Copy '/root/demos/wiggle_test/a.vhd' to '/root/demos/wiggle_test/test_bmc/src/a.vhd'.
SBY  9:56:03 [test_bmc] Copy '/root/demos/wiggle_test/top.vhd' to '/root/demos/wiggle_test/test_bmc/src/top.vhd'.
SBY  9:56:03 [test_bmc] engine_0: smtbmc boolector -- --keep-going
SBY  9:56:03 [test_bmc] base: starting process "cd test_bmc/src; yosys -m ghdl -ql ../model/design.log ../model/design.ys"
SBY  9:56:03 [test_bmc] base: a.vhd:8:10:error: unit "fpga_config_pkg" not found in library "work"
SBY  9:56:03 [test_bmc] base: use work.fpga_config_pkg.all;
SBY  9:56:03 [test_bmc] base: ^
SBY  9:56:03 [test_bmc] base: a.vhd:13:40:error: no declaration for "active_bits_c"
SBY  9:56:03 [test_bmc] base: CONSTANT NUM_BITS : integer := ACTIVE_BITS_C
SBY  9:56:03 [test_bmc] base: ^
SBY  9:56:03 [test_bmc] base: ERROR: vhdl import failed.
SBY  9:56:03 [test_bmc] base: finished (returncode=1)
SBY  9:56:03 [test_bmc] base: task failed. ERROR.
```

I see a Makefile include in the libraries directory and I'm wondering what are all the things I'd have to do/change to create my own library to use? Or is it possible to throw the packages under an existing library (say std or ieee for ex.).

AdamKeith1, 2024-07-24:
I ran `ghdl -a --work=work fpga_config_pkg.vhd` in the command line and it created a file called `work-obj93.cf` in my directory with contents...
```
v 4
file . "fpga_config_pkg.vhd" "07c4529063c827d480a05a1a4e6020b8c5f65907" "20240724174732.416":
  package fpga_config_pkg at 7( 251) + 0 on 12;

```
But when I go to run...
```
SBY 10:50:09 [test_bmc] engine_0: smtbmc boolector -- --keep-going
SBY 10:50:09 [test_bmc] base: starting process "cd test_bmc/src; yosys -m ghdl -ql ../model/design.log ../model/design.ys"
SBY 10:50:09 [test_bmc] base: a.vhd:9:10:error: unit "fpga_config_pkg" not found in library "work"
SBY 10:50:09 [test_bmc] base: use work.fpga_config_pkg.all;
SBY 10:50:09 [test_bmc] base: ^
```
Here's my lib call...

```
library work;
use work.fpga_config_pkg.all;
```
and my sby file...

```
[tasks]
bmc
cover

[options]
bmc:
mode bmc
depth 50
--
cover:
mode cover
depth 50
--

[engines]
smtbmc boolector -- --keep-going

[script]
read -formal tb.sv
ghdl --std=08 --ieee=synopsys a.vhd -e mod_a
ghdl --std=08 --ieee=synopsys b.vhd -e mod_b
ghdl --std=08 --ieee=synopsys top.vhd -e top
prep -top test -flatten

[files]
tb.sv
b.vhd
a.vhd
top.vhd
fpga_config_pkg.vhd
```

AdamKeith1, 2024-07-24:
Important to note that when I ran the ghdl -a command in the scripts section instead of command line it fails and doesn't recognize the -a flag.

tgingold, 2024-07-24:
Try: `ghdl -a --std=08 fpga_config_pkg.vhd`

`--work=work` is useless as the default name of the work library is `work`.

tgingold, 2024-07-24:
Probably in the sby script you should do:
`ghdl --std=08 --ieee=synopsys fpga_config_pkg.vhd a.vhd -e mod_a`

AdamKeith1, 2024-07-24:
Worked. Thanks!

tgingold, 2024-07-24:
Great!

closed 2024-07-28 by tgingold