ghdl/ghdl #2704: reading procedure ports defined as `out`

jeras, 2024-07-28:
latest Git version

I have some VIP code with procedures handling an AXI handshake (the usual stuff).
The `valid` signal is an `out`put while `ready` is in `in`put (or the opposite, depending on context).
I wrote a few transfer checks (`valid and ready`) inside the procedure and for the output signal I get the following error:
```
error: signal interface "src" cannot be read
```
Where `src` is the output record containing the `valid` signal.

This seems like an error from before VHDL-2008 allowed read access to output signals, but instead of a component it happens with a procedure.

I can prepare a testcase, but for this report is at least a placeholder.

Paebbels, 2024-07-28:
Please provide a reproducer and the commands how you called GHDL. In addition, your GHDL version could help.

diogratia, 2024-07-28:
A procedure is a subprogram and has parameters, not ports.

IEEE Std 1076-2008  

4.2 Subprogram declarations  
4.2.1 General  

>A subprogram declaration declares a procedure or a function, as indicated by the appropriate reserved word.  
>  
>subprogram_declaration ::=  
&nbsp;&nbsp;&nbsp;&nbsp;subprogram_specification ;
>  
>subprogram_specification ::=  
&nbsp;&nbsp;&nbsp;&nbsp;procedure_specification | function_specification  
>  
>procedure_specification ::=  
&nbsp;&nbsp;&nbsp;&nbsp;**procedure** designator  
>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;subprogram_header  
>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;[ [ **parameter** ] ( formal_parameter_list ) ]

6.5.2 Interface object declarations

>An interface object has one of the following modes:  
>>..  
>>— **out**. The value of the interface object is allowed to be updated and, provided it is not a signal parameter, read. Reading the attributes of the interface object is allowed, unless the interface object. is a signal parameter and the attribute is one of 'STABLE, 'QUIET, 'DELAYED, 'TRANSACTION, 'EVENT, 'ACTIVE, 'LAST_EVENT, 'LAST_ACTIVE, or 'LAST_VALUE.

jeras, 2024-07-28:
Following the standard details posted by @diogratia (could you check, if this shanged in VHDL-2019?), it appears the GHDL behavior is correct and Questa would have to report an error. I dislike using `inout` so I will wait a bit before deciding on how to approach this.

Here is the example code:
```VHDL
library ieee;
use ieee.std_logic_1164.all;

use std.env.all;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ghdl_bug_tb is
end ghdl_bug_tb;

architecture behavior of ghdl_bug_tb is

   -- Clock period definitions
   constant T_C  : time := 10.0 ns; -- Clock period constant

   -- system signals
   signal clk    : std_logic := '1'; -- clock
   -- handshake
   signal valid : std_logic;
   signal ready : std_logic;

   -- Wait for a number of rising edges and TPD
   procedure tbClkPeriod(
      signal   clk_i : in std_logic;
      constant num_i : in natural := 1
   ) is
   begin
      for i in 0 to num_i-1 loop
         wait until rising_edge(clk_i);
      end loop;
   end tbClkPeriod;

begin

   -- clock source
   p_Clock : process
   begin
      clk <= '1'; wait for T_C/2;
      clk <= '0'; wait for T_C/2;
   end process p_Clock;
   
   p_test_valid : process
      procedure drv_vld (
         signal clk : in  std_logic;
         signal vld : out std_logic;
      -- signal vld : inout std_logic;
         signal rdy : in  std_logic
      ) is
      begin
         tbClkPeriod(clk);
         vld <= '1';
         loop
            tbClkPeriod(clk);
            -- exit on transfer
            exit when (vld = '1' and rdy = '1');
         end loop;
         vld <= '0';
      end procedure drv_vld;
   begin
      valid <= '0';
      tbClkPeriod(clk, 2);
      drv_vld(clk, valid, ready);
      tbClkPeriod(clk, 4);
      finish;
   end process p_test_valid;

   p_test_ready : process
      procedure drv_rdy (
         signal clk : in  std_logic;
         signal vld : in  std_logic;
         signal rdy : out std_logic
      -- signal rdy : inout std_logic
      ) is
      begin
         tbClkPeriod(clk);
         rdy <= '1';
         loop
            tbClkPeriod(clk);
            -- exit on transfer
            exit when (vld = '1' and rdy = '1');
         end loop;
         rdy <= '0';
      end procedure drv_rdy;
   begin
      ready <= '0';
      tbClkPeriod(clk, 2);
      drv_rdy(clk, valid, ready);
      tbClkPeriod(clk, 4);
      finish;
   end process p_test_ready;

end behavior;
```

And the reported error is:
```sh
ghdl_bug_tb.vhd:46:17:warning: declaration of "clk" hides signal "clk" [-Whide]
         signal clk : in  std_logic;
                ^
ghdl_bug_tb.vhd:57:24:error: signal interface "vld" cannot be read
            exit when (vld = '1' and rdy = '1');
                       ^
ghdl_bug_tb.vhd:71:17:warning: declaration of "clk" hides signal "clk" [-Whide]
         signal clk : in  std_logic;
                ^
ghdl_bug_tb.vhd:82:38:error: signal interface "rdy" cannot be read
            exit when (vld = '1' and rdy = '1');
                                     ^
*command line*:1:1:error: unit entity "ghdl_bug_tb" has not been analyzed
```

jeras, 2024-07-28:
For the other error it will take me more time to bisect the code, and it will be a separate issue.

jeras, 2024-08-02:
Since this example did not output an error in Questa, I tried it on other simulators on Edaplayground and got the following results:

https://www.edaplayground.com/x/SPWm

| simulator | result |
|---|---|
| Aldec Rivera Pro (`-2019` option) | "ERROR COMP96_0393: "Cannot read output" |
| Cadence Xcelium | no issues |
| Siemens Questa | no related issues |
| Synopsys VCS | no issues |

Based on the results, I would say, most commercial simulators would not report reading the output. And without me actually looking into the standard, from the cited portion, it seems like the standard could just allow reading of output, as it is allowed for components ports, instead it seems they forgot changing it. Again without looking at the 2019 standard, Aldec Rivera Pro is reporting this as an error according to the 2019 version of the VHDL LRM.

While changing the parameter direction to `inout` is a trivial workaround in most use cases, it is not the same.

There is no obvious solution which would follow the standard and be compatible with other simulators, and while my usual approach would be to follow the standard, this does seem like an omission in the standard. So I am unable to decide, but I see the path of least resistance.

Paebbels, 2024-08-03:
Besides the things specific to GHDL, all general ideas about your problem and/or enhancement ideas can be discussed here: https://gitlab.com/IEEE-P1076/VHDL-Issues 😄