ghdl/ghdl #2705: Range as type, regression?

pat-pgt, 2024-07-28:
**Description**
Is it a regression? I am sure to already used that.
However I may be wrong as it can be on commercial software (not GHDL). It be can an unofficial extension.

type xyz_rng is integer range <> := 31 downto 0;

**Expected behaviour**
Use it in declarations as
signal xyz : std_logic_vector( xyz_rng );

GHDL 4.1.0 compiled on Linux using Gcc 8.5.0

Paebbels, 2024-07-29:
`subtype xyz_rng is integer range 31 downto 0;`

`type` declares a new type, while `subtype` constrains an existing type.

A new integer type would be `type t is range 31 downto 0;`, but this type wouldn't be compatible with std_logic_vector as it's a new type.

pat-pgt, 2024-07-29:
Once more some commercial tools extends the language to make incompatible with the free software.
We could have imagine range<> can be defined.

Some time ago, I had the same problem with a configuration of an instantiation under a generic. Since one does not have the instances names, the configure can not do anything. GHDL return a clear warning, some other accepts or says nothing.

closed 2024-07-29 by pat-pgt

Paebbels, 2024-07-29:
Sorry, but your request makes no sense.

There are no such vendor extensions to the language. At first, this is not following the Language Reference Manual (LRM). At second, vendors are lazy, so they barely implement all language features, why should they add a new feature? If you like to discuss language extensions and enhancements, please add your request to https://gitlab.com/IEEE-P1076/VHDL-Issues and/or join the IEEE working group.

At next, all what you requested is possible with the current language as it is. You just mixed up your syntax. More over as I tried to explain, you request makes not even sense, because ranges of types can't be used to constrain arrays when there is already a type defined for that constrained. Only subtypes can be used.

-------
For you syntax: `type xyz_rng is integer range <> := 31 downto 0;`  
* `integer` is an existing type, but type defines new types
* `range <>` is only allowed in unconstrained array definitions
* `:=` initial values are only allowed for objects
* `31 downto 0` is a range. Ranges are no first class entities in VHDL.

tgingold, 2024-07-29:
@pat-pgt It would be interesting to know which commercial tool allows such an extension!