ghdl/ghdl #2706: Simulation time in waves

oscargus, 2024-07-29:
**Description**
It seems like the last event(?) is not stored in the waveform output. (This is checked with both GTKWave and Surfer. From reading the VCD source, it is clear that nothing else can be expected.)

**Expected behaviour**
The waveforms should include the whole simulation time.

**How to reproduce?**
See https://github.com/oscargus/fulladder (the GHW-waveforms can be downloaded and viewed from https://oscargus.github.io/fulladder/ which shows the work-in-progress purpose of the project).

This is a (very simple) vunit-project, but I doubt it is related to that. Running the testbenches, the correct one states that it ends after 8 ns, but the waveform only contains values up to 7 ns. This is consistent among all three formats: vcd, fst, and ghw.  Now, the last event is indeed at 7 ns, but it is a bit confusing that it ends there.

The failing case may be more relevant as motivation as the failing check happens at 5 ns, but only 4 ns are shown in the simulation.

(See the "System output" in https://oscargus.github.io/fulladder/ which confirms that the simulation time is 8 and  5 ns, respectively.)

Not sure how FST and GHW work in detail, but for VCD it would most likely be enough to always add a time stamp with the final simulation time. Maybe something similar can be done for GHW and FST.

**Context**
Please, provide the following information:

(The CI-run linked above is using ghdl 1.0, but this also happens locally on:)

- OS: WSL Ubuntu 24.04 LTS
- Origin:
  - [x] Package manager: `version`
```
GHDL 4.1.0 (Ubuntu 4.1.0+dfsg-0ubuntu2) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 static elaboration, mcode JIT code generator
```

tgingold, 2024-07-30:
Make sense, but we need to be careful as simulation also stops at `time'high`.

tgingold, 2024-07-31:
But I started to investigate with the vcd output, and no: the last time in the vcd file is the last time of the simulation.

Do you have the vcd file for your testbench ?

oscargus, 2024-07-31:
Sure, I changed the webpage to get the VCDs instead (as one cannot upload vcd-files here). I can, however, confirm that that locally generated files (with ghdl 4.1) ends with:

```
#6000000
1%
0&
1'
b110 (
b10 )
1,
0-
1.
#7000000
```

and 

```
#3000000
#4000000
1$
0&
1'
b011 (
b10 )
1+
0-
1.
```

respectively.

I also note that the check happening at 7 ns leads to a time stamp, but the failing check at 5 ns in the failing test does not generate one.

Maybe this is fixed in the main branch? Will see if I can build it.

tgingold, 2024-07-31:
Ah OK, I didn't realize the time was missing only in case of error.

tgingold, 2024-07-31:
But when I run using `run.py  --gtkwave-fmt=vcd` with vunit `d93c087ce`, I got this vcd file (end only):

```
#3000000
#4000000
1$
0&
1'
b011 (
b10 )
1+
0-
1.
#5000000
```

So it looks ok for me...

tgingold, 2024-08-01:
I cannot reproduce with the mcode backend, but it happens with the llvm backend.
However, you mentioned you are using the mcode backend.

oscargus, 2024-08-01:
It is missing in both cases. The simulation time for the "correct" test bench is 8 ns, but the last time stamp is 7ns (log says `simulation stopped @8ns with status 0`). But also the failing one is missing the final time step. 

But indeed, your result looks good! (And different to mine.)

Yes, the versions used are both mcode.

On CI (vunit 4.7):
```
GHDL 1.0.0 (Ubuntu 1.0.0+dfsg-6) [Dunoon edition]
 Compiled with GNAT Version: 10.3.1 20211117
 mcode code generator
```

Locally (vunit d93c087ce, earlier I used a slightly older version but still from source):
```
GHDL 4.1.0 (Ubuntu 4.1.0+dfsg-0ubuntu2) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 static elaboration, mcode JIT code generator
```

I'll try to install some other versions on different computers and see if I can find some additional pattern.

oscargus, 2024-08-01:
Minor update, I installed:
```
GHDL 4.1.0 (4.0.0.r39.g7188e92cf) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 static elaboration, mcode JIT code generator
´``
natively on Windows (the latest UCRT64 in Releases on GitHub) and get the same result, missing final time stamp in both files. 

This makes me wonder if it may be some flushing issue (on Windows?)? Except that the CI run has the same issue on non-Windows...

tgingold, 2024-08-01:
For CI, the ghdl version is very old.
So I have to try on windows.

oscargus, 2024-08-02:
Now I compiled the latest ghdl from source on an ArchLinux machine.
```
GHDL 5.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 14.1.1 20240522
 static elaboration, mcode JIT code generator
```
Final time stamps are missing.

```
GHDL 5.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 14.1.1 20240522
 llvm 18.1.8 code generator
```
Final time stamps are missing.

(I have not yet managed to build with gcc.)

tgingold, 2024-08-03:
I have added two tests (in `testsuite/gna/iss2706`).  They don't fail.
Can you run it on your machine(s) ?  If they pass, it means the reproducer is not correct.

oscargus, 2024-09-06:
Sorry for letting this slip. I hope to find time in the upcoming few weeks.