ghdl/ghdl #2708: GHDL can't be correctly installed on macOS (arm64) using Homebrew

Paebbels, 2024-07-31:
**Description**
There are many problems with installing GHDL on macOS using Homebrew.
(Not all is directly related to GHDL or the GHDL repository, but we know we have many macOS users and it should be in our interest to support this platform especially as it's also available as a CI environment for cross-platform testing.)

My usecase is setting up GHDL on GitHub Actions to run some testcases.

**So what goes all wrong?**
1. The current Formular at Homebrew doesn't list any dependencies to e.g. LLVM@15. Besides that, all other dependencies should be listed, too. Currently the Formular relies on certain libraries magically existing in macOS. That's not good.
2. The current Formular at Homebrew installs on arm64 platforms with LLVM backend and on Intel x86-64 platforms as mcode. Why do I need different hardware to change the GHDL backend?
3. The current Formular builds from GHDL v4.1.0 release, but GHDL isn't precompiled for macOS 12, macOS13 and macOS 14. So there is not even an arm64 pre-build that could be used. (see also #2707)
4. On ARM based macOS (13, 14), Homebrew installs packages to a new location in `/opt/homebrew` compared to `usr/local` for Intel based macOS. Moreover the `lib` directory is not in search path. As a result, LLVM@15 (`libLLVM.dylib`) can't be found and loaded.  
   According to this article, the problem can't be solved: https://github.com/Homebrew/brew/issues/13481
   ```
   dyld[4314]: Library not loaded: /usr/local/opt/llvm@15/lib/libLLVM.dylib
     Referenced from: <3CC36E74-F6A6-3EF6-95CD-7BD3ECE10FBC> /opt/homebrew/Caskroom/ghdl/4.1.0/bin/ghdl1-llvm
     Reason: tried: '/usr/local/opt/llvm@15/lib/libLLVM.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/llvm@15/lib/libLLVM.dylib' (no such file), '/usr/local/opt/llvm@15/lib/libLLVM.dylib' (no such file), '/usr/local/lib/libLLVM.dylib' (no such file), '/usr/lib/libLLVM.dylib' (no such file, not in dyld cache)
   /opt/homebrew/Caskroom/ghdl/4.1.0/bin/ghdl:error: exec error
   ```
   GHDL installation directory: `/opt/homebrew/Caskroom/ghdl/4.1.0/bin/ghdl1-llvm`
   GHDL symlinks: `/opt/homebrew/bin/ghdl`
   LLVM searched at: `/usr/local/opt/llvm@15/lib/libLLVM.dylib`
   LLVM found at: `/opt/homebrew/opt/llvm@15/libLLVM.dylib`

**Suggested changes**
* [ ] GHDL should create a new release to offer pre-compiled results like in `nightly`, so Homebrew and other package managers can pick it up.  
      See #2707
* [ ] GHDL should create ARM64 builds for macOS 13 and macOS14.  
      See #2707
* [ ] Should GHDL provide *Universal macOS binaries* (combined binaries for x86-64 and arm64)?
      See also: https://developer.apple.com/documentation/apple-silicon/building-a-universal-macos-binary
* [ ] Homebrew should offer ghdl as mcode and LLVM backends, independent of the hardware platform.

**Open questions:**
I have no idea how to solve the `$DYLD_LIBRARY_PATH` problem.

Paebbels, 2024-07-31:
@nickg suggested this in Gitter:
> The Cask definition might be missing a  
> `depends_on formula: "llvm@15"`  
> See https://docs.brew.sh/Cask-Cookbook#stanza-depends_on

Paebbels, 2024-07-31:
Checking out the cask overview page, an installation on a ARM platform shoudn't be possible:
[image]

But brew installs GHDL without any error or warning.

tgingold, 2024-07-31:
Why not simply build ghdl for arm64 ?

Paebbels, 2024-07-31:
You mean via make in the CI run of my Python code?

It takes many minutes to compile it. The overall Python project usually has just a few minutes for the complete pipeline. This would drastically increase the pipeline runtime.

Moreover, the unittest template allows to install packages via apt, pacman and homebrew before running pytest. I don't like to add more exceptions to the template like user defined download and compile code. Because this needs to be added then to all platforms (Ubuntu, Windows, MSYS, ...).

I could compile it in a separate job and then ship it as an artifact... Such a job could run in parallel to some tests in my Python pipeline.

------
I think I can find a workaround for my initial problem, but I think in general this brew stuff should work (or it can be deleted if it doesn't work).

Another option I think about is switching the whole macOS pipeline from ARM to Intel on GitHub. OTOH, my Python pipeline templates have no concept for two macOS platforms at all :(.

tgingold, 2024-07-31:
I am not sure to understand the relation between homebrew and the ghdl CI builds.
Does homebrew simply install the ghdl release built by CI or do they have their own builders ?

Paebbels, 2024-07-31:
As I read the ruby code (https://github.com/Homebrew/homebrew-cask/blob/master/Casks/g/ghdl.rb), it refers directly to the assets created by the GHDL pipeline.

It uses:
`https://github.com/ghdl/ghdl/releases/download/v#{version}/ghdl-macos-11-#{arch}.tgz`
to refer to the linked asset in the 4.1.0 release for a tgz file.

So compared to Debian/Ubuntu or even MSYS2 with own CI jobs and maybe patches builds, they don't create there own binary files. Maybe homebrew can do so, but that Cask code just refers to a tgz we "unintentionally" provide by own GHDL pipeline.

Maybe someone with better homebrew knowledge can support here. I just tried to dig into that code, settings etc. and that's what my common sense reads from the files I found.

Paebbels, 2024-07-31:
I think you, me, @umarcor were not aware that a package manager system would directly use the provided assets. I know that user will download maybe manually zips or files for manual installation, but not that our pipeline and assets are a direct and unbuffered input.

Paebbels, 2024-08-01:
Today I tried to switch my Pipeline back to use Intel macOS images. Then I got this message:
> The job was not started because recent account payments have failed or your spending limit needs to be increased. Please check the 'Billing & plans' section in your settings.

So looks like switching back to Intel machines is no option for macOS.

Paebbels, 2024-08-06:
@tgingold 
> Why not simply build ghdl for arm64 ?

I can now build for macOS, MinGW64, UCRT64 and Ubuntu 24.04 with mcode and llvm backends.

When linking `ghdl` (actually `ghdl1-llvm`) with static LLVM, `libghdl` isn't linked with static dependencies.

Paebbels, 2025-03-08:
@khipp updated homebrew's casks. The casks was modified, so maybe the described problem is solve. I'm waiting for confirmation.

Diff: https://github.com/Homebrew/homebrew-cask/commit/fbfd23742363d23fe1efe43f6f529fbeb7702d66

Paebbels, 2025-03-08:
It was confirmed to be fixed.

closed 2025-03-08 by Paebbels