ghdl/ghdl #2710: GHDL is not correctly installed on Ubuntu 2024.04 LTS

Paebbels, 2024-08-03:
**Description**
When checking GitHub Action pipeline that utilize GHDL, I found already multiple problems. E.g. with GHDL installation on macOS using homebrew (see #2708).

Now I checked GHDL on Ubunto 2024.04.
It can be installed as llvm or mcode backend with shared `src` and `include` directories. Unfortunately, some essential parts of GHDL are missing:
* `lib/libghdl.so` is not (build and) shipped
* `lib/libghdl.a`
* `lib/libghdl.link`
* `includes/ghdl/synth.h` and
* `includes/ghdl/synth_gates.h`
* `lib/ghdl/vendors/*.sh`

Paebbels, 2024-08-03:
OK, I found some parts as:
* `libghdl` and
* `libghdl-dev`

packages.

When installing an error is printed:
> Processing triggers for libc-bin (2.39-0ubuntu8.2) ...
> /sbin/ldconfig.real: /usr/lib/wsl/lib/libcuda.so.1 is not a symbolic link

Paebbels, 2024-08-03:
GHDL has 2 different backends. This is also important for libghdl, but only one (unknown) backend variant is installed.

```
paebbels@Paebbels:~$ find /usr -name "libghdl*"
/usr/share/doc/libghdl-dev
/usr/share/doc/libghdl-4-1-0
/usr/lib/x86_64-linux-gnu/libghdl-4_1_0.so
/usr/lib/x86_64-linux-gnu/libghdl.so
/usr/lib/x86_64-linux-gnu/libghdl.a
/usr/lib/ghdl/llvm/libghdlvpi.so
/usr/lib/ghdl/mcode/libghdlvpi.so
```

Paebbels, 2024-08-03:
@tgingold does GHDL know where it's libghdl is?

I see many different locations for Ubuntu, MSYS2, macOS, ... It get's tricky finding the right dll/so/dylib when the library path is not from ghdl binary: `..../bin/ghdl../../lib/libghdl.so`.

Paebbels, 2024-08-04:
# A new summary of investigation results

**How GHDL is currently installed on Ubuntu 2024.04:**
* `/usr/`
  * `bin/`
    * `ghdl`
    * `ghdl-mcode`
    * `ghdl-llvm`
  * `include/`
    * `ghdl/`
      * `synth.h`
      * `synth_gates.h`
  * `lib/`
    * `ghdl/`
      * `include`
        * `ghdl`
          * `vhpi_user.h`
          * `vpi_user.h`
      * `llvm`
        * `ghdl1-llvm`
        * `libghdlvpi.so`
        * `vhdl`
          * `src` -> `../../src`
          * `...`
      * `mcode`
        * `libghdlvpi.so`
        * `vhdl`
          * `src` -> `../../src`
          * `...`
      * `src`
        * `ieee`
        * `ieee2008`
        * `std`
        * `synopsys`
        * `upf`
    * `x86_64-linux-gnu`
      * `libghdl-4_1_0.so`
      * `libghdl.a`
      * `libghdl.so` -> `libghdl-4_1_0.so`

**Problems:**  
`libghdl.so` can't find the precompiled libraries. `GHDL_PREFIX` was set to `/usr/lib/x86_64-linux-gnu`, because it needs to point to the directory, the `*.so`-file is in. By splitting all components into many packages and scattering these components over dozens of directories, GHDL (especially libghdl) is broken. And obviously it was never checked by the package maintainers.

(Do/can packages have sanity checks?)

**Good:**
* 3 variants are available: mcode, llvm, gcc
* Version: 4.1.0 (latest release)
* Backend can be selected by undocumented `GHDL_BACKEND` environment variable.
  * In case of an error, exit code 2 is used, which is also used by GHDL itself !!

**Bad:**
* `/usr/bin/ghdl` is not a symlink to the selected backend.
* No alternatives are registered for GHDL:
  ```
  paebbels@Paebbels:~$ update-alternatives --list ghdl
  update-alternatives: error: no alternatives for ghdl
  ```
* The backend for the installed `libghdl.so` file is unclear.
  * The package description in `apt` has also no hint.
* The backend for `libghdl` can't be selected like for ghdl itself.
* The header file for `libghdl` in not installed by the `libghdl-dev` package !
* Some headers are in `usr/include/ghdl` others are in `usr/lib/ghdl/include`.
* The binary `ghdl1-llvm` is installed to `/usr/lib` !
* VHPI and VPI header files have 2x `ghdl in path`: `usr/lib/ghdl/include/ghdl` !
* As `libghdl` could be installed as a different version then GHDL itself, the precompiled VHDL libraries will be incompatible depending on what package was updated as last.

-----
I haven't installed the GCC variant, only mcode and llvm. But to summarize, GHDL on Ubuntu is as broken as on macOS...

Paebbels, 2024-08-04:
I just created an issue at launchpad: https://bugs.launchpad.net/ubuntu/+source/ghdl/+bug/2076043

> The list of complains is quite long, so let me try to structure all individual flaws, mistakes and bugs step-by-step. As a background: I'm a co-maintainer and developer at GHDL. I support GHDL for pyGHDL and using libghdl as standalone as well as providing vendor library compile scripts, helping with CI and make/compile scripts for MSYS2 and Windows.
> 
> All findings and investigations are also collected here: https://github.com/ghdl/ghdl/issues/2710
> 
> 1. **Multiple *.deb packages and parallel installs**  
>    I noticed GHDL was split into multiple packages, so the user can decided on the GHDL backend as well having multiple backends in parallel. This is so far good, but a generic /usr/bin/ghdl is installed as a shell script to switch between backend.
>    * 1.1 The environment variable GHDL_BACKEND is not documented.
>    * 1.2 The internal exit code 2 is already used by GHDL. (If I'm right minimum 10 error codes are used by GHDL itself. For details and getting a free error code, it's recommended to contact the developers, but don't pick randomly a new exit code!
>    * 1.3 Why is alternatives not used to allow variant switching?
> 
> 
> 2. **Unclear backend for libghdl**
>    libghdl is also build for different backend. Thus, it's important which variant is installed.
>    * 2.1 The description in APT does not explain which backend is used by libghdl.
>    * 2.2 A user should be able to select the libghdl backend variant.
>      Note: For using libghdl with pyGHDL, a libghdl with mcode backend is very suitable. Others
>      might prefer the LLVM backend as it's faster, but produces *.o files and an executable on disk,
>      while mcode runs completely in memory.
> 
> 
> 3. **Missing header for libghdl**
>    * 3.1 When installing libghdl-dev, no libghdl header file is installed.
>    * 3.2 In reference to 2.*, each variant of libghdl might have different headers, due to varying APIs.
>      Note: This should be checked with the developers if yes/no it's different and if it might happen in the future.  
>      Please also note: headers might be modified/constructed by ./configure e.g. for default paths, etc.
> 
> 4. **Headers in differing directories**
>    * 4.1 Some headers are installed into /usr/include/ghdl, others to /usr/lib/ghdl/include
>    * 4.2 The VHPI and VPI headers are located in /usr/lib/ghdl/include/ghdl.  
>      The second ghdl needs to be removed.
> 
> 
> 5. **Wrong installation space for ghdl1-llvm**
>    ghdl1-llvm is a binary and should not be installed into /usr/lib
> 
> 
> 6. **libghdl can't find it's pre-compiled VHDL source files**
>    libghdl is installed into /usr/lib/x86_64-linux-gnu.
>    * 6.1 The libghdl-4_1_0.so is not following the conventions of so files in Linux.
>    * 6.2 libghdl cannot find it's pre-compiled VHDL sources, because it was moved to a different directory without patching the paths.
>    * 6.3 If fixed, libghdl would share it's pre-compiled VHDL sources with ghdl itself. It's currently unsure if packages rules are setup to keep ghdl, libghdl, libghdl-dev in sync, otherwise ich backend variant and each version needs it's own pre-compiled VHDL source files.
>    * 6.4 If libghdl will be installed as libghdl-mcode and libghdl-llvm, how to switch *.so files, when alternatives isn't used?  
>      Note: It might be handy for users to use ghdl with LLVM backend and libghdl with mcode backend.
> 
> 7. **Missing pyGHDL**
>    The most popular reason for libghdl is the usage with pyGHDL as either pyGHDL.libghdl for low-level Python bindings to libghdl.so or with pyGHDL.DOM to get a VHDL CodeDOM (Code Document Object Model). The pyGDL version is tightly linked to the libghdl version and vise versa. So it's recommended to install pyGHDL with same GitHash as libghdl.
> 
>    That said:
>    * 7.1 It should be part of libghdl or another package to install python3-pyGHDL 4.1.0 (depending on libghdl-mcode)
> 
> **As a summary:**
> * A) The packaging of GHDL as a binary is questionable (1.*, 5.*).
> * B) libghdl is completly broken and useless (2.*, 6.*).
> * C) I strongly believe libghdl was never used and tested.
> 
> 
> **Suggestions:**
> * a) Please rething if alternative switching shouldn't be done by alternatives. This could also have (independent) rules for libghdl.
>   If not, fix the shell script and use a different return code.
> * b) Patch the libghdl-....so file name to follow Linux rules until GHDL has fixed it in it's makefiles.
> * c) Create at-least 2 variants of libghdl for mcode and llvm backends.  
>   GCC support will be dropped soon, so no need for libghdl-gcc. Moreover, GCC - like LLVM - creates *.o files, which is back for a Python in memory execution.
> * d) Fix the installation directory of ghdl1-llvm.  
>   See also cpp, which called by gcc and still present in /usr/bin
> * e) Check installation directory of libghdl. If directory needs to be /usr/lib/x86_64-linux-gnu, then you need to patch the relative paths to the pre-compiled sources.  
>   And/or request an update to the make files in GHDL if the current installation directory is wrong.
> * f) Check if all ghdl packages are synchronzed by rules, otherwise an update of ghdl-llvm will break the others due to shared src and vhdl directories.
> * g) Check if you want to offer a python3-pyGHDL package.

Paebbels, 2024-08-04:
@tgingold

* We should think about the libghdl file name, because Linux `*.so` files have this scheme:
  ```
  $ ls -la /usr/lib/libXpm.so*
  lrwxrwxrwx 1 root root    16  3. Jul 21:08 /usr/lib/libXpm.so -> libXpm.so.4.11.0
  lrwxrwxrwx 1 root root    16  3. Jul 21:08 /usr/lib/libXpm.so.4 -> libXpm.so.4.11.0
  -rwxr-xr-x 1 root root 75784  3. Jul 21:08 /usr/lib/libXpm.so.4.11.0
  ```
  Can this be modified and adjusted in make?
* libghdl has no header file. Shouldn't we provide one for C programmers?  
  I also checked my MinGW installation directories, there is no header file.
* I suspect they installed libghdl to `/usr/lib/x86_64-linux-gnu`, because libghdl is based on LLVM and generates x86-64 code as outputs. So this directory is prepared for "cross-compiling".
  If Debian/Ubuntu sticks to this pattern, how can libghdl find it's pre-compiled sources?
  Or should we also install libghdl to that directory instead of plain `/usr/lib`?

tgingold, 2024-08-06:
One way for pylibghdl to find libghdl.so is through the location of the ghdl binary.

I don't think the name of `libghdl.so` needs to be changed.

I also don't think it needs C headers, nobody has requested them.

And finally, its content doesn't really depend on the backend.

doko42, 2024-10-15:
please consider filing a bug report in Debian for any packaging issues. while I am looking at the ghdl Ubuntu packages from time to time, I'm not doing any packaging changes. See https://tracker.debian.org/pkg/ghdl, and https://www.debian.org/Bugs/Reporting for reporting a packaging issue in Debian.

Paebbels, 2024-10-16:
@doko42 thanks for pointing it out. I was not sure if the package is maintained at Debian or at Ubuntu or both in parallel. I know both are *.deb files and use APT and they are very alike, but I never got involved with such packages and packaging before. The bug was reported to Ubuntu (see above: https://github.com/ghdl/ghdl/issues/2710#issuecomment-2267665515).

I didn't report to Debian, because Debian has currently no GHDL. Looking at the link, I see new attempts to add it again. Will the Debian package be identical to the Ubuntu package? Should I test and report it?

Paebbels, 2025-03-15:
I opened a discussing at the Debian mailing list. No results so far.

Now it's up to them to contact me or Tristan if the want to fix the Debian / Ubuntu packaging. I'm not spending my time in installing a full blown Debian incl. GUI + a bug reporting tool to report the findings in a year 202x.

If someone has access to such a system, please feel free to report the finding listed in https://github.com/ghdl/ghdl/issues/2710#issuecomment-2267665515

doko42, 2025-03-16:
which Debian ML? Did you open a bug report?

Paebbels, 2025-03-16:
The mail list was `pkg-electronics-devel@alioth-lists.debian.net`. No bug report opened from my side, because the requirements of Debian are too demanding (setting up a Debian system with GUI and installing a special GUI application). I don't have a Debian desktop system, only Debian CI servers.

If someone reaches out to me, I'm happy to open a Zoom session and explain my findings and discuss solutions with the package maintainers.

Paebbels, 2025-04-28:
@DanielG Three bugs have been reported to `submit@bugs.debian.org` several days ago - no response so far.

DanielG, 2025-04-28:
@Paebbels Excellent.

You're reaching the right audience for what you want to do now, but please be patient Debian works on long timescales and pressuring volunteers is considered very, very rude.

We're in the middle of the release process for Trixie right now and everyone is already busy because of that. Consider that people have many responsibilities inside and outside of Debian -- even DDs have to eat.

--Daniel

Paebbels, 2025-04-28:
@DanielG 

> We're in the middle of the release process for Trixie right now and everyone is already busy because of that.

I understand this, but shouldn't it be in anyone's interest to deliver working GHDL packages rather then broken packages? Such a package for Trixie will be broken in Debian for the next 2 years, right?

DanielG, 2025-04-28:
> Such a package for Trixie will be broken in Debian for the next 2 years, right?

Perhaps, but we do have a process for point releases if the user pain is large enough to generate interest in fixing it, but it's limited in a number of ways.

Unfortunately the timespan to *enure* issues are not present in Trixie was over the last one and a half years. We can and will still try but now we're talking crunchtime -- on Debian timescale anyway ;-).

This would have been much easier a couple months ago.

If you're interested in Debian being high quality in the areas you care about you should consider sticking around to make it so not show up at the last minute <3.

--Daniel