ghdl/ghdl #2711: GHDL not properly compiling on MacOS 14.6 with command ```ghdl -a --ieee=synopsy``` command

TeinkBR, 2024-08-05:
**Description**
I am trying to compile ```.vhd``` file with ```ghdl``` using ```ghdl -a``` command at terminal on MacOS 14.6. I encountered the error ```ghdl:error: installation problem: ghdl1-llvm not found```

**Expected behaviour**
# No output, if successfully compiled with no error in vhdl code
or warnings/errors in the vhdl code.

**How to reproduce?**
```brew install ghdl```
```brew install llvm```
```ghdl -a --ieee=synopsys --work=work example.vhd```

**Context**
Both ```ghdl``` and ```llvm``` are installed with ```brew```, and the system OS is MacOS 14.6
- OS: MacOS 14.6

Paebbels, 2024-08-05:
Is this a macOS on `arm64` or `x86-64`?

Please copy the outputs of these commands into your issue text:
1. `which ghdl`
2. `ghdl version`

------
I found a few days ago that Ubuntu and macOS packages are partially broken.  
See also:
* #2708 
* #2710

TeinkBR, 2024-08-05:
My machine is arm64 

source: ```arch``` command returns ```arm64```
```which ghdl``` gives output ```/usr/local/bin/ghdl```
```ghdl version``` gives output ```GHDL 4.1.0 (4.0.0.r39.g7188e92cf) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm 15.0.7 code generator
Written by Tristan Gingold.```

Paebbels, 2024-08-05:
@TeinkBR wrote in #114 
> I am using Apple M1 macbook pro, i wonder it has anything to do with this fact.

The current cask receipt for GHDL is not correct for ARM-based platforms. See #2708.

**Fast solution:**
clone GHDL and call the following commands:
```
mkdir build
cd build
../configure --prefix=/opt/ghdl --with-llvm-config
make
make install
```
to compile it locally.

**Slow solution:**  
wait until GHDL pipeline compiles more variants for macOS and the homebrew receipts get fixed.

karshPrime, 2024-08-05:
i've been stuck in the same issue, here's a fix I worked out-
```sh
$ ln -s /opt/homebrew/Caskroom/ghdl/4.1.0/bin/ghdl1-llvm /opt/homebrew/bin/ghdl1-llvm 
```

what I’m doing here is manually adding symlink to ghdl1-llvm to a dir that's in path

TeinkBR, 2024-08-05:
```
> mkdir build
> cd build
> ../configure --prefix=/opt/ghdl --with-llvm-config
> make
> make install
> ```
> 
```

When I tried this line ```> ../configure --prefix=/opt/ghdl --with-llvm-config ``` , the terminal returned ```Sorry, you need GNAT to build GHDL.  See the README```

From what I did a bit of search on the internet, to download and install GNAT on MacOS seems to be quite cumbersome...

Paebbels, 2024-08-05:
@karshPrime  
Can you analyze and run the design? I found that ghdl installed by homebrew is a x86-64 binary, but the system and llvm.dylib is arm64 (aarch64).

@TeinkBR  
I noticed this myself 60 minutes ago, sorry. GHDL has a [shell script](https://github.com/ghdl/ghdl/blob/master/scripts/macosx/install-ada.sh#L23) to "install" GNAT by downloading a precompiled variant from here: [GNAT 14.1.0-3](https://github.com/alire-project/GNAT-FSF-builds/releases/tag/gnat-14.1.0-3).

--------------
I'm currently checking how to improve that code for GitHub Actions.

TeinkBR, 2024-08-05:
The previous issue seem to have resolved, but now when i run ```ghdl -a State_Estimation.vhd``` it returns an error :
```dyld[32221]: Library not loaded: /usr/local/opt/llvm@15/lib/libLLVM.dylib
  Referenced from: <3CC36E74-F6A6-3EF6-95CD-7BD3ECE10FBC> /opt/homebrew/Caskroom/ghdl/4.1.0/bin/ghdl1-llvm
  Reason: tried: '/usr/local/opt/llvm@15/lib/libLLVM.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/llvm@15/lib/libLLVM.dylib' (no such file), '/usr/local/opt/llvm@15/lib/libLLVM.dylib' (no such file), '/usr/local/lib/libLLVM.dylib' (no such file), '/usr/lib/libLLVM.dylib' (no such file, not in dyld cache)
ghdl:error: exec error```,  this seems to indicate that there is discrepancies of my path where I installed my ```llvm```, I tried  the following ```# Uninstall the current LLVM version
brew uninstall llvm

# Install the correct version of LLVM
brew install llvm@15

# Verify the installation
if [ -f "/usr/local/opt/llvm@15/lib/libLLVM.dylib" ]; then
  echo "LLVM library found."
else
  echo "LLVM library not found. Installation failed."
  exit 1
fi

# Add paths to .zshrc or .bashrc
echo 'export PATH="/usr/local/opt/llvm@15/bin:$PATH"' >> ~/.zshrc
echo 'export LDFLAGS="-L/usr/local/opt/llvm@15/lib"' >> ~/.zshrc
echo 'export CPPFLAGS="-I/usr/local/opt/llvm@15/include"' >> ~/.zshrc
echo 'export DYLD_LIBRARY_PATH="/usr/local/opt/llvm@15/lib:$DYLD_LIBRARY_PATH"' >> ~/.zshrc

# Apply the changes
source ~/.zshrc

# Verify GHDL installation
ghdl --version
```
and it doesn't seem to work, now I am trying to directly clone from ```llvm-project``` git repository and build it:
```
# Clone the LLVM repository
git clone https://github.com/llvm/llvm-project.git
cd llvm-project

# Create a build directory
mkdir build
cd build

# Configure the build
cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local/llvm ../llvm

# Build and install LLVM
make
sudo make install

# Add paths to .zshrc
echo 'export PATH="/usr/local/llvm/bin:$PATH"' >> ~/.zshrc
echo 'export LDFLAGS="-L/usr/local/llvm/lib"' >> ~/.zshrc
echo 'export CPPFLAGS="-I/usr/local/llvm/include"' >> ~/.zshrc
echo 'export DYLD_LIBRARY_PATH="/usr/local/llvm/lib:$DYLD_LIBRARY_PATH"' >> ~/.zshrc

# Apply the changes
source ~/.zshrc

# Verify the installation
llvm-config --version
```

karshPrime, 2024-08-05:
interesting you got it done that way @TeinkBR 

here's when I tried to build the project by just

```sh
$ ../configure --prefix=/opt/ghdl --with-llvm-config

Apple clang version 15.0.0 (clang-1500.3.9.4)
Target: arm64-apple-darwin23.5.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
Use full IEEE library
Build machine is: arm64-apple-darwin23.5.0
Debugging is enabled with llvm 18.1.8
create pic/ subdirectory
Creating Makefile
Creating default_paths.ads
Creating ghdl.gpr
Creating scripts/gcc/Make-lang.in
for d in ieee/v87 ieee/v93 ieee/v08 std/v87 std/v93 std/v08 src/ieee src/ieee/v87 src/ieee/v93 src/ieee2008 src/std src/std/v87 src/std/v93 src/std/v08 src/synopsys src/synopsys/v08 src/upf src/vital95 src/vital2000; do \
          mkdir -p lib/ghdl/$d; \
        done
Generate ghdlsynth_maybe.ads
Generate grt-readline.ads
```

Here I did not clone or build llvm to get things done, instead I found it somewhere in homebrew already n made sure to add it to path

[image]
\* `~/Project/.bin` is a local repo I use to add my own programs to path, but used for this purpose here.

it failed, as the logs show above, with no clear error code. so I guess I should do what you did.

karshPrime, 2024-08-06:
hmm I cloned and sudo make install llvm, installed gnat, made sure they were all on path, yet i'm having this issue-
```sh
$ ../configure --prefix=/opt/ghdl --with-llvm-config

Apple clang version 15.0.0 (clang-1500.3.9.4)
Target: arm64-apple-darwin23.5.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
Use full IEEE library
Build machine is: arm64-apple-darwin23.5.0
Unhandled version llvm 20.0.0git
```

karshPrime, 2024-08-06:
attempt #lostCount

I've cloned and manually built it on my system, confirming its in path. tried to do the same with ghdl, failed, as well, I’m not really sure of the build instructions. steps mentioned above by @Paebbels dont seem to work given there's no makefile in ./build/.

anyways, instead I just downloaded compiled binary from releases and added that to path instead, to see if that'd help in any way.

good news: no llvm or build fail error
bad news: still no working setup

current error:
```sh
$ ghdl -a ./Multiplier1Cycle.vhd

dyld[97505]: Library not loaded: /usr/local/opt/llvm@15/lib/libLLVM.dylib
  Referenced from: <3CC36E74-F6A6-3EF6-95CD-7BD3ECE10FBC> /Users/alias/Downloads/ghdl-macos-11-llvm/bin/ghdl1-llvm
  Reason: tried: '/usr/local/llvm/lib/libLLVM.dylib' (no such file), '/libLLVM.dylib' (no such file), '/usr/local/opt/llvm@15/lib/libLLVM.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/llvm@15/lib/libLLVM.dylib' (no such file), '/usr/local/opt/llvm@15/lib/libLLVM.dylib' (no such file), '/usr/local/lib/libLLVM.dylib' (no such file), '/usr/lib/libLLVM.dylib' (no such file, not in dyld cache)
ghdl:error: exec error
```

diogratia, 2024-08-06:
There's a nightly ghdl binary distribution of a standalone [ghdl-macos14-ARM64-llvm-jit.tgz](https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-macos14-ARM64-llvm-jit.tgz) that should run for MacOS 13 - 15. Like the mcode version it produces no object code files, requiring elaboration command line arguments be passed during -r simulation commands.

It has no dynamic library dependencies not met by installing Apple's Command Line Tools. All llvm libraries are statically linked.

Using any of the other pre built ghdl binary llvm distributions for MacOS that give object code file access depend on finding libLLVM.so dynamic libraries in a particular location matching the build's installation of homebrew and may require a specific llvm version be installed due to API differences. There can also be a libgcc dynamic library rpath dependency.

I don't think I captured the [ghdl-macos14-ARM64-llvm.tgz](https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-macos14-ARM64-llvm.tgz) build log from the Jun 30 (may be international date line dependent) - the last nightly build but installing homebrew with the specified llvm cask in the same location as the build as well as the same gnat in the same location as the build should be guaranteed to satisfy limiting build rpath restrictions.

A properly constructed homebrew ghdl cask ghdl.rb pointing to a hombrew prebuilt distribution should be possible where the particular version of llvm would be a prerequisite of a ghdl cask, along with the particular gnat as yet not supported in homebrew which could benefit from a gnat/ghdl knowledgeable maintainer for both casks. Herein lies your original homebrew installation problem. 

In the meantime the build process used to produce nightly or released distribution archives can be duplicated using information found in the ghdl github tree. It should also be possible to modify the build to use statically linked libraries and avoid version dependencies. The big libLLVM dynamically loaded library should add around 45 MB or a bit more when statically linked and the libgcc around 2 MB. The larger size binary is offset by not requiring hombrew/gnat installations (approximately 2.6 GB). The Command Line Tools installation can't be evaded, supplying the linking loader and various execution frameworks as well as access to a build environment.

And lastly we should be aware of the difference in -2008 feature coverage in the mcode version which would require a volunteer to add ARM64 ABI support. The mcode version is inherently standalone.

tgingold, 2024-08-06:
There is no version yet for mac os arm 64.
But as @diogratia said, there is a nightly build.

TeinkBR, 2024-08-06:
> attempt #lostCount
> 
> I've cloned and manually built it on my system, confirming its in path. tried to do the same with ghdl, failed, as well, I’m not really sure of the build instructions. steps mentioned above by @Paebbels dont seem to work given there's no makefile in ./build/.
> 
> anyways, instead I just downloaded compiled binary from releases and added that to path instead, to see if that'd help in any way.
> 
> good news: no llvm or build fail error bad news: still no working setup
> 
> current error:
> 
> ```shell
> $ ghdl -a ./Multiplier1Cycle.vhd
> 
> dyld[97505]: Library not loaded: /usr/local/opt/llvm@15/lib/libLLVM.dylib
>   Referenced from: <3CC36E74-F6A6-3EF6-95CD-7BD3ECE10FBC> /Users/alias/Downloads/ghdl-macos-11-llvm/bin/ghdl1-llvm
>   Reason: tried: '/usr/local/llvm/lib/libLLVM.dylib' (no such file), '/libLLVM.dylib' (no such file), '/usr/local/opt/llvm@15/lib/libLLVM.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/llvm@15/lib/libLLVM.dylib' (no such file), '/usr/local/opt/llvm@15/lib/libLLVM.dylib' (no such file), '/usr/local/lib/libLLVM.dylib' (no such file), '/usr/lib/libLLVM.dylib' (no such file, not in dyld cache)
> ghdl:error: exec error
> ```

I have encountered a similar error with my ```GHDL``` also not able to locate ```libLLVM.dylib```, I tried to git clone from ```LLVM-project``` repository and build directly from there, then tried to rebuilt my GHDL from the cloned repository again with the ```LLVM``` backend :
# Remove the previous GHDL installation
sudo rm -rf /opt/ghdl

# Navigate to the directory where you cloned the GHDL repository
cd /path/to/ghdl

# Clean the build directory
rm -rf build
mkdir build
cd build

# Configure GHDL with LLVM support
../configure --prefix=/opt/ghdl --with-llvm-config=/usr/local/llvm/bin/llvm-config

# Build and install GHDL
make
sudo make install

# Add paths to .zshrc
echo 'export PATH="/opt/ghdl/bin:$PATH"' >> ~/.zshrc
echo 'export LDFLAGS="-L/usr/local/llvm/lib"' >> ~/.zshrc
echo 'export CPPFLAGS="-I/usr/local/llvm/include"' >> ~/.zshrc
echo 'export DYLD_LIBRARY_PATH="/usr/local/llvm/lib:$DYLD_LIBRARY_PATH"' >> ~/.zshrc

# Apply the changes
source ~/.zshrc

# Verify the GHDL installation
ghdl --version
```
but the same error still persists:
```
dyld[35498]: Library not loaded: /opt/homebrew/opt/llvm/lib/libLLVM.dylib
  Referenced from: <0B19825F-5392-3BCA-AE66-0F63E39F98DB> /Users/jingyili/Downloads/ghdl/build/ghdl1-llvm
  Reason: tried: '/opt/homebrew/opt/llvm/lib/libLLVM.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/llvm/lib/libLLVM.dylib' (no such file), '/opt/homebrew/opt/llvm/lib/libLLVM.dylib' (no such file)
```
seems like this doesn't work also, maybe i should try to build ```ghdl``` from the downloaded nigthly built```ghdl-macos14-ARM64-llvm-jit.tgz```, as mentioned by @diogratia , i will try it a bit later and share the result.

Paebbels, 2024-08-06:
@karshPrime 

> I’m not really sure of the build instructions. steps mentioned above by @Paebbels dont seem to work given there's no makefile in ./build/.

You forgot to run `../configure` or you cleaned the build directory, so you need to rerun `configure`.

Configure writes a `Makefile` into the current working directory. That's why I created `build/` and moved the working directory into `build/`. Also I referenced configure with `../`.

Look at your comment: https://github.com/ghdl/ghdl/issues/2711#issuecomment-2270064505, here it reported a `Makefile` was created. Together with some other files.

TeinkBR, 2024-08-06:
[Problem Solved] 
I downloaded the nightly ghdl binary distribution of a standalone [ghdl-macos14-ARM64-llvm-jit.tgz](https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-macos14-ARM64-llvm-jit.tgz) and moved the binary files to the local directories:
```
# Download the Nightly GHDL Binary
wget https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-macos14-ARM64-llvm-jit.tar.gz -O ghdl-macos14-ARM64-llvm-jit.tar.gz

# Extract the Archive
tar -xvf ghdl-macos14-ARM64-llvm-jit.tar.gz
cd ghdl-macos14-ARM64-llvm-jit

# Verify the contents of the directories
ls -l bin/
ls -l lib/
ls -l include/

# If directories are not empty, move the files
if [ -d "bin" ] && [ "$(ls -A bin)" ]; then
  sudo rsync -av bin/ /usr/local/bin/
else
  echo "bin directory is empty or does not exist."
fi

if [ -d "lib" ] && [ "$(ls -A lib)" ]; then
  sudo rsync -av lib/ /usr/local/lib/
else
  echo "lib directory is empty or does not exist."
fi

if [ -d "include" ] && [ "$(ls -A include)" ]; then
  sudo rsync -av include/ /usr/local/include/
else
  echo "include directory is empty or does not exist."
fi

# Set environment variables
echo 'export PATH="/usr/local/bin:$PATH"' >> ~/.zshrc
echo 'export LDFLAGS="-L/usr/local/lib"' >> ~/.zshrc
echo 'export CPPFLAGS="-I/usr/local/include"' >> ~/.zshrc
echo 'export DYLD_LIBRARY_PATH="/usr/local/lib:$DYLD_LIBRARY_PATH"' >> ~/.zshrc

# Apply the changes
source ~/.zshrc

# Verify the GHDL installation
ghdl --version
```
and there is no error ```dyld[35498]: Library not loaded: /opt/homebrew/opt/llvm/lib/libLLVM.dylib``` anymore.

Paebbels, 2024-08-06:
@TeinkBR you found a workaround, but the problem isn't solved yet.

diogratia, 2024-08-07:
See Reddit [r/VHDL Issues with GHDL running with LLVM backend, not being able to locate ```libLLVM.dylib```](https://www.reddit.com/r/VHDL/comments/1elixcv/issues_with_ghdl_running_with_llvm_backend_not/) which reveals sthe issue is non solely confined to hombrew and may be more about communications and documentation.

sm13, 2024-09-02:
Hello,

I'm far from a homebrew expert but I've managed to install the nightly version (from the binary artifacts) adapting the homebrew cask recipe. I'm sharing this in the hopes that this can shed some light for someone.

The main changes to the recipe from the one currently provided for version 4.1.0 are:

- Modified the `arch` method values to match the artifacts from the nightly build, the url and the sha sum etc.
- Added the apparently missing bin that was not listed before `ghdl1-llvm`
- Added a symlink to the `libgcc_s.1.1.dylib`, which my system fails to find it because it is expecting it in the `bin` folder, rather than the `lib`, I don't know if this is an issue in the build system or my own system.
- Added the dependency on llvm
- NOTE: this recipe is only valid for an aarch64 macos Sonoma and for ghdl running the LLVM backend!

Here's the listing.

```ruby
cask "ghdl" do
  version "4.2.0"

  arch arm: "ARM64"
  sha256 "3cd01ca86bcd6bb9ec0bdb086af1468c3c4f7a170fd003a879ffcbd567036c6b"
  url "https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-macos14-#{arch}-llvm.tgz"
  name "ghdl"
  desc "VHDL 2019/2008/93/87 simulator"
  homepage "https://github.com/ghdl/ghdl/"

  livecheck do
    url :url
    strategy :github_latest
  end

  depends_on formula: "llvm"
  ghdlbins = ["ghdl", "ghwdump", "ghdl1-llvm"]

  ghdlbins.each { |_bin|
    binary "bin/#{_bin}"
  }

  postflight do
    puts "Creating symlink to libgcc_s.1.1.dylib"
    File.symlink("#{HOMEBREW_PREFIX}/lib/gcc/current/libgcc_s.1.1.dylib", "#{staged_path}/bin/libgcc_s.1.1.dylib")
    puts "Creating library symlinks in #{HOMEBREW_PREFIX}/include and #{HOMEBREW_PREFIX}/lib"
    File.symlink("#{staged_path}/include/ghdl", "#{HOMEBREW_PREFIX}/include/ghdl")
    File.symlink("#{staged_path}/lib/ghdl", "#{HOMEBREW_PREFIX}/lib/ghdl")
    puts "Setting files as being from a nice developer"
    ghdlbins.each { |_bin|
      `xattr -dr com.apple.quarantine #{HOMEBREW_PREFIX}/bin/#{_bin}`
    }
  end

  uninstall_postflight do
    puts "Removing symlink to libgcc_s.1.1.dylib"
    File.unlink("#{staged_path}/bin/libgcc_s.1.1.dylib")
    puts "Removing library symlinks in #{HOMEBREW_PREFIX}/include and #{HOMEBREW_PREFIX}/lib"
    File.unlink("#{HOMEBREW_PREFIX}/include/ghdl", "#{HOMEBREW_PREFIX}/lib/ghdl")
  end

  # No zap stanza required
end
```

If someone wants to test it:

```bash
- brew create --cask ghdl https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-macos14-ARM64-llvm.tgz
- brew install --cask ghdl /opt/homebrew/Library/Taps/homebrew/homebrew-cask/Casks/g/ghdl.rb
```

Final thoughts, as I wrote this I think it seems that the error messages point to versions of LLVM which are not installed in the default path for homebrew on aarch64 macos. If a user moved from an intel-based mac to an arm-based mac, they may find programs and libraries installed in various locations and for different targets. Perhaps this is their issue?

tgingold, 2024-09-11:
Thank you for this effort, I hope it will help people.

AdrianLuchaco, 2024-10-08:
Hi, I have been reading this entire discussion and it has not been clear to me what the ideal solution is so that I do not get the llvm error that appears or did appear to the creator of this discussion. My Mac is an M3 Sonoma.
When I put ghdl --verison in the terminal, it says that I have the compiler correctly installed and also the llvm, but when I do ghdl -a... I get the llvm error. 
Thank you so much.

tgingold, 2024-10-08:
@AdrianLuchaco, as there are nightly builds for macos/ARM, we recommend to use them.

logansommerhoff, 2024-10-25:
Hey all, I've been trying to get my install to work for a few days now and hit a wall. After following @sm13 's instructions (I removed `ghdl` from the create command to stop homebrew from throwing a syntax error) finally fixed all the errors that've been discussed in the thread, attempting to analyze the hello world test presented a new error that I haven't seen anywhere as of yet:
`$ ghdl -a hello.vhdl`

`dyld[41948]: Symbol not found: __ZN4llvm11PassBuilder20crossRegisterProxiesERNS_15AnalysisManagerINS_4LoopEJRNS_27LoopStandardAnalysisResultsEEEERNS1_INS_8FunctionEJEEERNS1_INS_13LazyCallGraph3SCCEJRSA_EEERNS1_INS_6ModuleEJEEE`
`Referenced from: <656EB84A-6E81-32AC-AC51-0C6D1874B101> /opt/homebrew/Caskroom/ghdl/4.2.0/bin/ghdl1-llvm`
`Expected in:     <D34D5573-504D-38E6-9857-26B95A3E86E5> /opt/homebrew/Cellar/llvm/19.1.2/lib/libLLVM.dylib`

The closest thing I was able to find said something about fragments from attempting to build GHDL multiple times which could definitely be the case given how many times I've tried and failed, but fixing it on my own seems entirely out of my depth. I'm just trying to set up TerosHDL so I can practice verilog for my digital logic class; this debacle has honestly taught me more about unix than our actual unix course and I still don't even know where to start on an error like that.

tgingold, 2024-10-26:
Did you try to download and run the nightly build of GHDL ?  It is probably the easiest way.

sm13, 2024-10-26:
Note that the brew recipe I provided links to the nightly build. Perhaps at some point it will need the `sha256` updated when new binaries are provided.

@logansommerhoff the good news is that I was able to reproduce the issue on my side. The bad news is that I don't have a _good_ fix for you. The workaround at the moment is to use LLVM@18 instead of LLVM@19, of course this will have repercussions beyond GHDL. It really should be possible to provide a brew recipe that actually builds and links from source code, and thus will force to use LLVM < 19. I tried, but I wasn't able to.

I'm trying to find a better workaround than reverting LLVM versions. If I find any I'll post an update.

sm13, 2024-10-26:
I found a solution, although it cannot come with enough disclaimers:

- I barely know what I'm doing
- I have not tested this beyond compiling a single file
- I don't know if other GHDL binaries may be affected by this change, or may need this change, etc.
- etc.

Relinking the `ghdl1-llvm` binary to LLVM@18 will permit you to have LLVM updated to the latest in your brew installation at the expense of having yet another LLVM version installed.

The additional commands:

```
install_name_tool -change #{HOMEBREW_PREFIX}/opt/llvm/lib/libLLVM.dylib \
      ${HOMEBREW_PREFIX}/opt/llvm@18/lib/libLLVM.dylib \
      ${HOMEBREW_PREFIX}/Caskroom/ghdl/4.2.0/bin/ghdl1-llvm
```
I have edited my original comment to include it.

CC: @logansommerhoff

logansommerhoff, 2024-10-27:
`Hello world!`
What a journey, thanks a ton for the help @sm13 and everyone else who contributed to the thread. To make my own small contribution, here are the steps I took to get a working pipeline from a relatively fresh homebrew install:

**DISCLAIMER:** I have _far, far less_ of an idea what I'm doing than anyone else here and probably you yourself. If **any** of their solutions worked for you out of the box or you're proficient enough to try the methods which I couldn't, I really cannot stress enough how much better of an idea it would be for you to go with one of them instead.

```zsh
## Making sure you're in the default directory (sanity check, almost certainly doesn't matter)
$ cd

## Getting homebrew ready to run custom formulae [Source 1]
$ export HOMEBREW_NO_INSTALL_FROM_API=1
$ echo $HOMEBREW_NO_INSTALL_FROM_API      ## Sanity check, should return "1"
$ brew tap --force homebrew/cask

## Switching out the broken formula for sm13's
$ brew edit --cask ghdl  
   ## Following instructions assume you haven't changed homebrew's default editor from VIM:
      ## 1. Type "ggVG" to select the whole file
      ## 2. Copy and paste sm13's code block [Source 2]
      ## 3. Type ":wq" (write and quit)
      ## 4. Hit enter

$ brew install ghdl

## Done! Here's how I tested it if you'd like to do the same
$ vim hello.vhdl
   ## 1. Copy and paste the code block from GHDL's Quick Start Guide [Source 3]
   ## 2. Type ":wq"
   ## 3. Hit enter

$ ghdl -a hello.vhdl
$ ghdl -e hello.vhdl
$ ghdl -r hello.vhdl      ## Should return "Hello world!"

## Cleanup
$ rm hello.vhdl
$ rm hello.o
$ rm hello_world
$ rm e~hello_world.o 
$ rm work-obj93.cf
```

Sources
1. [Homebrew FAQ "Can I edit formulae myself?"](https://docs.brew.sh/FAQ#can-i-edit-formulae-myself)
2. [sm13's revised GHDL brew formula](https://github.com/ghdl/ghdl/issues/2711#issuecomment-2324563920)
3. [GHDL Hello world program](https://ghdl.github.io/ghdl/quick_start/simulation/hello/index.html)

Some Notes
1. You don't need to manually `brew install llvm@18`, the updated formula will do that for you.
2. You'll notice that my commands to install GHDL using sm13's formula are a bit different than their own. While I can't say for certain why those commands didn't work in my case, the method I used comes straight from Homebrew's documentation on running custom formulae [Source 1].
3. The `HOMEBREW_NO_INSTALL_FROM_API` value is specific to the terminal session. If you terminate that session, you'll need to set it again before running `install` or else the vanilla formula will be used.
5. I'm running an M3 on OSX 15 Sequoia. If that's you, fantastic. If not, YMMV.

Paebbels, 2024-11-28:
**Update:**  
GHDL has now the new build pipeline and also a new mechanism for nightly releases. This will be triggered soon. As casks are referring to assets in the release/nightly release, we're now able to update the casks recipes.