ghdl/ghdl #2712: Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Covered_Flag

ru551n, 2024-08-07:
**Description**
Exception when running VUnit testbenches.

**Expected behaviour**
No crash

**How to reproduce?**
Currently unsure what exactly causes the issue, investigation ongoing.

**Context**
- OS: Almalinux 8.8 podman image running on Almalinux 9.3
- Origin:
  - Built from sources: `3b5854759c14a50dbaefd1cc48f9be4606f600d6`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

This error has occured on an air gapped machine (not connected to the internet), so I will do my best to paste the info I can here

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r102.g3b5854759) [Dunoon edition]
Compiled with GNAT Version: 8.5.0 20210514 (Red Hat 8.5.0-23
Target: x86_64-redhat-linux
Command Line:
/usr/local/ghdl/bin/ghdl --elab-run --std=08 --work=XXX --workdir=/usr/local/repo/build/sim/vunit_out_questa/ghdl/libraries/XXX -P/usr/local... (too much to type)
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE RAISED
Exception information:
raised SYSTE.ASSERTIONS.ASSERT_FAILURE : no field Covered_Flag
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

ru551n, 2024-08-07:
If it helps, this error occurs in every test case and in every testbench we have.

ru551n, 2024-08-07:
I did some more digging, and it turns out that the issue was using `--coverage` option. 

So the issue here boils down to coverage not working on Linux with mcode backend.

tgingold, 2024-08-07:
It would be nice to have a reproducer.  And ideally a reproducer without VUnit.
There are a few tests about `--coverage` in the testsuite, that pass on linux with the mcode backend.
So there is a particular scenario.