ghdl/ghdl #2713: Synthesis crash: Constraint Error

JJ-Gaisler, 2024-08-08:
**Description**
When trying to synthesize a large design into verilog I run into a GHDL crash. The design I am synthesizing is both complex and large. Synthesis up until the crash used about 85 GiB of memory.

**How to reproduce?**
Unfortunately the design is proprietary and thus I can't provide an reproducible. I have provided a back trace from when ada throws the exception. Maybe the issue is as simple as having to small data types for this large design?

**Context**
Please, provide the following information:

- OS: Linux
- Origin:
  - [x] Built from sources: `9567e46f7df97b7d04979f8f6036a825edf879cb`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r141.g9567e46f7) [Dunoon edition]
Compiled with GNAT Version: Community 2021 (20210519-103)
Target: x86_64-pc-linux-gnu
Command line:
ghdl --synth --out=verilog -fexplicit --ieee=synopsys --mb-comments --warn-no-binding --workdir=gnu/work --work=work -Pgnu -Pgnu/grlib -Pgnu/dware -Pgnu/synplify -Pgnu/techmap -Pgnu/spw -Pgnu/spfi -Pgnu/wizl -Pgnu/eth -Pgnu/opencores -Pgnu/gaisler -Pgnu/esa -Pgnu/bsc -Pgnu/cypress -Pgnu/micron -Pgnu/hynix -Pgnu/tmtc --warn-no-hide --warn-no-specs --warn-no-port design
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : dyn_tables.adb:138 range check failed
Call stack traceback locations:
0x43c241 0x440b3b 0x44857c 0x44aa25 0x44d9d3 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9f2 0x44d9d3 0x44d9f2 0x44d9f2
```
```
#0  <__gnat_debug_raise_exception> (e=0xa30c00 <constraint_error>, message=...) at s-excdeb.adb:43                                                                                                                                                                                                                           
#1  0x0000000000668579 in ada.exceptions.complete_occurrence (x=x@entry=0xa54210) at a-except.adb:929                                                                                                                                                                                                                        
#2  0x0000000000668589 in ada.exceptions.complete_and_propagate_occurrence (x=x@entry=0xa54210) at a-except.adb:940                                                                                                                                                                                                          
#3  0x0000000000668976 in ada.exceptions.raise_with_location_and_msg (e=0xa30c00 <constraint_error>, f=f@entry=(system.address) 0x6cc9e0, l=l@entry=138, c=c@entry=0, m=m@entry=(system.address) 0x7897c0) at a-except.adb:1151                                                                                              
#4  0x000000000066854c in <__gnat_raise_constraint_error_msg> (file=file@entry=0x6cc9e0, line=line@entry=138, column=column@entry=0, msg=msg@entry=0x7897c0 <ada.exceptions.rmsg_12>) at a-except.adb:895              
#5  0x0000000000668ad9 in <__gnat_rcheck_CE_Range_Check> (file=file@entry=0x6cc9e0, line=line@entry=138) at a-except.adb:1268                                                                                                                                                                                                
#6  0x000000000043c243 in netlists.snames_table.dyn_table.last (t=...) at ../../../gcc-12.2.0/gcc/vhdl/dyn_tables.adb:138                                                                                                                                                                                                    
#7  0x0000000000440b3d in netlists.snames_table.dyn_table.last (t=...) at ../../../gcc-12.2.0/gcc/vhdl/dyn_tables.adb:161                                                                                                                                                                                                    
#8  netlists.snames_table.dyn_table.append (val=..., t=...) at ../../../gcc-12.2.0/gcc/vhdl/dyn_tables.adb:161                                                                                                                                                                                                               
#9  netlists.snames_table.append (val=...) at ../../../gcc-12.2.0/gcc/vhdl/tables.adb:65                                                                                                                                                                                                                                     
#10 netlists.new_sname_version (ver=1073717928, prefix=prefix@entry=0) at /scratch/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/netlists.adb:56                                                                                                                                                                                 
#11 0x000000000044857e in netlists.builders.new_internal_name (prefix=0, ctxt=<optimized out>) at /scratch/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/netlists-builders.adb:801                                                                                                                                               
#12 netlists.builders.new_internal_instance (ctxt=0x45bb2c0, m=51) at /scratch/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/netlists-builders.adb:811                                                                                                                                                                           
#13 0x000000000044aa27 in netlists.builders.build_extract (ctxt=ctxt@entry=0x45bb2c0, i=341769, off=11, w=w@entry=1) at /scratch/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/netlists-builders.adb:1662                                                                                                                        
#14 0x000000000044d9b2 in netlists.folds.build2_extract_push (ctxt=ctxt@entry=0x45bb2c0, i=<optimized out>, off=off@entry=11, w=w@entry=1) at /scratch/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/netlists-folds.adb:306                                                                                                      
#15 0x000000000044d9d5 in netlists.folds.build2_extract_push (ctxt=ctxt@entry=0x45bb2c0, i=<optimized out>, off=off@entry=11, w=w@entry=1) at /scratch/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/netlists-folds.adb:298                                                                                                      
#16 0x000000000044d9f4 in netlists.folds.build2_extract_push (ctxt=ctxt@entry=0x45bb2c0, i=<optimized out>, off=off@entry=11, w=w@entry=1) at /scratch/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/netlists-folds.adb:298                                                                                                      
#17 0x000000000044d9f4 in netlists.folds.build2_extract_push (ctxt=ctxt@entry=0x45bb2c0, i=<optimized out>, off=off@entry=11, w=w@entry=1) at /scratch/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/netlists-folds.adb:298                                                                                                      
#18 0x000000000044d9f4 in netlists.folds.build2_extract_push (ctxt=ctxt@entry=0x45bb2c0, i=<optimized out>, off=off@entry=11, w=w@entry=1) at /scratch/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/netlists-folds.adb:298                                                                                                      
#19 0x000000000044d9f4 in netlists.folds.build2_extract_push (ctxt=ctxt@entry=0x45bb2c0, i=<optimized out>, off=off@entry=11, w=w@entry=1) at /scratch/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/netlists-folds.adb:298                                                                                                      
#20 0x000000000044d9f4 in netlists.folds.build2_extract_push (ctxt=ctxt@entry=0x45bb2c0, i=<optimized out>, off=off@entry=11, w=w@entry=1) at /scratch/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/netlists-folds.adb:298                                                             [... cut ...]
#134 0x0000000000633c8b in ghdlmain.main () at /scratch/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/ghdldrv/ghdlmain.adb:679
#135 0x000000000065f287 in ghdl_gcc () at /scratch/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/ghdldrv/ghdl_gcc.adb:43
#136 0x000000000040a036 in main (argc=<optimized out>, argv=<optimized out>, envp=<optimized out>) at /scratch/ghdl-tmp/ghdl_sources/build/gcc-objs/gcc/b~ghdl_gcc.adb:976
#137 0x00007ffff7c9724d in __libc_start_main () from /lib64/libc.so.6
#138 0x000000000040a08a in _start () at ../sysdeps/x86_64/start.S:120

JJ-Gaisler, 2024-08-08:
I modified the last function slightly to see what the values are in the last function:
  ```
   function Last (T : Instance) return Table_Index_Type is
     tmp : Table_Index_Type;
   begin
      tmp := Table_Index_Type'Val
        (Table_Index_Type'Pos (Table_Low_Bound)
           + Unsigned'Pos (T.Priv.Last_Pos) - 1);
      return tmp;
      exception
        when constraint_error =>
          Ada.Text_IO.Put_Line("error: ");
          Ada.Text_IO.Put_Line("Table_Index_Type'pos (Table_Low_Bound) = " & Unsigned'image(Table_Index_Type'pos (Table_Low_Bound) ));
          Ada.Text_IO.Put_Line( "Unsigned'Pos(T.Priv.Last_Pos) - 1: " & Unsigned'image(Unsigned'Pos (T.Priv.Last_Pos) - 1));
   end Last;

```
Without knowing too much about ada it looks like an overflow issue, which may be due to the size of the design?
```
error:
Table_Index_Type'pos (Table_Low_Bound) =  0
Unsigned'Pos(T.Priv.Last_Pos) - 1:  1073741824
```

tgingold, 2024-08-09:
It looks like you have more than 1e9 nets.  Does this correspond to the size of the design ?

JJ-Gaisler, 2024-08-09:
It does sound like a lot. I don't really know if a GDHL net is equivalent to what other tools calls a net since the netlist isn't optimized. I have seen prior synthesis results from other tools end up several magnitudes smaller than what I am currently seeing from ghdl. If I run one of the CPU's through ghdl synth with a complete subsystem I typically end up with ~5e5 nets. The design I'm currently synthesizing is larger and I would expect that a magnitude larger would be in the reasonable range. Could it be that some construct has gone recursive and that ghdl fails to recognize the exit condition. I looked at the backtrace and without knowing much about it, the following function seems to be called many times recursively netlists.folds.build2_extract_push. I will try moving down the hierarchy of the design to see where things start to work again.

JJ-Gaisler, 2024-08-09:
I have managed to find the problematic code and I can finally synthesize. I have prepared a reproducible and hopefully I am allowed to share the problematic code next week.

Furthermore, I also noticed an issue with namespace collisions where a wire would be declared multiple times. I'll look into the issue further and I'll try to create a reproducible for this too.

Another issue I found is that ghdl sometimes directly translates vhdl signal names into verilog signal names. This is an issue if a signal has the name 'do, break, etc' then the verilog wire will at least in some cases get the same name. Maybe some obfuscation can be done if a verilog keyword is detected?

tgingold, 2024-08-10:
If you can provide a reproducer, that would be very useful.

Yes, there can be collisions (this was already reported).  You can use yosys (and the ghdl plugin) to avoid that.  I need to fix that.

The verilog keywords should be automatically escaped.  But `do` or `break` are new SV keywords I think.

JJ-Gaisler, 2024-08-12:
See the attached file for a reproducible. I can't say for sure if this is exactly the same issue since I can't get GHDL to reach the constraint error any more, however I can see that GHDL is stuck with a relatively high memory usage and a high cpu load without really getting anywhere. Let's start here and see if solving this issue resolves the other issues I have had with the design exploding in size. Commenting the call to the fillin function makes the design synthesize.

[reprod.txt](https://github.com/user-attachments/files/16581364/reprod.txt)

Something unrelated: With the ghdl + yosys plugin I am seeing the same verilog module being imported twice in Yosys. Is this a GHDL issue or is this a Yosys issue? I am wondering so I can open an issue in the right project.

tgingold, 2024-08-14:
I am able to reproduce the problem.  I have further reduced the reproducer, but I am still investigating.

tgingold, 2024-08-14:
For the yosys issue, what commands are you using ?

closed 2024-08-14 by tgingold

tgingold, 2024-08-17:
@JJ-Gaisler Is the original issue also fixed ?

JJ-Gaisler, 2024-08-19:
> @JJ-Gaisler Is the original issue also fixed ?

Sorry for not getting back to you. I have tried the patch and it does indeed solve the original issue so thank you for that!
However, I am getting a similar issue in another file where ghdl allocates a bunch of memory and then crashes. I have been working on a reproducible for that as well, but the ip block is several thousands of lines long with packages from all over grlib so it will probably take some time to get down to a mwe.

tgingold, 2024-08-25:
Thank you for the status update.

Note: you don't really need to reduce the reproducer.  A large one is OK, as usually once the issue is understood, it is easy to reduce it.

JJ-Gaisler, 2024-08-26:
> Thank you for the status update.
> 

Hello! @tgingold  It looks like commit: 488e8addb25b486c382f18902051157c87f333ef broke the Verilog synthesis functionally. When I tested the fix I only checked that synthesis no longer crashed, I had other issues in my design which kept me from testing the functionality. However I have fixed those issues now and I did a git bisect from a version of ghdl I knew was working. This led me to the aforementioned commit.

It simply looks like after this commit my design no longer works. 

> Note: you don't really need to reduce the reproducer. A large one is OK, as usually once the issue is understood, it is easy to reduce it.

The reason for reducing the example is because the code is proprietary so I need to make it as minimal as possible to get approval to share the code.

commit 488e8add (2024-08-14): netlists-inference: delay extract push.

Push extract only when inference has detected a loop.
This reduces the number of gates.

Fix #2713