ghdl/ghdl #2714: Failed tests of synth under UCRT64 LLVM

Paebbels, 2024-08-08:
The following tests from testsuite `synth` fail under latest UCRT64 + LLVM 18.
Pipeline: https://github.com/Paebbels/ghdl/actions/runs/10301345088/job/28512846005#step:7:1211

1. synth issue1850: failed
   ```
   Synthesis of pulse.vhdl detector.psl -e
   D:\a\ghdl\ghdl\install\bin\ghdl.exe:note: top entity is "rising_pulse_detector"
   Synthesis of issue2.vhdl -e issue
   issue2.vhdl:122:7:note: found ROM "n276", width: 8 bits, depth: 16
         SERE_4_a : assert always
         ^
   
   ******************** GHDL Bug occurred ***************************
   Please report this bug on https://github.com/ghdl/ghdl/issues
   GHDL release: 5.0.0-dev (tarball) [Dunoon edition]
   Compiled with GNAT Version: 14.2.0
   Target: x86_64-w64-mingw32
   D:\a\ghdl\ghdl\testsuite\synth\issue1850\
   Command line:
   D:\a\ghdl\ghdl\install\bin\ghdl.exe --synth --std=08 issue2.vhdl -e issue
   Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
   Exception information:
   raised ADA.ASSERTIONS.ASSERTION_ERROR : netlists.adb:75
   ******************************************************************
   ```
2. synth issue1366: failed
   ```
   Synthesis of issue.vhdl issue_psl.vhdl -e
   D:\a\ghdl\ghdl\install\bin\ghdl.exe:note: top entity is "issue"
   issue.vhdl:43:21:note: found ROM "n58", width: 8 bits, depth: 10
     data <= to_bit(seq(index));
                       ^
   
   ******************** GHDL Bug occurred ***************************
   Please report this bug on https://github.com/ghdl/ghdl/issues
   GHDL release: 5.0.0-dev (tarball) [Dunoon edition]
   Compiled with GNAT Version: 14.2.0
   Target: x86_64-w64-mingw32
   D:\a\ghdl\ghdl\testsuite\synth\issue[1366](https://github.com/Paebbels/ghdl/actions/runs/10301345088/job/28512846005#step:7:1372)\
   Command line:
   D:\a\ghdl\ghdl\install\bin\ghdl.exe --synth --std=08 issue.vhdl issue_psl.vhdl -e
   Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
   Exception information:
   raised ADA.ASSERTIONS.ASSERTION_ERROR : netlists.adb:75
   ******************************************************************
   ```
3. synth issue1850: failed
   ```
   Synthesis of pulse.vhdl detector.psl -e
   D:\a\ghdl\ghdl\install\bin\ghdl.exe:note: top entity is "rising_pulse_detector"
   Synthesis of issue2.vhdl -e issue
   issue2.vhdl:122:7:note: found ROM "n276", width: 8 bits, depth: 16
         SERE_4_a : assert always
         ^
   
   ******************** GHDL Bug occurred ***************************
   Please report this bug on https://github.com/ghdl/ghdl/issues
   GHDL release: 5.0.0-dev (tarball) [Dunoon edition]
   Compiled with GNAT Version: 14.2.0
   Target: x86_64-w64-mingw32
   D:\a\ghdl\ghdl\testsuite\synth\issue1850\
   Command line:
   D:\a\ghdl\ghdl\install\bin\ghdl.exe --synth --std=08 issue2.vhdl -e issue
   Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
   Exception information:
   raised ADA.ASSERTIONS.ASSERTION_ERROR : netlists.adb:75
   ******************************************************************
   ```
4. synth issue1366: failed
   ```
   Synthesis of issue.vhdl issue_psl.vhdl -e
   D:\a\ghdl\ghdl\install\bin\ghdl.exe:note: top entity is "issue"
   issue.vhdl:43:21:note: found ROM "n58", width: 8 bits, depth: 10
     data <= to_bit(seq(index));
                       ^
   
   ******************** GHDL Bug occurred ***************************
   Please report this bug on https://github.com/ghdl/ghdl/issues
   GHDL release: 5.0.0-dev (tarball) [Dunoon edition]
   Compiled with GNAT Version: 14.2.0
   Target: x86_64-w64-mingw32
   D:\a\ghdl\ghdl\testsuite\synth\issue1366\
   Command line:
   D:\a\ghdl\ghdl\install\bin\ghdl.exe --synth --std=08 issue.vhdl issue_psl.vhdl -e
   Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
   Exception information:
   raised ADA.ASSERTIONS.ASSERTION_ERROR : netlists.adb:75
   ******************************************************************
   ```
5. summray
   ```
   synth test failed (issue1850 issue1366 )
   ```
	 Is it correct to report the cases twice?

tgingold, 2024-08-09:
I think the errors are reported first when they are detected and then at the end.

I am aware of these crashes, but I was never able to reproduce them outside github CI.

Paebbels, 2024-08-09:
Should I run it on my local machine and confirm it?
Do you have access to UCRT64?

I find it strange that it fails in UCRT64 but not in MinGW64 and only for
LLVM.

tgingold, 2024-08-10:
Yes, I tried it locally, without failure.

avelure, 2024-08-14:
I can recreate these locally with UCRT64 LLVM.
Same tests pass with UCRT64 mcode.
This is on win11 with fully updated msys.
```
synth issue1850: failed
Synthesis of pulse.vhdl detector.psl -e
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe:note: top entity is "rising_pulse_detector"
Synthesis of issue2.vhdl -e issue
issue2.vhdl:122:7:note: found ROM "n276", width: 8 bits, depth: 16
      SERE_4_a : assert always
      ^

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r144.gf7663855b.dirty) [Dunoon edition]
Compiled with GNAT Version: 14.2.0
Target: x86_64-w64-mingw32
C:\proj\public\ghdl\testsuite\synth\issue1850\
Command line:
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe --synth --std=08 issue2.vhdl -e issue
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : netlists.adb:75
******************************************************************

synth issue1366: failed
Synthesis of issue.vhdl issue_psl.vhdl -e
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe:note: top entity is "issue"
issue.vhdl:43:21:note: found ROM "n58", width: 8 bits, depth: 10
  data <= to_bit(seq(index));
                    ^

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r144.gf7663855b.dirty) [Dunoon edition]
Compiled with GNAT Version: 14.2.0
Target: x86_64-w64-mingw32
C:\proj\public\ghdl\testsuite\synth\issue1366\
Command line:
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe --synth --std=08 issue.vhdl issue_psl.vhdl -e
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : netlists.adb:75
******************************************************************

synth issue2025: failed
Synthesis of --std=08 wb_standard_axi4_lite_bridge_rtl.vhd wb_standard_axi4_lite_bridge.psl wb_standard_formal_psl.vhd -e wb_standard_axi4_lite_bridge

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r144.gf7663855b.dirty) [Dunoon edition]
Compiled with GNAT Version: 14.2.0
Target: x86_64-w64-mingw32
C:\proj\public\ghdl\testsuite\synth\issue2025\
Command line:
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe --synth --std=08 wb_standard_axi4_lite_bridge_rtl.vhd wb_standard_axi4_lite_bridge.psl wb_standard_formal_psl.vhd -e wb_standard_axi4_lite_bridge
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : netlists.adb:75
******************************************************************

synth issue1609: failed
Synthesis of --std=08 exp_psl.vhdl -e
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe:note: top entity is "exp"

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r144.gf7663855b.dirty) [Dunoon edition]
Compiled with GNAT Version: 14.2.0
Target: x86_64-w64-mingw32
C:\proj\public\ghdl\testsuite\synth\issue1609\
Command line:
C:\msys64\opt\ghdl_llvm\bin\ghdl.exe --synth --std=08 exp_psl.vhdl -e
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : netlists.adb:75
******************************************************************
```

Paebbels, 2024-08-14:
Thanks @avelure.

-------------
I was not able to recreate it.
Windows 11 Pro, MSYS2 was up to date.

avelure, 2024-08-15:
Strange.
How do I best provide more debug info?

tgingold, 2024-08-15:
If you know how to use gdb, set a breakpoint on `__gnat_debug_raise_exception` and print the backtrace on it.

But the interesting point is this is not github CI specific.

avelure, 2024-08-15:
I noticed now that it works fine if I run it natively in windows (e.g. not inside of the ucrt64 mintty).

tgingold, 2024-08-15:
Now fixed.  Thank you for your help!

closed 2024-08-15 by tgingold