ghdl/ghdl #2719: GNATdoc produces no outputs due to broken cross-references

Paebbels, 2024-08-11:
When running gnatdoc .P/ghdl, all source files report broken cross-references:

GNATdoc version from Ubuntu 20.04 - package `gnat-gps`:  
`GNATdoc: /usr/bin/gnatdoc (GNATdoc 19.2 (@GPS_DATE@) hosted on x86_64-linux-gnu))`

`./configure` was used to write the `ghdl.gpr` file.

**Output:**
```
Computing cross-reference information
Collecting source files
warning: cross references for file default_paths.ads are not up-to-date. Documentation not generated.
warning: cross references for file grt-readline.ads are not up-to-date. Documentation not generated.
warning: cross references for file ortho_code-x86-flags.ads are not up-to-date. Documentation not generated.
warning: cross references for file ghdlsynth_maybe.ads are not up-to-date. Documentation not generated.
warning: cross references for file elf_arch.ads are not up-to-date. Documentation not generated.
warning: cross references for file dyn_tables.ads are not up-to-date. Documentation not generated.
warning: cross references for file errorout.ads are not up-to-date. Documentation not generated.
warning: cross references for file std_names.ads are not up-to-date. Documentation not generated.
warning: cross references for file dyn_htables.ads are not up-to-date. Documentation not generated.
warning: cross references for file filesystem.ads are not up-to-date. Documentation not generated.
warning: cross references for file libraries.ads are not up-to-date. Documentation not generated.
warning: cross references for file errorout-console.ads are not up-to-date. Documentation not generated.
warning: cross references for file bug.ads are not up-to-date. Documentation not generated.
warning: cross references for file utils_io.ads are not up-to-date. Documentation not generated.
warning: cross references for file flists.ads are not up-to-date. Documentation not generated.
warning: cross references for file types.ads are not up-to-date. Documentation not generated.
warning: cross references for file str_table.ads are not up-to-date. Documentation not generated.
warning: cross references for file interning.ads are not up-to-date. Documentation not generated.
warning: cross references for file logging.ads are not up-to-date. Documentation not generated.
warning: cross references for file lists.ads are not up-to-date. Documentation not generated.
warning: cross references for file file_comments.ads are not up-to-date. Documentation not generated.
warning: cross references for file name_maps.ads are not up-to-date. Documentation not generated.
warning: cross references for file mutils.ads are not up-to-date. Documentation not generated.
warning: cross references for file areapools.ads are not up-to-date. Documentation not generated.
warning: cross references for file hash.ads are not up-to-date. Documentation not generated.
warning: cross references for file files_map-editor.ads are not up-to-date. Documentation not generated.
warning: cross references for file debuggers.ads are not up-to-date. Documentation not generated.
warning: cross references for file tables.ads are not up-to-date. Documentation not generated.
warning: cross references for file flags.ads are not up-to-date. Documentation not generated.
warning: cross references for file files_map.ads are not up-to-date. Documentation not generated.
warning: cross references for file dyn_interning.ads are not up-to-date. Documentation not generated.
warning: cross references for file errorout-memory.ads are not up-to-date. Documentation not generated.
warning: cross references for file options.ads are not up-to-date. Documentation not generated.
warning: cross references for file simple_io.ads are not up-to-date. Documentation not generated.
warning: cross references for file name_table.ads are not up-to-date. Documentation not generated.
warning: cross references for file dyn_maps.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-prints.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-std_env.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-sensitivity_checks.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-sem.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-comments.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-ieee-numeric.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-configuration.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-ieee-numeric_std_unsigned.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-sem_stmts.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-sem_scopes.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-lists.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-nodes_walk.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-xrefs.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-std_package.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-back_end.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-flists.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-types.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-ieee-std_logic_1164.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-nodes_utils.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-sem_types.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-evaluation.ads are not up-to-date. Documentation not generated.
warning: cross references for file vhdl-tokens.ads are not up-to-date. Documentation not generated.
[... 292 lines cut ...]
warning: cross references for file binary_file-format.ads are not up-to-date. Documentation not generated.
warning: cross references for file elf64.ads are not up-to-date. Documentation not generated.
warning: cross references for file coff.ads are not up-to-date. Documentation not generated.
warning: cross references for file ortho_code-x86.ads are not up-to-date. Documentation not generated.
warning: cross references for file ortho_code-x86-flags_macosx.ads are not up-to-date. Documentation not generated.
warning: cross references for file ortho_mcode.ads are not up-to-date. Documentation not generated.
warning: cross references for file ortho_code-decls.ads are not up-to-date. Documentation not generated.
warning: cross references for file ortho_code-flags.ads are not up-to-date. Documentation not generated.
warning: cross references for file ortho_code-x86-flags_linux.ads are not up-to-date. Documentation not generated.
warning: cross references for file binary_file-elf.ads are not up-to-date. Documentation not generated.
warning: cross references for file ortho_code-x86-flags_windows64.ads are not up-to-date. Documentation not generated.
warning: cross references for file disa_x86.ads are not up-to-date. Documentation not generated.
warning: cross references for file ortho_code-x86-flags_macosx64.ads are not up-to-date. Documentation not generated.
warning: cross references for file ortho_code-binary.ads are not up-to-date. Documentation not generated.
warning: cross references for file dwarf.ads are not up-to-date. Documentation not generated.
warning: cross references for file macho_arch32.ads are not up-to-date. Documentation not generated.
warning: cross references for file elfdumper.ads are not up-to-date. Documentation not generated.
warning: cross references for file ortho_code.ads are not up-to-date. Documentation not generated.
No files to process

Check for generated GNATdoc files/folders:
-rw-r--r-- 1 runner docker [10](https://github.com/Paebbels/ghdl/actions/runs/10339133532/job/28618112312#step:5:11)8K Aug 11 10:46 gnatinspect.db
```

Pipeline: https://github.com/Paebbels/ghdl/actions/runs/10339133532/job/28618112312

Paebbels, 2024-08-11:
Is it possible that GNAT needs a full compilation to get its `*.ali` file, which is needed by GNATdoc?

Paebbels, 2024-08-11:
The list of complaints is now shorter.  
@tgingold Please have a look at the remaining files.

Are these e.g. files for LLVM?  
The code was configured for `mcode` on Ubuntu.

tgingold, 2024-08-11:
Where can we see the list of remaining files ?
(I am not sure how gnatdoc output is useful).

Paebbels, 2024-08-11:
@tgingold I updated the original post with the shorter list and hide the old list in a collapsible region :).

328 files are auto-documented. Looks like 43 are remaining.

------------
The generated documentation looks almost empty/useless due to missing GNATdoc annotations in the code, right?

What would be interesting to have documented?
1. For GHDL developers the internals in case people want to support in the core development. But this would require writing lots of GNATdoc annotations.
2. The libghdl interface, which is public via pyGHDL. Here we have the pyGHDL documentation and higher level documentation e.g. pyGHDL.DOM via Sphinx. Many documentation is imported from pyVHDLModel, as classes are inherited.

For for end users, item 2 might be interesting. For me as a developer of pyGHDL.DOM, I'm struggling with the interface and structure of IIR. Maybe we don't need the API documented, but more like how the IIR looks like (Nodes, subnodes, how to iterate them, how to get child nodes, how to find a parent node, etc.

For now I just recreated what the old pipeline was able to produce, but yes, it's 3MB od empty pages :).

tgingold, 2024-08-12:
Yes, the remaining files are not used in this configuration.

The structure of the tree is defined in `vhdl-nodes.ads`, I am not sure gnatdoc is very useful for this project.

Paebbels, 2024-08-12:
Then I would disable it and remove the placeholder from Sphinx documentation. I'll add a checkbox to the PR for future work.

closed 2024-08-12 by Paebbels