ghdl/ghdl #2721: LLVM-JIT fails with 'version' command in MinGW64 and UCRT64

Paebbels, 2024-08-11:
**Description**
When calling `ghdl version`, the ghdl executable (or the loader?) exits.

Backend:
* `llvm-jit`

Environment:
* `MinGW64` &rarr; https://github.com/Paebbels/ghdl/actions/runs/10339583212/job/28619207680
* `UCRT64` &rarr; https://github.com/Paebbels/ghdl/actions/runs/10339583212/job/28619204508

According to the CI pipeline, the return code is `127`.  
(Isn't this a code used by Bash?)

[image]

**Expected behaviour**
GHDL's version should be printed.

**How to reproduce?**
See pipeline/job links above.

**Context**
Please, provide the following information:

- OS: Windows + MSYS2
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 9567e46f7df97b7d04979f8f6036a825edf879cb

**Additional context**
*None*

tgingold, 2024-08-12:
Probably llvm dll are not in the PATH.

Paebbels, 2024-08-19:
This should be solved. See https://github.com/Paebbels/ghdl/actions/runs/10462074048

closed 2024-08-19 by Paebbels