ghdl/ghdl #2722: Unexpected exit code 0xC0000005 in Windows

jorkrohn, 2024-08-12:
**Description**
When running the provided testbench GDHL returns the exit code -1073741819 (0xC0000005). This seems to only happen in windows. The problem also only occurs when the constant C_ARRAY_SIZE is >= 1022.

**Expected behaviour**
Expected exit code = 0. 

**How to reproduce?**

```vhd :file: ghdl_bug_2_tb.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ghdl_bug_2_tb is
end entity ghdl_bug_2_tb;

architecture sim of ghdl_bug_2_tb is

  type t_slv_array is array (natural range <>) of std_logic_vector;

  constant C_ARRAY_SIZE : natural := 2000;

  type t_record is
  record
    slv_array  : t_slv_array(0 to C_ARRAY_SIZE-1)(7 downto 0);
  end record t_record;

  function generate_record
  return t_record is
    variable v_record : t_record;
  begin
    return v_record;
  end function generate_record;

begin

  p_sequencer : process
    variable v_record : t_record;
  begin
    v_record := generate_record;
    std.env.finish;
  end process p_sequencer;

end architecture sim;
```

```batch :image: ghdl/ghdl:bullseye-mcode
echo off
ghdl -a --std=08 ghdl_bug_2_tb.vhd
ghdl -r --std=08 ghdl_bug_2_tb
echo %ERRORLEVEL%
```

**Context**

- OS: Windows 11 Pro
- Origin:
  - Nightly build

```
GHDL 5.0.0-dev (4.1.0.r141.g9567e46f7) [Dunoon edition]
 Compiled with GNAT Version: 14.2.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

tgingold, 2024-08-14:
This is probably due to a limited stack size.

jorkrohn, 2024-08-14:
Ok, thanks!
So how can I increase the stack size?

Isn't this equal between windows and linux?

tgingold, 2024-08-15:
The handling of the stack is very different between those OSes.
On windows, I suppose there is a tool to change the default stack size.

jorkrohn, 2024-08-16:
Ok, but I still believe there must be a bug in GHDL. 

- It's only when the array is inside a record that the problem occurs. If the array is by itself it doesn't fail. 
- The size of the data structure is not really that big. 2000 bytes is smaller than what's needed in a lot of real use cases.
- The code works with the other simulators I have tried (ModelSim, Riviera-PRO, NVC)

By the way, functionally all my test cases seem to pass. So whatever happens doesn't seem to cause any issues for my simulations except giving a wrong exit code.

tgingold, 2024-08-16:
Does it fail if you replace `std.env.finish;` by `wait;` ?

jorkrohn, 2024-08-16:
With `wait;` it returns exit code 0 as expected

jorkrohn, 2024-08-16:
In my real testbenches I use VUnit where I stop my tests with the test_runner_cleanup() function. This is in turn calling std.env.stop(0); which has the same problem as std.env.finish

tgingold, 2024-08-16:
Ok, I think I understand the issue.

tgingold, 2024-08-16:
Now fixed.  Thank you for the issue and your patience.

jorkrohn, 2024-08-16:
Works great, thanks!