ghdl/ghdl #2723: Passing test with VUnit v4.7.0 with i686 mcode version, failing test with x86_64 mcode version (GHDL 4.1.0 release)

mbermal, 2024-08-13:
**Description**
I have a test suite that uses VUnit (v4.7.0). With different GHDL binaries of the same 4.1.0 release, I get different results. This is the output for --version for both binaries:
```
C:\programs\ghdl>mingw32\bin\ghdl --version
GHDL 4.1.0 (4.0.0.r39.g7188e92cf) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

C:\programs\ghdl>mingw64\bin\ghdl --version
GHDL 4.1.0 (4.0.0.r39.g7188e92cf) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

**Expected behaviour**
The tests should pass in both cases. 
With the i686 version, test pass as expected with the following message:
```
Starting lib.axi_base_tb.Set/Get varying number bytes
Output file: C:\projects\common\misc_ghdl_debug\test\vunit_out\test_output\lib.axi_base_tb.Set_Get_varying_number_bytes_31bc3918fcc32bad43eba686a4c6aa709d76e113\output.txt
pass (P=1 S=0 F=0 T=1) lib.axi_base_tb.Set/Get varying number bytes (0.7 seconds)
```

With the x86_64 version, the following message occurs:
```
Starting lib.axi_base_tb.Set/Get varying number bytes
Output file: C:\projects\common\misc_ghdl_debug\test\vunit_out\test_output\lib.axi_base_tb.Set_Get_varying_number_bytes_31bc3918fcc32bad43eba686a4c6aa709d76e113\output.txt
simulation stopped @0ms with status 0
fail (P=0 S=0 F=1 T=1) lib.axi_base_tb.Set/Get varying number bytes (3.0 seconds)
```
This is not what is expected.

**How to reproduce?**
I have attached a minimal working example. It requires:
Python 3
latest released version of vunit_hdl package (4.7.0)

Run the run.py in a python virtual enviroment with the vunit_hdl package installed.

Set enviroment variable VUNIT_GHDL_PATH to the path to the GHDL binary to use.
By switching the value of VUNIT_GHDL_PATH between the 32-bit and the 64-bit versions of the GHDL package, I can reproduce this error.

**Context**
Please, provide the following information:

- OS: Microsoft Windows 11 Enterprise, 10.0.22631
- Origin:
  - [X] Released binaries: 
  - https://github.com/ghdl/ghdl/releases/download/v4.1.0/mingw-w64-i686-ghdl-mcode-ci-1-any.pkg.tar.zst
  - https://github.com/ghdl/ghdl/releases/download/v4.1.0/mingw-w64-x86_64-ghdl-mcode-ci-1-any.pkg.tar.zst

**Additional context**
Example of issue attached.
[misc_ghdl_debug.zip](https://github.com/user-attachments/files/16604532/misc_ghdl_debug.zip)

mbermal, 2024-08-14:
I tested the same released binaries versions on a Windows 7 Service Pack 1 version of Microsoft Windows. On Windows 7 Service Pack 1, the behavior is the same with VUnit, but Windows 7 does report a crash in the x86_64 version with the generic windows message: "ghdl.exe has stopped working" and "Windows is checking for a solution to the problem".

As with on Windows 11 Enterprise, the i686 version works fine on Windows 7.

tgingold, 2024-08-14:
Interesting.  It would be also interesting to test with the development version in case the issue was already fixed.

mbermal, 2024-08-14:
I tested the following binaries today from the nightly tag (https://github.com/ghdl/ghdl/commit/f7663855bb9db2790fd6234b391072df62c55e0b):
- i686: https://github.com/ghdl/ghdl/releases/download/nightly/mingw-w64-i686-ghdl-mcode-ci-1-any.pkg.tar.zst
- x86_64: https://github.com/ghdl/ghdl/releases/download/nightly/mingw-w64-x86_64-ghdl-mcode-ci-1-any.pkg.tar.zst

The behavior is the same: passing test using i686 version, failing with x86_64, with what I presume is a crash.

Another interesting detail:
I tested it both by deleting the vunit_out directory before switching GHDL versions, and without deleting the vunit_out directory and switching GHDL versions.

When deleting vunit_out directory before switching GHDL versions, a fresh re-compile happens with each GHDL binary test run, as expected. The x86_64 version still fails.

If I don't delete vunit_out directory before switching versions, re-compile does not happen:
~~~
===========================================================================
===========================================================================
Re-compile not needed

Starting lib.axi_base_tb.Set/Get varying number bytes
Output file: C:\projects\common\misc_ghdl_debug\test\vunit_out\test_output\lib.axi_base_tb.Set_Get_varying_number_bytes_31bc3918fcc32bad43eba686a4c6aa709d76e113\output.txt
simulation stopped @0ms with status 0
fail (P=0 S=0 F=1 T=1) lib.axi_base_tb.Set/Get varying number bytes (2.0 seconds)

==== Summary ========================================================
fail lib.axi_base_tb.Set/Get varying number bytes (2.0 seconds)
=====================================================================
pass 0 of 1
fail 1 of 1
=====================================================================
Total time was 2.0 seconds
Elapsed time was 2.0 seconds
=====================================================================
Some failed!
~~~

The behavior is still the same: passing test with i686 version, fail with x86_64 version.

tgingold, 2024-08-15:
Are you sure you are using vunit 4.7.0 ?
It's a little bit old and doesn't support ghdl 4.1

mbermal, 2024-08-15:
I am using vunit 4.7.0, which is the latest official release.

I resolved the issue with vunit and GHDL compatibility by modifying the vunit_hdl site package in my virtual environment. I described it here: https://github.com/VUnit/vunit/issues/1052

This is the output of python -m pip freeze showing the packages and versions of everything in the virtual environment.
~~~
(hdl_test) C:\projects\python_venv\hdl_test\Scripts>python -m pip freeze
click==8.1.7
colorama==0.4.6
edalize==0.5.4
importlib_resources==6.4.0
Jinja2==3.1.4
MarkupSafe==2.1.5
platformdirs==4.2.2
PyYAML==6.0.2
teroshdl==3.0.0
vsg==3.25.0
vunit_hdl==4.7.0
wasmtime==23.0.0
yowasp-runtime==1.63
yowasp-yosys==0.44.0.0.post760
~~~

tgingold, 2024-08-16:
I was not able to reproduce the failure on Linux.

LarsAsplund, 2024-08-20:
I got a similar problem in my project when running the latest pre-release of VUnit (@mbermal install with `pip install --pre`). It works for an older version of GHDL (5.0.0-dev (4.1.0.r13.g0d97a760f, Windows mcode JIT) but fails with 4.1.0.r102.g3b5854759.

GHDL says `simulation stopped @0ms with status 0` but the return code is in fact 0xc0000005 which is a memory access violation I think. I managed to track the problem to this function, i.e. if working around it everything pass again.

``` vhdl
function div_ceil(a, b : integer) return integer is
begin
  return integer(ceil(real(a) / real(b)));
end;
```

My workaround was to simply put the expression inline but in my use case `b` is a constant so I also replaced `real(b)` with a `real`literal. The `div_ceil` function doesn't fail when I put it in an MRE so there is something more to it. It is also used in other places in my project without any issues.

tgingold, 2024-08-25:
@LarsAsplund, is it fixed by #2727

LarsAsplund, 2024-08-26:
@tgingold Unfortunately not, it seems to be something completely unrelated. I will see if I can reduce it to a minimal example that I can submit.