ghdl/ghdl #2725: Crash on analyze:   Local Package with Generic (constants)

JimLewis, 2024-08-19:
**Description**
Local Generic Package Instance with signal declarations (to reduce the rats nest of signal declarations in structural code).

There are two netlists in this issue.   I included the bug reports from both.   The first references signals in the Local Generic Package using selected names.   The second references aliases in the Local Generic Package using selected names.

**Expected behaviour**
Successful Analyze 

**How to reproduce?**
Get the dev branch of OSVVM
     git clone --recursive --branch dev https://github.com/osvvm/OsvvmLibraries

Using OSVVM scripts do:
build $OsvvmLibraries
build $OsvvmLibraries/AXI4/Axi4/RunDemoTests_GenericSignals

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
analyze TbAxi4.vhd
ghdl -a --std=08 -Wno-library -Wno-hide --work=osvvm_TbAxi4_GenericSignals --workdir=C:/tools/VHDL_L
IBS/GHDL-5.0.0-dev/osvvm_tbaxi4_genericsignals/v08 -PC:/tools/VHDL_LIBS/GHDL-5.0.0-dev C:/OsvvmLibra
ries/AXI4/Axi4/testbench_GenericSignals/TbAxi4.vhd
Error:
Error: ******************** GHDL Bug occurred ***************************
Error: Please report this bug on https://github.com/ghdl/ghdl/issues
Error: GHDL release: 5.0.0-dev (4.1.0.r141.g9567e46f7) [Dunoon edition]
Error: Compiled with GNAT Version: 14.2.0
Error: Target: x86_64-w64-mingw32
Error: C:\tools\sim\ghdl\
Error: Command line:
Error: C:\tools\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -Wno-library -Wno-hide --work=osvvm_TbAxi
4_GenericSignals --workdir=C:/tools/VHDL_LIBS/GHDL-5.0.0-dev/osvvm_tbaxi4_genericsignals/v08 -PC:/to
ols/VHDL_LIBS/GHDL-5.0.0-dev\ -PC:\tools\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\tools\msys64\mingw64
\lib/ghdl\ -c -o C:/tools/VHDL_LIBS/GHDL-5.0.0-dev/osvvm_tbaxi4_genericsignals/v08\TbAxi4.o C:/Osvvm
Libraries/AXI4/Axi4/testbench_GenericSignals/TbAxi4.vhd
Error: Exception TYPES.INTERNAL_ERROR raised
Error: Exception information:
Error: raised TYPES.INTERNAL_ERROR : trans-chap2.adb:1539
Error: Load address: 0x7ff712a20000
Error: Call stack traceback locations:
Error: 0x7ff712df56ff 0x7ff712df5b2c 0x7ff712df1f79 0x7ff712df669f 0x7ff712df6cd6 0x7ff712df6dcd 0x7
ff712e04348 0x7ff712e06957 0x7ff712e31d0e 0x7ff712de662c 0x7ff712dd886e 0x7ff712e4d9c1 0x7ff712e4f03
6 0x7ff712e503b0 0x7ff712a2322e 0x7ff712a212ec 0x7ff712a21404 0x7ffe02e97372 0x7ffe0309cc8f
Error: ******************************************************************
C:\tools\msys64\mingw64\bin\ghdl.exe:error: compilation error

... and ...

ghdl -a --std=08 -Wno-library -Wno-hide --work=osvvm_TbAxi4_GenericSignals --workdir=C:/tools/VHDL_L
IBS/GHDL-5.0.0-dev/osvvm_tbaxi4_genericsignals/v08 -PC:/tools/VHDL_LIBS/GHDL-5.0.0-dev C:/OsvvmLibra
ries/AXI4/Axi4/testbench_GenericSignals/TbAxi4Memory.vhd
Error:
Error: ******************** GHDL Bug occurred ***************************
Error: Please report this bug on https://github.com/ghdl/ghdl/issues
Error: GHDL release: 5.0.0-dev (4.1.0.r141.g9567e46f7) [Dunoon edition]
Error: Compiled with GNAT Version: 14.2.0
Error: Target: x86_64-w64-mingw32
Error: C:\tools\sim\ghdl\
Error: Command line:
Error: C:\tools\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -Wno-library -Wno-hide --work=osvvm_TbAxi
4_GenericSignals --workdir=C:/tools/VHDL_LIBS/GHDL-5.0.0-dev/osvvm_tbaxi4_genericsignals/v08 -PC:/to
ols/VHDL_LIBS/GHDL-5.0.0-dev\ -PC:\tools\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\tools\msys64\mingw64
\lib/ghdl\ -c -o C:/tools/VHDL_LIBS/GHDL-5.0.0-dev/osvvm_tbaxi4_genericsignals/v08\TbAxi4Memory.o C:
/OsvvmLibraries/AXI4/Axi4/testbench_GenericSignals/TbAxi4Memory.vhd
Error: Exception TYPES.INTERNAL_ERROR raised
Error: Exception information:
Error: raised TYPES.INTERNAL_ERROR : trans-chap2.adb:1539
Error: Load address: 0x7ff712a20000
Error: Call stack traceback locations:
Error: 0x7ff712df56ff 0x7ff712df5b2c 0x7ff712df1f79 0x7ff712df669f 0x7ff712df6cd6 0x7ff712df6dcd 0x7
ff712e04348 0x7ff712e06957 0x7ff712e31d0e 0x7ff712de662c 0x7ff712dd886e 0x7ff712e4d9c1 0x7ff712e4f03
6 0x7ff712e503b0 0x7ff712a2322e 0x7ff712a212ec 0x7ff712a21404 0x7ffe02e97372 0x7ffe0309cc8f
Error: ******************************************************************
C:\tools\msys64\mingw64\bin\ghdl.exe:error: compilation error

```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

JimLewis, 2024-08-19:
@Paebbels   WRT your gitter:
GHDL release: 5.0.0-dev (4.1.0.r141.g9567e46f7) [Dunoon edition]  (in the message above)
Running on Windows 10.

closed 2024-08-25 by tgingold

JimLewis, 2024-08-26:
@tgingold    Thank you.
I have downloaded a new version of GHDL from the nightly updates and tested this on my machine and can verify it passes.

JimLewis, 2024-08-26:
BTW, I presented the usage of local packages in my Part 1 of my AXI4 VC presentation on  8/22 and said that while it is not currently working GHDL today, my bet is that by Part 2 of my AXI4 VC presentation on 9/5 that it will be.

tgingold, 2024-08-26:
Great!

commit 2a545eec (2024-08-25): trans-chap2: handle kind_alias in Copy_Info

Fix #2725