ghdl/ghdl #2728: Elaborated design available through pyGHDL

nrhint, 2024-08-20:
**Is your feature request related to a problem? Please describe.**
Summary:
Is there a plan to allow elaboration through the LSP? (https://github.com/ghdl/ghdl/issues/1753#issuecomment-834634597)

Background:
I am working on a program that parses VHDL and then converts it to other formats and manipulates it. In the past we wrote code that parses VHDL. Recently I have been working on integrating GHDL into the program to use GHDL to parse and analyze the code. I have been using the LSP for this and it has been very excellent. However I ran into a snag today where I needed to get the value of a constant from a package. Here are minimum examples:

my_pkg.vhd:
`
library IEEE;
use IEEE.math_real.all;
use work.my_pkg.all;

package my_other_pkg is

  constant FOO : real := 100;
  constant BAR : natural := my_func(integer(FOO));

end package my_other_pkg;
`

my_other_pkg.vhd:
`
library IEEE;
use IEEE.math_real.all;
use IEEE.numeric_std.all;

package my_pkg is

  function my_func (num : natural) return natural is
  begin
    return num / 3 + 1;
  end function my_func;

end package my_pkg;
`

Looking at what I can of the tree there is no elaborated value for the constant `BAR`. Other times there are values such as: `constant BAR : natural := FOO/3'` does provide a value (accessed through `nodes.Get_Default_Value`) From what I can tell if a function is builtin to the IEEE library the value is elaborated/resolved however if it is a user function it is not. I know that GHDL can actually resolve this because if I simulate the code it does have the correct value in either case.

**Describe the solution you'd like**
I would like to be able to get the elaborated values of constants and other declarations through the LSP

**Describe alternatives you've considered**
The alternative would be to return the function to the other program and then run the function there however that is what we are trying to avoid and remove from that program.

**Additional context**
I did go through an try to create an endpoint in the LSP that would perform this action and it works for many constants and signals (haven't finished all the declarations yet) however it fails in this case (which is my problem) and I can not find the actual value anywhere in the LSP (or XML) tree

Paebbels, 2024-08-21:
I'm not sure LSP (`pyGHDL.LSP`) is the the right interface. There is `pyGHDL.libghdl` as a low-level binding to `libghdl...{so|dll|dylib}`. Both, LSP and DOM are using it to access the provided features (parse, analyze, ...).

`pyGHDL.DOM` is an implementation of `pyVHDLModel` and provides access to the parsed content like entities, ports, types, etc. It can also figure out the dependency graph, hierarchy and compile order. 

Currently, no elaborate is called, but this could be done. Are you able to contribute to that feature? I'm still bound in providing all parsed elements to the CodeDOM.

nrhint, 2024-08-21:
I am totally willing to contribute. I am not familiar with ada but am competent with python. What I have done currently interfaces with `pyGHDL.libghdl` to get the information that I am wanting. As for how to elaborate not quite sure how to do that. If you could though give me kind of a rough outline of what needs to be done I will muddle through it and it can be refined on a PR. So far I have spent most of the time looking into `pyGHDL.LSP` for what I want however I will also see if I can find the information in `pyGHDL.DOM`

tgingold, 2024-08-25:
I don't think you need to write any Ada code for this feature.
If you want to stay with the low-level `libghdl` binding, you need to add bindings to elaborate a design.

As a starting point, you can look at `src/ghdldrv/ghdlsynth.adb`, in particular `ghdl_synth` function.  This function calls
`Elab.Vhdl_Insts.Elab_Top_Unit` to elaborate a design.  It returns an instance. 

Then you need to write bindings to navigate through a design+instance and to get the value of an object (constant or variable).  You can see how it is done in `elab-vhdl_debug.adb`.