ghdl/ghdl #2731: make test; pyunit test path problems

avelure, 2024-08-23:
**Description**
When running `make && make test`, it fails during the pyunit test as `libghdl-5_0_0_dev.dll` is not found even though it is present in `./lib/libghdl-5_0_0_dev.dll` in the cwd build directory (<git_repo>/build).
I see in CI the test is skipped and the other tests are run manually from the testsuite.
The issue seems to be in `_get_libghdl_path` in `libGHDL/__init__.py`. When `GHDL` is set in the environment, which it is from the makefile, it expects ghdl_mcode.exe to be in a bin folder, but in the build folder it is not.
Further down in the function it has `Try when running from the build directory`, but then it seems to expect that you are building directly in the root of the repo, instead of a separate build folder.

Moving the lib folder from `build` to the root repo folder gives another error that the `bin` folder is not present in the root of the repo.

**Expected behaviour**
Test should pass.

**How to reproduce?**
```
mkdir build && cd build
../configure --prefix=/opt/ghdl_mcode
make && make test
```

```
    import pyGHDL.libghdl as libghdl
<frozen importlib._bootstrap>:1176: in _find_and_load
    ???
<frozen importlib._bootstrap>:1147: in _find_and_load_unlocked
    ???
<frozen importlib._bootstrap>:690: in _load_unlocked
    ???
C:/msys64/ucrt64/lib/python3.11/site-packages/_pytest/assertion/rewrite.py:174: in exec_module
    exec(co, module.__dict__)
../pyGHDL/libghdl/__init__.py:187: in <module>
    libghdl = _initialize()
../pyGHDL/libghdl/__init__.py:161: in _initialize
    _libghdl_path = _get_libghdl_path()
../pyGHDL/libghdl/__init__.py:155: in _get_libghdl_path
    raise Exception(f"Cannot find libghdl {basename}")
E   Exception: Cannot find libghdl libghdl-5_0_0_dev.dll
```
```
pyunit/libghdl/Comments.py:4: in <module>
    import pyGHDL.libghdl as libghdl
<frozen importlib._bootstrap>:1176: in _find_and_load
    ???
<frozen importlib._bootstrap>:1147: in _find_and_load_unlocked
    ???
<frozen importlib._bootstrap>:690: in _load_unlocked
    ???
C:/msys64/ucrt64/lib/python3.11/site-packages/_pytest/assertion/rewrite.py:174: in exec_module
    exec(co, module.__dict__)
../pyGHDL/libghdl/__init__.py:187: in <module>
    libghdl = _initialize()
../pyGHDL/libghdl/__init__.py:168: in _initialize
    os_add_dll_directory(str(p1))
<frozen os>:1119: in add_dll_directory
    ???
E   FileNotFoundError: [WinError 2] The system cannot find the file specified: 'C:/proj/public/ghdl/bin'
```

**Context**
Please, provide the following information:

- OS: win11 ucrt64 mcode
- Origin: Built from sources: ea01390c7

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

Paebbels, 2024-08-23:
It looks like `make install` is missing.

avelure, 2024-08-23:
It doesn't make a difference.

Paebbels, 2024-09-23:
I'm reworking GHDL's pipeline. See [Paebbels/ghdl:simplify-pipeline](https://github.com/Paebbels/ghdl/tree/simplify-pipeline).

Searching libghdl has changed as well as how pyGHDL wheels are built.

Paebbels, 2024-12-19:
@avelure does this still apply?

avelure, 2024-12-20:
`make test` seems to get a bit further now, most of the python tests run. 
I now only get 3 errors related to the language server.
Unless this is a setup issue, I could make a new issue maybe.
```
pyunit/lsp/LanguageServer.py::Test001_Simple::test_Request_Response PASSED
pyunit/lsp/LanguageServer.py::Test002_Coverage::test_Request_Response Test002_Coverage
[8].params.diagnostics[0].severity: mismatch (ref: 1, result: 2)
[8].params.diagnostics[1].severity: mismatch (ref: 1, result: 2)
[8].params.diagnostics[2].severity: mismatch (ref: 1, result: 2)
[8].params.diagnostics[3].severity: mismatch (ref: 1, result: 2)
Test002_Coverage
[10].params.diagnostics[0].severity: mismatch (ref: 1, result: 2)
[10].params.diagnostics[1].severity: mismatch (ref: 1, result: 2)
[10].params.diagnostics[2].severity: mismatch (ref: 1, result: 2)
[10].params.diagnostics[3].severity: mismatch (ref: 1, result: 2)
Test002_Coverage
[11].params.diagnostics[0].severity: mismatch (ref: 1, result: 2)
[11].params.diagnostics[1].severity: mismatch (ref: 1, result: 2)
[11].params.diagnostics[2].severity: mismatch (ref: 1, result: 2)
[11].params.diagnostics[3].severity: mismatch (ref: 1, result: 2)
Test002_Coverage
[13].params.diagnostics[0].severity: mismatch (ref: 1, result: 2)
FAILURE between output and C:/proj/public/ghdl/testsuite/pyunit/lsp/002coverage/replies.json (for C:/proj/public/ghdl/testsuite/pyunit/lsp/002coverage/cmds.json)
Writing result output to result.json
FAILED
pyunit/lsp/LanguageServer.py::Test003_Errors::test_Crash1 PASSED
pyunit/lsp/LanguageServer.py::Test003_Errors::test_Crash2 PASSED
pyunit/lsp/LanguageServer.py::Test003_Errors::test_Request_Response Test003_Errors
[1].params.diagnostics[0].severity: mismatch (ref: 1, result: 2)
FAILURE between output and C:/proj/public/ghdl/testsuite/pyunit/lsp/003errors/replies.json (for C:/proj/public/ghdl/testsuite/pyunit/lsp/003errors/cmds.json)
Writing result output to result.json
FAILED
pyunit/lsp/LanguageServer.py::Test004_Error_Project::test_Request_Response PASSED
pyunit/lsp/LanguageServer.py::Test005_Create::test_Request_Response PASSED
pyunit/lsp/LanguageServer.py::Test006_Option_Error::test_Request_Response PASSED
pyunit/lsp/LanguageServer.py::Test007_Error_Project::test_Request_Response PASSED
pyunit/lsp/LanguageServer.py::Test008_Error_NoFile::test_Request_Response PASSED
pyunit/lsp/LanguageServer.py::Test009_ls_122::test_Request_Response Test009_ls_122
[1].params.diagnostics[0].severity: mismatch (ref: 1, result: 2)
[1].params.diagnostics[1].severity: mismatch (ref: 1, result: 2)
[1].params.diagnostics[2].severity: mismatch (ref: 1, result: 2)
[1].params.diagnostics[3].severity: mismatch (ref: 1, result: 2)
FAILURE between output and C:/proj/public/ghdl/testsuite/pyunit/lsp/009ls122/replies.json (for C:/proj/public/ghdl/testsuite/pyunit/lsp/009ls122/cmds.json)
Writing result output to result.json
FAILED
pyunit/lsp/LanguageServer.py::Test010_ls_28::test_Request_Response PASSED
pyunit/lsp/LanguageServer.py::Test011_closediag::test_Request_Response PASSED
pyunit/lsp/LanguageServer.py::Test012_library::test_Request_Response PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_Comments PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_arch_before PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_arch_inside PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_arch_inside_2 PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_arch_inside_fail XFAIL
pyunit/libghdl/Comments.py::CommentAssociation::test_array PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_conf_before PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_conf_inside PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_conf_inside_fail XFAIL
pyunit/libghdl/Comments.py::CommentAssociation::test_const PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_const_fail XFAIL
pyunit/libghdl/Comments.py::CommentAssociation::test_ctxt_before PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_ctxt_inside PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_ctxt_inside_fail XFAIL
pyunit/libghdl/Comments.py::CommentAssociation::test_element_1 PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_element_2 PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_element_3 PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_element_4 PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_elements_fail XFAIL
pyunit/libghdl/Comments.py::CommentAssociation::test_entity_arch PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_entity_before PASSED
pyunit/libghdl/Comments.py::CommentAssociation::test_entity_inside PASSED
[... 119 lines cut ...]
PASSED pyunit/libghdl/Comments.py::MultiLine::test_Complex
PASSED pyunit/libghdl/Initialize.py::Instantiate::test_InitializeGHDL
XFAIL pyunit/libghdl/Comments.py::CommentAssociation::test_arch_inside_fail - reason:
XFAIL pyunit/libghdl/Comments.py::CommentAssociation::test_conf_inside_fail - reason:
XFAIL pyunit/libghdl/Comments.py::CommentAssociation::test_const_fail - reason:
XFAIL pyunit/libghdl/Comments.py::CommentAssociation::test_ctxt_inside_fail - reason:
XFAIL pyunit/libghdl/Comments.py::CommentAssociation::test_elements_fail - reason:
XFAIL pyunit/libghdl/Comments.py::CommentAssociation::test_enum_fail - reason:
XFAIL pyunit/libghdl/Comments.py::CommentAssociation::test_enumlit_fail - reason:
[... cut ...]
FAILED pyunit/lsp/LanguageServer.py::Test002_Coverage::test_Request_Response - AssertionError: None
FAILED pyunit/lsp/LanguageServer.py::Test003_Errors::test_Request_Response - AssertionError: None
FAILED pyunit/lsp/LanguageServer.py::Test009_ls_122::test_Request_Response - AssertionError: None
====================================================================================== 3 failed, 49 passed, 16 xfailed in 4.71s ======================================================================================
make: *** [Makefile:243: test.mcode] Error 1

```

Paebbels, 2024-12-20:
This is pyunit in the CI server (Ubuntu + Python 3.12): https://github.com/ghdl/ghdl/actions/runs/12407567493/job/34638452639#step:14:48

It tests `pyghdl.libghdl` and `pyghdl.lsp`. `pyghdl.dom` has currently some failing tests and is disabled in CI.

Can you compare `make test` with calling pytest manually (like in CI):
```bash
cd <ghdlRoot>

export PYTHONPATH=$(pwd)
  
python -m pytest -raP --junitxml=report/unit/TestReportSummary.xml --color=yes testsuite/pyunit/libghdl testsuite/pyunit/lsp
```

avelure, 2024-12-20:
I get the same errors.
```
$ export PYTHONPATH=$(pwd)

$ python -m pytest -raP --junitxml=report/unit/TestReportSummary.xml --color=yes testsuite/pyunit/libghdl testsuite/pyunit/lsp
================================================================================================ test session starts =================================================================================================
platform win32 -- Python 3.12.7, pytest-8.3.3, pluggy-1.5.0
rootdir: C:/proj/public/ghdl/testsuite/pyunit
configfile: pytest.ini
collected 68 items

testsuite/pyunit/libghdl/Comments.py ....x...x.x..x....x....x...x.x.x.....xx..x.x..x.x.x..                                                                                                                      [ 77%]
testsuite/pyunit/libghdl/Initialize.py .                                                                                                                                                                        [ 79%]
testsuite/pyunit/lsp/LanguageServer.py .F..F.....F...                                                                                                                                                           [100%]

====================================================================================================== FAILURES ======================================================================================================
_______________________________________________________________________________________ Test002_Coverage.test_Request_Response _______________________________________________________________________________________

self = <LanguageServer.Test002_Coverage testMethod=test_Request_Response>

    def test_Request_Response(self):
>       self._RequestResponse("cmds.json", "replies.json")

testsuite/pyunit/lsp/LanguageServer.py:176:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
testsuite/pyunit/lsp/LanguageServer.py:162: in _RequestResponse
    self.fail()
E   AssertionError: None
------------------------------------------------------------------------------------------------ Captured stdout call ------------------------------------------------------------------------------------------------
Test002_Coverage
[8].params.diagnostics[0].severity: mismatch (ref: 1, result: 2)
[8].params.diagnostics[1].severity: mismatch (ref: 1, result: 2)
[8].params.diagnostics[2].severity: mismatch (ref: 1, result: 2)
[8].params.diagnostics[3].severity: mismatch (ref: 1, result: 2)
Test002_Coverage
[10].params.diagnostics[0].severity: mismatch (ref: 1, result: 2)
[10].params.diagnostics[1].severity: mismatch (ref: 1, result: 2)
[10].params.diagnostics[2].severity: mismatch (ref: 1, result: 2)
[10].params.diagnostics[3].severity: mismatch (ref: 1, result: 2)
Test002_Coverage
[11].params.diagnostics[0].severity: mismatch (ref: 1, result: 2)
[11].params.diagnostics[1].severity: mismatch (ref: 1, result: 2)
[11].params.diagnostics[2].severity: mismatch (ref: 1, result: 2)
[11].params.diagnostics[3].severity: mismatch (ref: 1, result: 2)
Test002_Coverage
[13].params.diagnostics[0].severity: mismatch (ref: 1, result: 2)
FAILURE between output and C:/proj/public/ghdl/testsuite/pyunit/lsp/002coverage/replies.json (for C:/proj/public/ghdl/testsuite/pyunit/lsp/002coverage/cmds.json)
Writing result output to result.json
________________________________________________________________________________________ Test003_Errors.test_Request_Response ________________________________________________________________________________________

self = <LanguageServer.Test003_Errors testMethod=test_Request_Response>

    def test_Request_Response(self):
>       self._RequestResponse("cmds.json", "replies.json")

testsuite/pyunit/lsp/LanguageServer.py:189:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
testsuite/pyunit/lsp/LanguageServer.py:162: in _RequestResponse
    self.fail()
E   AssertionError: None
------------------------------------------------------------------------------------------------ Captured stdout call ------------------------------------------------------------------------------------------------
Test003_Errors
[1].params.diagnostics[0].severity: mismatch (ref: 1, result: 2)
FAILURE between output and C:/proj/public/ghdl/testsuite/pyunit/lsp/003errors/replies.json (for C:/proj/public/ghdl/testsuite/pyunit/lsp/003errors/cmds.json)
Writing result output to result.json
________________________________________________________________________________________ Test009_ls_122.test_Request_Response ________________________________________________________________________________________

self = <LanguageServer.Test009_ls_122 testMethod=test_Request_Response>

    def test_Request_Response(self):
[... 14 lines cut ...]
Writing result output to result.json
======================================================================================================= PASSES =======================================================================================================
[... cut ...]
FAILED testsuite/pyunit/lsp/LanguageServer.py::Test002_Coverage::test_Request_Response - AssertionError: None
FAILED testsuite/pyunit/lsp/LanguageServer.py::Test003_Errors::test_Request_Response - AssertionError: None
FAILED testsuite/pyunit/lsp/LanguageServer.py::Test009_ls_122::test_Request_Response - AssertionError: None
====================================================================================== 3 failed, 49 passed, 16 xfailed in 5.38s ======================================================================================
```