ghdl/ghdl #2732: Simualtion taking very long time

Biswajitks1, 2024-08-23:
We have a very big design nearly 100 files to analyze. When we attempted to run the simulation, it quit before executing the run simulation command of ghdl. By using GDB(GNU project debugger) it came to knew that there is issue with the stack size of system when the stack size of system was changed to unlimited the ghdl simulation did run but it takes around 3 hours of time to do the simulation. While the Xilinx Vivado does it in few minutes.
Can anyone help me to figure out this?
The ghdl version used is:-

GHDL 5.0.0-dev (4.1.0.r63.g9bcaa92a2.dirty) [Dunoon edition]
 Compiled with GNAT Version: 10.5.0
 llvm 14.0.0 code generator
Written by Tristan Gingold.

We tried to use the mcode ghdl as it has time resolution option but that didn't solve the issue.

tgingold, 2024-08-25:
Unfortunately, we need a reproducer to investigate this issue.  The description is too vague.

Biswajitks1, 2024-08-31:
I have grouped all the files in the zip atached. The vhdl_simulation.sh script runs the ghdl in herarchial way. If the ulimit -s unlimited is commented out then the ghdl stops the run command(please refer the terminal picture).
[image](https://github.com/user-attachments/assets/255473c4-d435-470d-bdd1-691a910a5c23)

Explanation:-
The vhdl files define a FPGA fabric of the [FABulous framework](https://github.com/FPGA-Research-Manchester/FABulous). The user design is a sequential counter and the generated bitstream is converted to hex and loaded in the testbench the output generated by FPGA and user_design is compared.
[test_files.zip](https://github.com/user-attachments/files/16824460/test_files.zip)

tgingold, 2024-08-31:
I can run the design.
You have a very large number of processes (~800K), and currently there is no inter-process optimizations.
I will have a closer look on how to deal with such a design.

Biswajitks1, 2024-08-31:
Did the run complete? I mean how much time did it take to run the design.
Thank you for looking into this.

tgingold, 2024-09-01:
No, I have stopped it before the end as it progressed too slowly.

Biswajitks1, 2024-09-11:
Hi, I did some observations on RAM consumption.The RAM consumption is not exceeding more than 1.3GB. I don't know whether this is a issue or not.

tgingold, 2024-09-13:
No, that should be OK.  It would be nice if it were lower.

Biswajitks1, 2024-10-07:
Hi, were you able to identify the issue? I was thinking is there any way that we can say ghdl to use more resources and can that reduce simulation time?

tgingold, 2024-10-09:
I haven't worked on it yet.  It's a long time project.

Biswajitks1, 2025-05-18:
This issue has been resolved. It was related to the clock speed, and @EverythingElseWasAlreadyTaken solved the issue. He may be able to share more details or other possible causes.

closed 2025-05-18 by Biswajitks1