ghdl/ghdl #2734: Unnecessary time tags printed in VCD output file

Umoma, 2024-08-25:
**Description**
In simulation, in order to reduce the size of a VCD output file to the only relevant signals, a filter Wave option file is provided.
Only expected signal value changes are printed as expected. However, unnecessary time tags are printed, since no signal value changes takes place.

I have been wondering if this can be tagged as bug or feature. I've finally decided to mark as bug. My argument is due to the impact in some designs (like my case), it could be the difference from generating VCD dump files from several GiBs to a few MiBs. Note this file are later machine processed (testing,...) and the performance hit is considerable.

**Expected behaviour**
Only meaningful timestamps tags should be printed.

**How to reproduce?**
Three signals are declared updated at different time intervals. Only the slowest `low_freq` will be printed in the VCD file. 

Test VHDL file `tb.vhd`:
```vhd :file: tb.vhd
library IEEE;

use ieee.std_logic_1164.all;

entity tb is 
end entity;

architecture behav of tb is
    constant semiperiod      : time := 5 ns;
    signal clk               : std_logic := '0';
    signal low_freq          : std_logic := '0';
    signal counter           : natural   := 0; 
begin
    clk <= not clk after semiperiod;
    low_freq <= '1' when counter = 0 else '0';

    process (clk)
    begin
        if rising_edge(clk) then
            if counter = 9 then
                counter <= 0;
            else
                counter <= counter + 1;
            end if;
        end if;
    end process;
end;
```
Wave option file provided `filter.signals`:
```vcd :file: filter.signals
$ version 1.1

# Signals in packages :

# Signals in entities :
/tb/low_freq

```

ghdl calls:
```sh :image: ghdl/ghdl:bullseye-llvm
ghdl -a tb.vhd
ghdl elab-run tb --vcd=out.vcd --read-wave-opt=filter.signals --stop-time=30ns
```
Output file `out.vcd` prints as follows:
```vcd :file: out.vcd
$date
  Sun Aug 25 16:07:33 2024
$end
$version
  GHDL v0
$end
$timescale
  1 fs
$end
$scope module tb $end
$var reg 1 ! low_freq $end
$upscope $end
$enddefinitions $end
#0
1!
#5000000
0!
#10000000
#15000000
#20000000
#25000000
#30000000
```

However, I would expect `out.vcd` to be like:
```vcd :file: out.vcd
$date
  Sun Aug 25 16:07:33 2024
$end
$version
  GHDL v0
$end
$timescale
  1 fs
$end
$scope module tb $end
$var reg 1 ! low_freq $end
$upscope $end
$enddefinitions $end
#0
1!
#5000000
0!
#30000000
```

**Context**

- OS:
`Linux debian 6.1.0-23-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.99-1 (2024-07-15) x86_64 GNU/Linux`
- Origin:
  - [ ] Package manager: 
  - [ ] Released binaries:
  - [x] Built from sources: `commit 7188e92cf3cc60f5d97cfe83bc2c85f0250373ab (HEAD, tag: v4.1.0)`

```sh
$ ghdl --version
GHDL 4.1.0 (4.1.0.r0.g7188e92cf) [Dunoon edition]
 Compiled with GNAT Version: 12.2.0
 llvm 14.0.6 code generator
```

tgingold, 2024-08-28:
Indeed, this could be optimized.

Umoma, 2024-08-29:
Thanks a lot for your reply. Unfortunately I am not familiar with Ada, so I am unable to contribute with code.