ghdl/ghdl #2738: VPI works differntly when compiling to shared library

reriv1, 2024-08-31:
**Description**
When I compile VHDL sources to an executable, VPI works as expected, when I compile to a shared library, VPI works unexpectedly.

**Expected behaviour**
See the example VHDL code below. The VHDL has a clk signal that is type std_logic. If I compile to an executable, VPI reports size of the clk signal as 1 (expected). When I compile to a shared library, size of clk is reported as 8 (unexpected).

**How to reproduce?**
Simple VHDL example code is as follows:

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity tb is
end entity;

architecture sim of tb is
    signal clk : std_logic := '0';
begin
    clk <= not clk after 5 ns;
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a tb.vhdl
ghdl -e -shared -o tb.so tb
gcc -I$GHDL_INC_DIR -ldl sim.c -o sim
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS: Linux Mint 21.3
- Origin: Nightly build from 1-2 weeks ago (ghdl-gha-ubuntu-22.04-llvm.tgz )
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
The C code that prints the size of the VHDL clk signal is as follows:

```
#include <vpi_user.h>
#include <dlfcn.h>
#include <stdio.h>

typedef void (*grt_init_t)(void);
typedef int (*grt_elab_t)(void);
typedef void (*sim_init_t)(void);
typedef vpiHandle (vpi_handle_by_name_t)(PLI_BYTE8, vpiHandle);
typedef PLI_INT32 (*vpi_get_t)(PLI_INT32, vpiHandle);

int main(int argc, void** argv) {
  void* h = dlopen("./[tb.so](http://tb.so/)", RTLD_LAZY);
  grt_init_t grt_init = (grt_init_t) dlsym(h, "grt_init");
  grt_elab_t grt_elab = (grt_elab_t) dlsym(h, "grt_main_elab");
  sim_init_t sim_init = (sim_init_t) dlsym(h, "__ghdl_simulation_init");
  vpi_handle_by_name_t vpi_handle_by_name = (vpi_handle_by_name_t) dlsym(
  h, "vpi_handle_by_name");
  vpi_get_t vpi_get = (vpi_get_t) dlsym(h, "vpi_get");

  grt_init();
  grt_elab();
  sim_init();

  vpiHandle net = vpi_handle_by_name("tb.clk", NULL);
  printf("size %d\n", vpi_get(vpiSize, net));

  dlclose(h);
  return 0;
}

```

tgingold, 2024-09-04:
I fear the shared library is not correctly built: `-fpic` is not used when `tb.vhdl` is analyzed.
In addition, the precompiled `ieee` or `std` libraries are not built with `-fpic`.

reriv1, 2024-09-07:
Thanks. I decided to download the 4.1.0 sources, and I'm now compiling those for the llvm backend. After configure and then running make, I can see the VHDL libraries being compiled, and I can see that -fpic isn't specified for them. Is there currently an easy way to specify that the VHDL libraries should be built with -fpic?

reriv1, 2024-09-07:
I couldn't see any recommended way of setting -fPIC for compiling the VHDL libraries, so what I did was in the 6 places "-P../.." appears in the libraries/Makefile.inc file, I changed it to read "-P../.. -fPIC", and now when I do a fresh configure and then a make I can see all the VHDL libraries being compiled with -fPIC. After make I do a make install to complete the process. Also, the analysis of the example VHDL file given above is now done with: ghdl -a -fPIC tb.vhd.

All to no avail at the moment, the C code is still showing the clock signal as size 8. I tried -fpic as well as fPIC to check if it made a difference but it doesn't.

tgingold, 2024-09-08:
I think what you want to do is possible.
But you need to follow the initialization steps of `src/grt/ghdl_main.adb`.  You probably need to also call `Run_Options` and `Register_Modules`.

However, you are using it in a way which is not really supported and will probably not be supported in the future.
Why do you need to use as a shared library ?

reriv1, 2024-09-08:
Thanks, I'll take another look and have another go.

The reason for trying the shared library option is because I'm working on a gui for interacting with simulators using VPI/VHPI, and viewing waves as a simulation progresses. The plan is to do things like e.g. start the sim gui, load a sim by loading a GHDL sim compiled to shared library, add signal waves, run sim, stop sim, restart sim by unloading GHDL sim shared library and reloading it to reset and run again, start a different sim by loading a different GHDL sim shared library. In other words, some of the things you would do with a commercial simulator gui.

If I don't use the shared library feature, then I wouldn't be able to do the things described above. For example, to restart/reset a sim, or switch to another sim, the user would need to close down the sim gui, and start over. So at the moment I'm thinking the GHDL shared library feature is a nice feature for this application.

My sim gui currently has quite a bit of the basics more or less done (draw/zoom/pan waves, add/delete/move cursors, Tcl command window, etc), so I decided to see if I could connect it up to GHDL. My plan is to make the gui code available on github (or equivalent) once it has enough functionality to be useful to others.

tgingold, 2024-09-09:
I fear using shared libraries this way is a dead end.
First, as far as I know, only GHDL supports this model, but as you can see not completely.  And in the near future, it is possible this model is not supported anymore.
Second, in case of crash, your application will crash, which is not ideal.
Finally, the resources might not be free after a simulation.

The simulation should run in a different process to benefit from isolation features provided by the OS.

reriv1, 2024-09-09:
OK, thanks. From time to time I get crashes with commercial simulators, which is irritating, and process isolation would be good for this. CXXRTL looks interesting as an alternative to VPI/VHPI, it'll be interesting to see how that progresses. For now I'll use VPI with GHDL sim executables instead of shared libraries, as this will allow me to continue.

gatk555, 2024-09-21:
I think this can work with a shared library, as I have rather similar code working since April.  During initialisation, the code scans the ports of the simulated entity and gets their sizes.  This bug is not seen.

My examples use std_logic, but I have not encountered any problems with PIC.

Some differences: for GHDL initialisation there is a call to ghdl_main().  That does not return until simulation ends, but the rest of the program runs in another thread.  All VPI calls are in a separate VPI module, loaded by an argument to ghdl_main().  Some shim code is included in the shared library built by GHDL, and ghdl_main() is called there.

If it will be helpful, I can show the Makefile, or complete code.  It is quite small, less than 3000 lines, as the code is just a shim to adapt GHDL execution to an extension interface of the host program, Ngspice.

reriv1, 2024-11-02:
Thanks gatk555, I don't need example makefile/code, thanks however for the offer, and also thanks for describing a method that works, i.e. passing separate VPI module to ghdl_main.

I did notice actually on this cosim page:

https://ghdl.github.io/ghdl-cosim/vhpidirect/dynamic.html 

this text:

"When GHDL is configured with --default-pic explicitly, it uses it implicitly when executing any -a, -e or -r command. Hence, it is not required to provide these arguments (fPIC/PIE) to GHDL"

When I was originally looking at this I wasn't adding --default-pic when configuring, so I tried again compiling GHDL with this option. I also changed my example C code to call ghdl_main instead of the more finely grained functions used in my original code, to take that out of the equation. My example C code is now:

```
#include <vpi_user.h>
#include <dlfcn.h>
#include <stdio.h>

typedef int (*ghdl_main_t)(int, char**);
typedef vpiHandle (*vpi_handle_by_name_t)(PLI_BYTE8*, vpiHandle);
typedef PLI_INT32 (*vpi_get_t)(PLI_INT32, vpiHandle);

int main(int argc, void** argv) {
  void* h = dlopen("./tb.so", RTLD_LAZY);

  ghdl_main_t ghdl_main = (ghdl_main_t) dlsym(h, "ghdl_main");
  vpi_handle_by_name_t vpi_handle_by_name = (vpi_handle_by_name_t) dlsym(
      h, "vpi_handle_by_name");
  vpi_get_t vpi_get = (vpi_get_t) dlsym(h, "vpi_get");

  printf("main %d\n", ghdl_main(0, NULL));
  vpiHandle net = vpi_handle_by_name("tb.clk", NULL);
  printf("size %d\n", vpi_get(vpiSize, net));

  dlclose(h);
  return 0;
}

```
The output from this C code is:

main 0
size 8

So the result is still the same, clk size is reported in error as 8.

My build script for the above was:

```
ghdl -a tb.vhdl
ghdl -e -shared -o tb.so tb
gcc -I$GHDL_INC_DIR -ldl sim.c -o sim
```

reriv1, 2024-11-02:
Just for completeness, I should also say I changed tb.vhdl to the code shown below, so that the call to ghdl_main returns.

```
library ieee;
use ieee.std_logic_1164.all;

entity tb is
end entity;

architecture sim of tb is
    signal clk : std_logic := '0';
begin
    clk_p : process is
    begin
        for i in 0 to 9 loop
            clk <= not clk;
        end loop;
        wait;
    end process;
end;
```