ghdl/ghdl #2740: GHDL Bug Occured no field Parameter

Haron123, 2024-09-01:
**Description**
When trying to convert a UART Core from VHDL to Verilog via GHDL to use it in Litex, this Issue arrises.

**Expected behaviour**
I expect to get a verilog file converted from vhdl but i get an error. 

**How to reproduce?**
To reproduce, get the UART Core from https://github.com/jakubcabal/uart-for-fpga
Then try to convert the VHD Files to verilog via

```
ghdl --synth --out=verilog --std=08 --no-formal -fsynopsys -gCLK_FREQ=4000000.0 -gBAUD_RATE=9600 -gPARITY_BIT=none -gUSE_DEBOUNCER=0 own_periphs/uart/uart.vhd own_periphs/uart/comp/uart_clk_div.vhd own_periphs/uart/comp/uart_debouncer.vhd own_periphs/uart/comp/uart_parity.vhd own_periphs/uart/comp/uart_rx.vhd own_periphs/uart/comp/uart_tx.vhd own_periphs/spi/spi_master.vhd -e UART
```
(Path should be changed to place of Installation of the Files for every Source)

**Context**
Please, provide the following information:

- OS: 6.6.46-1-MANJARO
- Origin:
  - [ x ] Package manager: `4.1.0 and 5.0.0(dev) have been tested`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.1.0 (tarball) [Dunoon edition]
Compiled with GNAT Version: 14.1.1 20240720
Target: x86_64-pc-linux-gnu
/home/haron/Desktop/gatemate_litex/
Command line:
ghdl --synth --out=verilog --std=08 --no-formal -fsynopsys -gCLK_FREQ=4000000.0 -gBAUD_RATE=9600 -gPARITY_BIT=none -gUSE_DEBOUNCER=0 own_periphs/uart/comp/uart.vhd own_periphs/uart/comp/uart_clk_div.vhd own_periphs/uart/comp/uart_debouncer.vhd own_periphs/uart/comp/uart_parity.vhd own_periphs/uart/comp/uart_rx.vhd own_periphs/uart/comp/uart_tx.vhd own_periphs/spi/spi_master.vhd -e UART
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : no field Parameter
Call stack traceback locations:
0x704601 0x540c4a 0x61828a 0x65d005 0x6d7a18 0x6d7c23 0x6ac784 0x40d3ee 0x7f80a1ed4e06 0x7f80a1ed4eca 0x40d453 0xfffffffffffffffe
******************************************************************
```

**Additional context**
This is an effort to use the UART VHDL Code in Litex as Core, so it uses the vhd2v converter Module from Litex
https://github.com/enjoy-digital/litex/blob/master/litex/build/vhd2v_converter.py

Incase there are any workarounds to this, please let me know.

tgingold, 2024-09-02:
The generic `CLK_FREQ` is an integer but you are giving a real value.

Haron123, 2024-09-02:
Thanks a lot tgingold, that really was the problem. Thanks for the quick response

commit f39edb8e (2024-09-03): vhdl-evaluation: do not use Get_Parameter for error

Fix #2740
closed 2024-09-02 by Haron123