ghdl/ghdl #2741: GHDL-gcc build fails on CS8

abyszuk, 2024-09-02:
Hello,

Some time ago my CI build of GHDL broke with a compile error.
I'm not sure if this is a bug in GHDL, or if you decided to drop support for older platforms. Here are some details:
OS: CentOS Stream 8 (installed GCC version: 8.5.0)
GHDL version: git master
I'm building the `gcc` backend against GCC 10

Build fails with the following error:
```
export PATH=$PATH:/usr/gnat/bin && \
    cd /tmp/gcc && \
    ./contrib/download_prerequisites && \
    cd /tmp/ghdl/ && \
    mkdir build && \
    cd build && \
    ../configure --with-gcc=/tmp/gcc --prefix=/usr/local && \
    make copy-sources && \
    mkdir gcc-objs && \
    cd gcc-objs && \
    /tmp/gcc/configure --prefix=/usr/local --enable-languages=c,vhdl \
    --disable-bootstrap --disable-lto --disable-multilib --disable-libssp \
    --disable-libgomp --disable-libquadmath && \
    make -j4 && make install && \
    cd /tmp/ghdl/build && \
    make ghdllib && \
    make install 

(... snip ...)

#17 703.1 gcc -c -I./ -I/tmp/gcc/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /tmp/ghdl/build/gcc-objs/gcc/vhdl/verilog-bn_tables.o /tmp/gcc/gcc/vhdl/verilog-bn_tables.ads
#17 703.4 gcc -c -I./ -I/tmp/gcc/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /tmp/ghdl/build/gcc-objs/gcc/vhdl/verilog-macros.o /tmp/gcc/gcc/vhdl/verilog-macros.ads
#17 703.7 gnatmake -o ghdl1 -aI/tmp/gcc/gcc/vhdl -aOvhdl ortho_gcc-main \
#17 703.7          -cargs -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC \
#17 703.7  -bargs -E \
#17 703.7  -largs --LINK=g++ -no-pie   -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  attribs.o vhdl/ortho-lang.o vhdl/grt-cstdio.o vhdl/grt-cdynload.o \
#17 703.7  libbackend.a libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a  -L/tmp/ghdl/build/gcc-objs/./isl/.libs  -lisl -L/tmp/ghdl/build/gcc-objs/./gmp/.libs -L/tmp/ghdl/build/gcc-objs/./mpfr/src/.libs -L/tmp/ghdl/build/gcc-objs/./mpc/src/.libs -lmpc -lmpfr -lgmp -rdynamic -ldl  -L./../zlib -lz -lzstd 
#17 703.7 gnatbind -aI/tmp/gcc/gcc/vhdl -aOvhdl -E -x /tmp/ghdl/build/gcc-objs/gcc/vhdl/ortho_gcc-main.ali
#17 704.7 error: elaboration circularity detected
#17 704.7 info:    "synth.vhdl_context (spec)" must be elaborated before "elab.vhdl_types (body)"
#17 704.7 info:       reason: with clause
#17 704.7 
#17 704.7 info:    "synth.vhdl_environment (spec)" must be elaborated before "synth.vhdl_context (spec)"
#17 704.7 info:       reason: with clause
#17 704.7 
#17 704.7 info:    "synth.verilog_stmts (body)" must be elaborated before "synth.vhdl_environment (spec)"
#17 704.7 info:       reason: implicit Elaborate_All in unit "synth.vhdl_environment (spec)"
#17 704.7 info:       recompile "synth.vhdl_environment (spec)" with -gnatel for full details
#17 704.7 info:          "synth.verilog_stmts (body)"
#17 704.7 info:             must be elaborated along with its spec:
#17 704.7 info:          "synth.verilog_stmts (spec)"
#17 704.7 info:             which is withed by:
#17 704.7 info:          "synth.verilog_insts (body)"
#17 704.7 info:             which must be elaborated along with its spec:
#17 704.7 info:          "synth.verilog_insts (spec)"
#17 704.7 info:             which is withed by:
#17 704.7 info:          "synthesis (body)"
#17 704.7 info:             which must be elaborated along with its spec:
#17 704.7 info:          "synthesis (spec)"
#17 704.7 info:             which is withed by:
#17 704.7 info:          "synth.vhdl_insts (body)"
#17 704.7 info:             which must be elaborated along with its spec:
#17 704.7 info:          "synth.vhdl_insts (spec)"
#17 704.7 info:             which is withed by:
#17 704.7 info:          "synth.vhdl_stmts (body)"
#17 704.7 info:             which must be elaborated along with its spec:
#17 704.7 info:          "synth.vhdl_stmts (spec)"
#17 704.7 info:             which is withed by:
#17 704.7 info:          "synth.vhdl_oper (body)"
#17 704.7 info:             which must be elaborated along with its spec:
[... 123 lines cut ...]
#17 704.7 info:             which is withed by:
#17 704.7 info:          "netlists.memories (body)"
#17 704.7 info:             which must be elaborated along with its spec:
#17 704.7 info:          "netlists.memories (spec)"
#17 704.7 info:             which is withed by:
#17 704.7 info:          "netlists.inference (body)"
#17 704.7 info:             which must be elaborated along with its spec:
#17 704.7 info:          "netlists.inference (spec)"
#17 704.7 info:             which is withed by:
#17 704.7 info:          "synth.environment (body)"
#17 704.7 info:             which must be elaborated along with its spec:
#17 704.7 info:          "synth.environment (spec)"
#17 704.7 info:             which is withed by:
#17 704.7 info:          "synth.environment.debug (spec)"
#17 704.7 info:             which is withed by:
#17 704.7 info:          "synth.verilog_environment (spec)"
#17 704.7 
#17 704.7 gnatmake: *** bind failed.
#17 704.7 make[2]: *** [/tmp/gcc/gcc/vhdl/Make-lang.in:66: ghdl1] Error 4
#17 704.7 rm gcov.pod fsf-funding.pod gfdl.pod gpl.pod cpp.pod gcc.pod gcov-dump.pod gcov-tool.pod
#17 704.7 make[2]: Leaving directory '/tmp/ghdl/build/gcc-objs/gcc'
#17 704.7 make[1]: *** [Makefile:4398: all-gcc] Error 2
#17 704.7 make[1]: Leaving directory '/tmp/ghdl/build/gcc-objs'
#17 704.7 make: *** [Makefile:961: all] Error 2
```

tgingold, 2024-09-03:
That's weird as this error has never been in the CI.
Maybe your gcc/gnat version is too old ?

Did this error appear recently ?  Do you often rebuild ghdl ?

abyszuk, 2024-09-03:
> Maybe your gcc/gnat version is too old ?  

That's what I've been thinking too, but according to GHDL documentation this should be fine
I use GNAT 2017. According to documentation that's the last version I can batch install without GUI.

The error appeared sometime between end of March and August. Do you want me to bisect?

tgingold, 2024-09-03:
Did you try to compile using gnat-gcc provided by centos ?
I don't think I have used GNAT GPL for a while, but I can test if (if I find it).

Paebbels, 2024-09-04:
GHDL's [CI pipeline](https://github.com/Paebbels/ghdl/blob/simplify-pipeline/.github/workflows/Build-MacOS.yml#L65) uses a precompiled GNAT from "Builds of the GNAT Ada compiler from FSF GCC releases" ([GNAT 14.2.0-1](https://github.com/alire-project/GNAT-FSF-builds/releases/tag/gnat-14.2.0-1)) for macOS builds, because there is no package in homebrew. The release also contains many other platforms. There is a generic Linux build: `gnat-x86_64-linux-14.2.0-1.tar.gz`.

Could this work?

abyszuk, 2024-09-04:
> Did you try to compile using gnat-gcc provided by centos ?  
So that's the thing - there's no gnat-gcc provided by centos. At least not for C8/CS8.

I'll try with the FSF GNAT build.

Meanwhile, I've tested that building against newer GCC (i.e. against GCC 13). So it's either problem with the GCC 8.5 or GNAT GPL.

abyszuk, 2024-09-04:
OK, it works with FSF GNAT version.
I think we can close this issue, but it would be good to update the documentation.

Thanks for your help!

tgingold, 2024-09-04:
The doc has been recently updated for #2730.

Thank you for the update!

closed 2024-09-04 by tgingold