ghdl/ghdl #2744: NULL access dereferenced on ArchLinux x86_64, llvm backend

augustofg, 2024-09-04:
**Description**
This is a very tricky issue that I'm having problems reproducing in other environments. Using the same GHDL version, but running on a Debian installation doesn't cause problems. Basically, the simulation aborts with an `NULL access dereferenced` error, but VHDL `access` types are not used anywhere in the source files.

**Expected behaviour**
Testbench executes successfully on all platforms.

**How to reproduce?**
Extract [ghdl-null-bug.zip](https://github.com/user-attachments/files/16853445/ghdl-null-bug.zip) and run on a x86_64 ArchLinux installation with [ghdl-llvm-git](https://aur.archlinux.org/packages/ghdl-llvm-git) (AUR) installed:
```
$ cd ghdl-null-bug
$ make run
ghdl -a --std=08 -g fofb_cc_version.vhd
ghdl -a --std=08 -g fofb_cc_pkg.vhd
ghdl -a --std=08 -g wishbone_pkg.vhd
ghdl -a --std=08 -g dot_prod_pkg.vhd
ghdl -a --std=08 -g fofb_sys_id_pkg.vhd
ghdl -a --std=08 -g fofb_ctrl_pkg.vhd
ghdl -a --std=08 -g fofb_tb_pkg.vhd
ghdl -a --std=08 -g sim_wishbone.vhd
ghdl -a --std=08 -g wb_fofb_shaper_filt_regs_consts_pkg.vhd
ghdl -a --std=08 -g xwb_fofb_shaper_filt_tb.vhd
ghdl -a --std=08 -g ifc_common_pkg.vhd
ghdl -a --std=08 -g wb_fofb_shaper_filt_regs.vhd
ghdl -a --std=08 -g xwb_fofb_shaper_filt.vhd
ghdl -a --std=08 -g iir_filt.vhd
ghdl -a --std=08 -g xwb_register_link.vhd
ghdl -a --std=08 -g wb_slave_adapter.vhd
ghdl -a --std=08 -g biquad.vhd
ghdl -a --std=08 -g wb_skidpad.vhd
ghdl -e --std=08 -g xwb_fofb_shaper_filt_tb
ghdl -r xwb_fofb_shaper_filt_tb 
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
./xwb_fofb_shaper_filt_tb:error: NULL access dereferenced
in process .xwb_fofb_shaper_filt_tb(test).uut@xwb_fofb_shaper_filt(behave).P4
./xwb_fofb_shaper_filt_tb:error: simulation failed
make: *** [Makefile:21: run] Error 1
```

**Context**
Please, provide the following information:

- OS: ArchLinux
- Origin:
  - Built from sources: `b467eacbafa2d0db270b5892c8ec1ed50c0aff31`

tgingold, 2024-09-05:
I can reproduce the crash and I am investigating

closed 2024-09-05 by tgingold

augustofg, 2024-09-05:
Thanks @tgingold !

commit 7865415f (2024-09-05): trans-chap7.adb: fix array aggregate with vector and others

Fix #2744