ghdl/ghdl #2746: Fix for synthesis bug broke synthesis

JJ-Gaisler, 2024-09-06:
It looks like commit: https://github.com/ghdl/ghdl/commit/488e8addb25b486c382f18902051157c87f333ef broke the Verilog synthesis functionally. When I tested the fix I only checked that synthesis no longer crashed, I had other issues in my design which kept me from testing the functionality. However I have fixed those issues now and I did a git bisect from a version of ghdl I knew was working. This led me to the aforementioned commit.

It simply looks like after this commit my design no longer works.

I have tried to run yosys equivalence checking on a design generated from the older working version of ghdl and I have gotten some simple modules to pass. However, I have also seen that when yosys tries to generate its .il representation the OOM killer comes in and kills the process when processing verilog from the newer version but not using the older working version. The system I am running on has 94 GiB of ram so I am suspecting bad code in that module.

If I try to run synthesis with ghdl + yosys I get the error: `ERROR: Wire not found for $posedge`.

The steps I used to do this.
Download latest version of [grlib ](https://www.gaisler.com/products/grlib/grlib-gpl-2024.2-b4293.tar.gz):

1. Unpack and cd to desgins/noelv-generic
2. Build ghdl with commit: ec315f0a6be497c70aaf7a06b7d3d9461d372b27 (golden)
3. Add the following to the Makefile:
```
PREFIX_GENERICS = \
-gfabtech=0 \
-gmemtech=0 \
-gmularch=0 \
-gncpu=1 \
-gnextmst=0 \
-gnextslv=1 \
-gnextapb=0 \
-gndbgmst=1 \
-gnintdom=0 \
-gneiid=63 \
-gcached=0 \
-gwbmask=16\#50F\# \
-gbusw=128 \
-gcmemconf=0 \
-grfconf=1 \
-gfpuconf=0 \
-gtcmconf=0 \
-gmulconf=0 \
-gdisas=0 \
-gahbtrace=0 \
-gcfg=0 \
-gdevid=0 \
-gnodbus=0 \
-gtrace=0 \
-gscantest=0

TOP = gaisler.noelvsys
VERILOG_OUT=noelv.v
ifeq ($(YOSYS), 1)

YOSYS_OPTS= proc; peepopt; fsm; opt;  memory; opt; clean; check; write_verilog -sv $(VERILOG_OUT)

$(VERILOG_OUT): ghdl
         yosys -m ghdl -p 'ghdl --no-formal -fexplicit --ieee=synopsys --mb-comments --warn-no-binding --warn-no-hide $(PREFIX_GENERICS) $(shell cat ghdl.path) -O0 --workdir=gnu/work --work=work $(TOP); $(YOSYS_OPTS)'
else
$(VERILOG_OUT): ghdl
        $(GHDL) --synth --out=verilog --no-formal -fexplicit --ieee=synopsys --mb-comments --warn-no-binding --workdir=gnu/work $(PREFIX_GENERICS) --work=work  $(shell cat ghdl.path) --warn-no-hide --warn-no-specs --warn-no-port $(TOP) > $(VERILOG_OUT)
        sed -i 's/\<do\>/dao/g' $(VERILOG_OUT) #verilog keywords
        sed -i 's/\<break\>/break_s/g' $(VERILOG_OUT) #verilog keywords
endif
```
4. Run make noelv.v GHDL=`<path to golden ghdl>`
5. move noelv.v to noelv_golden.v
6. Build ghdl with latest commit (gate)
7. move noelv.v noelv_bad.v
8. Make sure Yosys, Eqy and Yices are installed
9. run the following .eqy script: `eqy script.eqy`
```
[gold]
read_verilog -sv noelv_golden.v
prep -top div64_0_0_1_0_1
#prep -top iunv_0_0_0_32_41_786432_16_128_32_0_0_0_4_4_0_0_1_2_0_8_10_8_4_0_196624_196625_1_1_1_1_1_1_1_1_1_1_1_1_1_1_1_1_0_0_1_0_0_0_0_1_1_1_1_1_1_0_0_0_0_1_1_1_64_0_82_1_1_2_589824_4_0_0_1_1_1_0_0_1
#prep -top cctrlnv_0_32_4_8_4_4_8_4_0_0_0_1_0_1_16_16_16_2_0_8_10_8_4_1_1_1_1_1_1_0_1295_128_64_4_4_4_255_1_1_1_1_276e8824c88e117571aa19e570895b56ecbd3507
memory_map

[gate]
read_verilog -sv noelv_bad.v
prep -top div64_0_0_1_0_1
#prep -top iunv_0_0_0_32_41_786432_16_128_32_0_0_0_4_4_0_0_1_2_0_8_10_8_4_0_196624_196625_1_1_1_1_1_1_1_1_1_1_1_1_1_1_1_1_0_0_1_0_0_0_0_1_1_1_1_1_1_0_0_0_0_1_1_1_64_0_82_1_1_2_589824_4_0_0_1_1_1_0_0_1
#prep -top cctrlnv_0_32_4_8_4_4_8_4_0_0_0_1_0_1_16_16_16_2_0_8_10_8_4_1_1_1_1_1_1_0_1295_128_64_4_4_4_255_1_1_1_1_276e8824c88e117571aa19e570895b56ecbd3507
memory_map

[collect *]
group div64*
#group iunv*
#group cctrlnv*

[strategy sby]
use sby
depth 7
engine smtbmc
```
10. I expect div64 to pass successfully, when trying with cctrlnv I see the aforementioned error with OOM killer. If run yosys golden.ys that works fine if I instead run yosys gate.ys that works. So perhaps this is a way to start identifying the bad code?
11. Reproduce issue with wire for posedge, make noelv.v YOSYS=1

tgingold, 2024-09-07:
Thank you for this effort.

I am not sure about `cctrlnv.vhd` around line 8240:
```
  srstregs: if GRLIB_CONFIG_ARRAY(grlib_async_reset_enable) = 0 generate
    regs: process(clk, csro)
    begin
      if rising_edge(clk) then
        r   <= rin;
        if GRLIB_CONFIG_ARRAY(grlib_sync_reset_enable_all) /= 0 and rst = '0' then
          r <= RRES;
        end if;
      end if;
      -- These are really set in IUNV in NOEL-V.
      if is_riscv then
        -- Replace CCTRL with CSR_CCTRL
        r.cctrl.dsnoop  <= csro.cctrl.dsnoop;
        r.cctrl.dcs     <= csro.cctrl.dcs;
        r.cctrl.ics     <= csro.cctrl.ics;
        r.cctrl.ics_btb <= csro.cctrl.ics;
        r.cctrl.dfrz    <= '0';
        r.cctrl.ifrz    <= '0';
      end if;
    end process;
```

This part creates the `wire not found` error in yosys.

Are golden.ys and gate.ys part of grlib ?

tgingold, 2024-09-07:
I can synthesize with yosys the generated noelv.v (and my system has less than 92GB)

I haven't run formal equivalence yet.  Tell me if the issue is fixed for you.

TheSaltyLiquorice, 2024-09-07:
> 
> Are golden.ys and gate.ys part of grlib ?
> 

Sorry for not being clear about these. These files are generated by eqy when you run the equivalence checking. It looks to me like eqy is just generating scripts to be run in yosys. The first step of the equivalence checking seems to be to convert the designs into immediate representation. The scripts for doing that are called golden.ys and gate.ys. So if you run eqy, you should find these in the output folder of the run. The reason for mentioning these is that gold.ys works while running standalone, gate.ys doesn't despite coming from the same vhdl source.

JJ-Gaisler, 2024-09-09:
It looks like synthesis is working again with both yosys and ghdl :)

closed 2024-09-09 by JJ-Gaisler

tgingold, 2024-09-10:
Great!  Thank you for the status update.