ghdl/ghdl #2748: Interactive simulation using CXXRTL protocol

oscargus, 2024-09-06:
**Is your feature request related to a problem? Please describe.**
It would be great if one can interact with GHDL using a wave form viewer to simulate interactively.

**Describe the solution you'd like**
The wave form viewer Surfer, https://surfer-project.org/, supports the CXXRTL protocol https://gist.github.com/whitequark/59520e2de0947da8747061bc2ea91639
Hence, if GHDL supported CXXRTL it would be possible to interactively run the simulation.

**Describe alternatives you've considered**
Coming up with a separate protocol.

**Additional context**
This was briefly discussed at FSiC, but I thought I'd open a tracking and discussion issue for it.

It seems like the protocol has not define "force_signal" or anything related, but this can probably be added as well to enable simulations without a test bench.

tgingold, 2024-09-08:
If I understand correctly, I first have to implement read of saved waveforms.
Then I could add the CXXRTL protocol.

That's not a small task!

oscargus, 2024-09-09:
I *think* that an alternative option may be to add a different "waveform backend" that stores previous values in memory and responds to CXXRTL queries. That is no small task either, but *maybe* easier compared to reading saved waveforms. I am also a bit doubtful if writing and reading does not add an unnecessary overhead since the main benefit with this would be to run simulations in partial steps. If the whole waveform is saved, one may as well open it in Surfer directly.

However, from a Surfer perspective, I am sure that we can support any protocol that may seem suitable.