ghdl/ghdl #2749: Generic package in entity statement cannot include functions

rramunni, 2024-09-09:
**Description**
Trying to use a generic package within the generic map of an entity statement in order to set port widths. This works fine for the example in  #1262 , however when that package also has a function, GHDL throws an exception.

**Expected behaviour**
Expected to not throw an error, instead for the package instantiation to work without any issues.

**How to reproduce?**

```vhd :file: pkg.vhd
library ieee;
use ieee.std_logic_1164.all;

package slv is
	generic(
		N: integer
	);
	subtype slv_t is std_logic_vector(N-1 downto 0);

	function foo return integer;

end package;

package body slv is

	function foo return integer is
	begin
		return 12345;
	end function foo;

end package body slv;
```

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
	generic(
		 WIDTH: integer;
		 package slv_pkg is new work.slv
		 	generic map(N => WIDTH)
	);
	port (
		o_slv: out slv_pkg.slv_t
	);
end ent;

architecture beh of ent is
	constant ones : std_logic_vector(WIDTH-1 downto 0) := (others => '1');
begin
	o_slv <= ones;
end architecture beh;
```

```vhd :file: tb.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tb is
end entity;

architecture test of tb is
	constant w : integer := 3;

	package slv_tb_pkg is new work.slv generic map(N => w);

	component ent
		generic(
			WIDTH: integer
		);
		port (
			o_slv: out slv_tb_pkg.slv_t
		);
	end component ent;

	signal s_out : std_logic_vector(w-1 downto 0);
begin
	e : ent
		generic map(
			WIDTH => w
		)
		port map(
			o_slv => s_out
		);

	process
	begin
		wait for 1 ns;
		report integer'image(to_integer(unsigned(s_out)));
		wait;
	end process;
end architecture test;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 pkg.vhd
ghdl -a --std=08 ent.vhd
```

**Context**
Please, provide the following information:

- OS: Ubuntu 22.04
- Origin:
  - [ ] Package manager: `version`
  - [X ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r823.g9e2b6fc95) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
/home/rramunni/dev/test_bug/
Command line:
/home/rramunni/.local/bin/ghdl1-llvm --std=08 -P/home/rramunni/.local/lib/ghdl/ieee/v08/ -P/home/rramunni/.local/lib/ghdl/ -fpic -c -o ent.o ent.vhd
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
Call stack traceback locations:
0x7fc5a1a7751e 0x7fc5a32f64b3 0x7fc5a31c49ba 0x7fc5a31fc826 0x7fc5a31fc78e 0x556285d95876 0x556285f4eca4 0x556285f4ec2b 0x556285f4ef96 0x556285f4f251 0x55628611644f 0x556286126342 0x55628612688b 0x5562861c6052 0x5562861d4256 0x5562861d435d 0x55628618213f 0x556286174dd5 0x5562861d8609 0x5562861da05b 0x5562861db6af 0x556285d8efd1 0x7fc5a1a5ed8e 0x7fc5a1a5ee3e 0x556285d8d803 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error
```

tgingold, 2024-09-10:
It doesn't fail on the head of git.
So I suppose the bug has been fixed meanwhile as you aren't using the latest version of ghdl.

I will add your reproducer in the testsuite.

closed 2024-09-10 by tgingold

rramunni, 2024-09-10:
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r203.g6ace5ac8c) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
/home/rramunni/dev/test_bug/
Command line:
/home/rramunni/.local/bin/ghdl1-llvm --std=08 -P/home/rramunni/.local/lib/ghdl/ieee/v08/ -P/home/rramunni/.local/lib/ghdl/ -fpic -c -o ent.o ent.vhd
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
Call stack traceback locations:
0x7fbe8c11351e 0x7fbe8d9924b3 0x7fbe8d8609ba 0x7fbe8d898826 0x7fbe8d89878e 0x561134803aa6 0x5611349bfec4 0x5611349bfe4b 0x5611349c01b6 0x5611349c0471 0x561134b8be9d 0x561134b9bd90 0x561134b9c2d9 0x561134c2e0c6 0x561134c3c2f6 0x561134c3c3fd 0x561134be9bf9 0x561134bdc88f 0x561134c4f0af 0x561134c50b01 0x561134c52155 0x5611347fd0e1 0x7fbe8c0fad8e 0x7fbe8c0fae3e 0x5611347fb903 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error

Downloaded the nightly release and got the same issue.

rramunni, 2024-09-10:
Also get this result with the nightly version of ghdl-gcc 
```
GHDL 5.0.0-dev (4.1.0.r203.g6ace5ac8c) [Dunoon edition]
 Compiled with GNAT Version: 10.5.0
 GCC 12.1.0 code generator
```

```
/home/rramunni/dev/test_bug/ent.vhd: In function ‘work__ent__DECL_ELAB’:
/home/rramunni/dev/test_bug/ent.vhd:4: internal compiler error: Segmentation fault
    4 | entity ent is
      |
0xefbaf3 crash_signal
        ../../gcc-srcs/gcc/toplev.cc:322
0x7fc67843e51f ???
        ./signal/../sysdeps/unix/sysv/linux/x86_64/libc_sigaction.c:0
0x7b21bb new_selected_element
        ../../gcc-srcs/gcc/vhdl/ortho-lang.cc:1522
0x9f1ec8 trans__chap10__find_scope
        /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/gcc-srcs/gcc/vhdl/trans.adb:576
0x9f2049 trans__chap10__get_instance_ref
        /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/gcc-srcs/gcc/vhdl/trans.adb:612
0x9f2134 trans__chap10__get_var
        /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/gcc-srcs/gcc/vhdl/trans.adb:631
0x9fbd33 trans__chap3__get_composite_type_layout
        /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/gcc-srcs/gcc/vhdl/trans-chap3.adb:61
0xa0430f trans__chap3__get_composite_bounds
        /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/gcc-srcs/gcc/vhdl/trans-chap3.adb:2999
0xa0472d trans__chap3__get_array_length
        /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/gcc-srcs/gcc/vhdl/trans-chap3.adb:3065
0xa579d2 trans__chap9__merge_signals_rti
        ../../gcc-srcs/gcc/vhdl/trans-foreach_non_composite.adb:49
0xa57d62 trans__chap9__merge_signals_rti_of_port_chain
        /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/gcc-srcs/gcc/vhdl/trans-chap9.adb:2965
0xa5d780 trans__chap9__elab_block_declarations
        /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/gcc-srcs/gcc/vhdl/trans-chap9.adb:2983
0xa3160f trans__chap1__translate_entity_subprograms
        /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/gcc-srcs/gcc/vhdl/trans-chap1.adb:221
0xa2b20b translation__translate
        /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/gcc-srcs/gcc/vhdl/translation.adb:213
0xa2f4fc ortho_front__do_compile
        /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/gcc-srcs/gcc/vhdl/ortho_front.adb:422
0xa2f4fc ortho_front__parse
        /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/gcc-srcs/gcc/vhdl/ortho_front.adb:679
0x9e5d89 lang_parse_file
        /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/gcc-srcs/gcc/vhdl/ortho_gcc.adb:123
0x7add19 ortho_parse_file
        ../../gcc-srcs/gcc/vhdl/ortho-lang.cc:500
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
ghdl-gcc:error: exec error
```

tgingold, 2024-09-10:
You should try with the mcode backend.

rramunni, 2024-09-10:
Is this not supported in llvm/gcc?

tgingold, 2024-09-11:
They have a limited support on generics because they compile the files before elaboration.
While the jit backends elaborates before generating code.