ghdl/ghdl #2751: Synthesis Verilog Issue: Signed Division

m-stockinger, 2024-09-16:
**Description**
When running GHDL synth and output the netlist to Verilog, the signed division is done on unsigned values instead of signed values.

**Expected behaviour**
Signed Division should be done with signed values.

**How to reproduce?**
```vhd :file: top.vhd
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

entity top is
  generic (
    data_width : natural := 4
  );
  port (
    clk_i        : in    std_logic;
    rst_i        : in    std_logic;
    data_numer_i : in    std_logic_vector(data_width - 1 downto 0);
    data_denom_i : in    std_logic_vector(data_width - 1 downto 0);
    data_o       : out   std_logic_vector(data_width - 1 downto 0)
  );
end entity top;

architecture rtl of top is

  signal data, next_data : std_logic_vector(data_width - 1 downto 0);

begin

  reg_proc : process (clk_i, rst_i) is
  begin

    if (rst_i = '1') then
      data <= (others => '0');
    elsif rising_edge(clk_i) then
      data <= next_data;
    end if;

  end process reg_proc;

  next_data <= std_logic_vector(signed(data_numer_i) / signed(data_denom_i));

  data_o <= data;

end architecture rtl;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl --synth --out=verilog top.vhd -e > syn_top.v
```

Resulting in a Verilog netlist which contains the line:
`assign n7_o = data_numer_i / data_denom_i; // sdiv`
but I would expect it to be 
`assign n7_o = $signed(data_numer_i) / $signed(data_denom_i); // sdiv`

**Context**
- OS: Ubuntu 23.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `6ace5ac8cf2b89c3c217d12953de55ae20c64fc8`

**Additional context**
I think this issue can be solved quite easily by editing this file `/ghdl/src/synth/netlists-disp_verilog.adb` and replacing line 1018
`Disp_Template ("  assign \o0 = \i0 / \i1; // sdiv" & NL, Inst);`
with
`Disp_Template ("  assign \o0 = \si0 / \si1; // sdiv" & NL, Inst);`

tgingold, 2024-09-17:
Indeed.

Thank you for the issues and the testcases!

closed 2024-09-17 by tgingold

m-stockinger, 2024-09-17:
Thanks for the quick fixes and thanks a lot for your dedication to this project.

commit 3c2eaf57 (2024-09-17): netlists-disp_verilog: fix sdiv

Fix #2751